frequenz-floss/frequenz-sdk-python #5: Implement a data resampling actor

leandro-lucarella-frequenz, 2022-09-30:
We need to implement a data resampling actor that takes care of receiving data from the microgrid and resamples it.

### Problem

The data can come from the microgrid at different intervals for different components. The data could potentially even come out of order, or there can be missing samples. Also formulas can be calculated using samples at different times for different components. Dealing with this is not easy so it shouldn't be done by applications or other use case actors directly.

### Proposed solution

Limit the `MicrogridData` actor to only do the formula calculation (it should be renamed and split further, see the data pipeline redesign slides and #520). Create a new resampling actor (`MicrogridDataResamplingActor`) that takes care of resampling component's timeseries data, so the resulting timeseries are separated by a fixed amount of time.

* The resampling should be done in a separate class (`MicrogridDataResampler`)
  * The actor should just be a thin wrapper over this resampling class.
  * The `MicrogridDataResampler` should use another class, `ComponentDataResampler`, to handle each component's resampling.

* The resampling rate should be configurable in the constructor (and there is only one for all the components being resampled).

* We start with a fixed size ring buffer for each component, storing the last samples that were received.

* There is a timer configured to fire a new sample needs to be produced.

* There is a (configurable via the constructor) `maximum_data_age` that is the threshold to consider the data in the ring buffer relevant. Data older than `maximum_data_age` is considered irrelevant.

* When the timer fires, if there is no relevant data in the ring buffer, a special value (`NaN`, `Unknown`, `None` or similar) should be sent.

* When the timer fires, if there is relevant data in the ring buffer, the resampling is done by feeding the contents of the relevant data in the ring buffer to a configurable (also in the constructor) resampling function (average by default). The resampling function should also take the the resampling rate, This way it is possible for the resampling function to split samples into resampling windows if necessary.

* The data sent for a particular sampling window should have the **exact same timestamp** (i.e. the timestamp needs to be generate once per sampling window and reused in all outgoing messages for that sampling window).

* The timestamp should be the one at the end of the window (so the instant where the resampling timer fired). So if the current sampling window we are about to send if from 12:00:00 to 12:00:05 then the output timestamp should be 12:00:05.

* Users of the resampling actor/class should be able to subscribe to the components data they want to receive.

### Questions

* Should we read directly from `MicrogridApiClient` or should we create a new `MicrogridDataReadingActor` that just passes the data untouched?

  **Answer:** For now we use the `MicrogridApiClient` directly, but if possible (and makes sense) we should also extract this from `FormulaCalculator` and try to have the same subscription interface in the future we might split it out as an independent actor if message passing overhead is not an issue. See next steps, as it is related to the `DataSource` layer, see *Next steps*.

* Should a `NaN` be produced by the resampling class or by the resampling function? The former is more consistent and the latest is more flexible.

  **Answer:** `NaN` will be produced by the resampling class for consistency and simplicity, as we don't foresee any use cases where the resampling function will have to make that decision.

* Should we send one result per component or one result with all the components. The former makes it more complicated for users to compile the data for one sampling window (it will have to wait until the data for all components arrive and look at the timestamp to group them properly) but the later could send a lot of unneeded data if some user only needs the data for a couple of components. In both cases we'll need a timeout (smaller than the resampling rate probably) while receiving to detect if the resampling actor is broken or the messages channel is congested or something.

  If we want the output to have a common interface with the microgrid API we probably don't have other choice than to output individual component data in individual channels, but maybe this is a case for having the resampling actor a special case?

### Next steps

Not to do as part of this issue, but to have a more long term sight when implementing this.

* Infer the data rate of every individual input:
  * Use the input timestamps to estimate the rate.
  * Use dynamic ring buffer sizes depending on the input data rate, so components with higher data rates can get bigger buffers (up to a configurable maximum) and lower data rates can get smaller buffers.
  * Make the constructor take different upsampling and downsampling functions.

    If we know if a particular input needs to be upsampled (the frequency of the input is lower than the resampling frequency) or downsampled (if the frequency of the input is lower than the resampling frequency), we can use different functions for each, which could be more precise.

* Implement the `DataSource` layer proposed in https://github.com/frequenz-io/mos/discussions/520#discussioncomment-3707466 (and adapt the current actors to it). This is still pretty much in the air and related to the common interface with the microgrid API.

* Provide some helper function to create channels for components in batch. For example create all the channels for a component category by passing a component graph (or the `MicrogridApiClient` directly?). Otherwise it will be a bit overkill to do all the channel connections manually.

### Resources

* [Presentation on the data pipeline](https://docs.google.com/presentation/d/1LGivWQADWnxYk1NIcDIqpF40h0v1jYF6iKsuMUcnYM4/edit?pli=1#slide=id.gfedb8542fc_0_0)
  
  [image]

  [image]

leandro-lucarella-frequenz, 2022-10-12:
We continued specifying the resampling actor with Ela and Tiyash and I think it is ready for a final review for everybody. Please try not to get too much into bikeshedding so we can proceed with this if it is fairly OK :D 

@thomas-nicolai-frequenz @sahas-subramanian-frequenz @matthias-wende-frequenz @jakub-toptal

jakub-toptal, 2022-10-13:
Question regarding these:

> The resampling function signature should be: `resampling(samples: Sequence[(datetime, float)], resampling_period_s: float)`

```
@dataclass
class Sample:
	timestamp: datetime
	value: float

ResamplingFunction = Callable[(samples: Sequence[Sample], resampling_period_s: float), float]
```

Why do we assume the `value` will be of type `float`? I think the type of `value` will depend what metric you want to resample, it could be `float`, `EVChargerCableState`, `Tuple[float, float, float]` or the whole data point e.g. `MeterData`.

Also, I see you wanted to double check if we are really going to receive `Sample`s from the `MicrogridApiClient` here:

```
class MicrogridDataResamplingActor:
    ...
    # TODO: We don't really receive samples from the MicrogridApiClient probably, check
    _input_receivers: Dict[ComponentMetricKey, Receiver[Sample]]
```

I don't think so, because `MicrogridApiClient` returns objects of type `ComponentData`, e.g. `BatteryData`, `MeterData`, e.g.:
https://github.com/frequenz-floss/frequenz-sdk-python/blob/v0.x.x/src/frequenz/sdk/microgrid/client.py#L74

So imho the interface for the resampling function should rather be something like this:

`resampling(samples: Sequence[(Timestamp, ComponentData)], resampling_period_s: float) -> TimeSeriersEntry[ComponentData]`
or 
`resampling(samples: Sequence[(Timestamp, ComponentData)], resampling_period_s: float) -> TimeSeriersEntry[Any]`

thomas-nicolai-frequenz, 2022-10-13:
> resampling(data_points: Sequence[(Timestamp, ComponentData)], resampling_period_s: float) -> TimeSeriersEntry[ComponentData]

What is ComponentData? This reads to me as we still aim to pass on the entire component object which includes all metrics rather than just the metric of interest as discussed.

thomas-nicolai-frequenz, 2022-10-13:
> I don't think so, because MicrogridApiClient returns objects of type ComponentData, e.g. BatteryData, MeterData, e.g.:

And from that we have to extract the metric of interest for that channel we are going to write it to. This is what the data mapper will be about!

[image]

thomas-nicolai-frequenz, 2022-10-13:
@jakub-toptal now I added that only a metric is passed on from the common object layer to the resampler. The metric is derived from the channel name through the channel registry.

[image]

jakub-toptal, 2022-10-13:
> What is ComponentData? This reads to me as we still aim to pass on the entire component object which includes all metrics rather than just the metric of interest as discussed.

Yes, what I had in mind by saying `ComponentData` was an entire component object. 

Alright, so this is wrong:
`resampling(data_points: Sequence[(Timestamp, ComponentData)], resampling_period_s: float) -> TimeSeriersEntry[ComponentData]`

if we want to pass specific metric to the resampler, but I still think the type of the underlying metric could be something else than `float`. Simply, by looking at what's in the protobuf, there are values of different types than `float`.

----
Exactly, I was wondering what is supposed to happen in the Object Mapping Layer, thanks for pointing that out! 

I thought the resampling actor would still be getting the component data directly from the `MicrogridApiClient`, because it has this param in the constructor `source: MicrogridApiClient` and also it is mentioned as a comment in the `subscribe` method (`the stream will receive all the metrics for a component_id`):

```
class MicrogridDataResamplingActor:
    ....

    def __init__(self, 
            source: MicrogridApiClient,  <--- this here
            *,
            period_s: float = 0.2,
            max_data_age_in_periods: float = 3,
            function: ResamplingFunction = average,
            ):
        # create timer
        self._resampler = MetricsGroupResampler(period_s, function,
                metrics_buffer_size, max_data_age_in_periods)

    def subscribe(self, key: ComponentMetricKey) -> Receiver[Sample]:
        # if metric not in _output_channels:
        #    * create a new receiver from MicrogridApiClient for the metric and add it to
        #      _input_receivers.
        #      We will need to keep an index of component_id because metrics for a single
        #      component come packed in one single object in the MicrogridApiClient (and the
        #      gRPC interface), the stream will receive all the metrics for a component_id.
        #    * create a new _output_channels for the metric
        # return a new receiver from _output_channels[metric]
```

The follow up question is, why do we want to extract the metrics of interest in the Object Mapping Layer instead of doing that in the `ResamplingActor`?
- In case there was a user interested in a specific metric only, but not resampled?
- So that we store less data in the ring buffer and we resample only the metrics of interest instead of entire component data objects?

leandro-lucarella-frequenz, 2022-10-13:
> Why do we assume the `value` will be of type `float`? I think the type of `value` will depend what metric you want to resample, it could be `float`, `EVChargerCableState`,

Because this is what **always** comes from the microgrid API.

>  `Tuple[float, float, float]` or the whole data point e.g. `MeterData`.

No, we said we will resample each metric individually, no tuples.

> Also, I see you wanted to double check if we are really going to receive `Sample`s from the `MicrogridApiClient` here
> ...
> 
> I don't think so, because MicrogridApiClient returns objects of type ComponentData

As things are now, you are write, we would need a layer to extract individual metrics from a whole `ComponentData` package. But we talked about this with @tiyash-basu-frequenz and we decided to extend the microgrid API to be able to subscribe to individual metrics.

No sure when this will be ready and if we should wait for it or going for individual metric extraction from the `ComponentData`. @tiyash-basu-frequenz ?

> So imho the interface for the resampling function should rather be something like this:
> 
> `resampling(data_points: Sequence[(Timestamp, ComponentData)], resampling_period_s: float) -> TimeSeriersEntry[ComponentData]` or `resampling(data_points: Sequence[(Timestamp, ComponentData)], resampling_period_s: float) -> TimeSeriersEntry[Any]`

No, we are not going to resample all metrics from a component, only the metrics that are actually requested.

leandro-lucarella-frequenz, 2022-10-13:
> if we want to pass specific metric to the resampler, but I still think the type of the underlying metric could be something else than `float`. Simply, by looking at what's in the protobuf, there are values of different types than `float`.

The protobuf offers a "hierarchical" view on the components data, but each leave is a float value (or should be, if you find any that it isn't, please point them out!). But it is true that we need to take into account the "tree" when using metric names (`ComponentMetricKey.metric_id` in terms of code). @tiyash-basu-frequenz also mention that somewhere else even numeric values are being used to identify particular metrics, so maybe we can use that. For now, for this first version, I would just use strings for simplicity. So for example: inverter5.dc.voltage could be `ComponentMetricKey(component_id=5, metric_id='dc.voltage')`.

leandro-lucarella-frequenz, 2022-10-13:
> I thought the resampling actor would still be getting the component data directly from the MicrogridApiClient, because it has this param in the constructor source: MicrogridApiClient and also it is [...]

I was leaving the "common object mapping layer" out of this issue to reduce the scope and have a first version that just deals with the microgrid API (which is the most pressing, as we don't have yet other sources in the SDK). FYI @thomas-nicolai-frequenz

I think we should have a first version that gets the stuff directly from the microgrid API. Although if we don't get the new gRPC function in the microgrid API to get individual metrics, then we will probably need this layer anyway to be able to deconstruct the `ComponentData` into individual metrics as you pointed out.

> The follow up question is, why do we want to extract the metrics of interest in the Object Mapping Layer instead of doing that in the `ResamplingActor`?

As I said, for the first version I think is fine to do it in the resampling actor if we are only handling microgrid API data. But in following versions we want this layer to be a common interface for other data sources, like the parquet files or the cloud.

It is basically for modularization, we want the resampler to do just resampling. A data adaptor to only adapt data from one format to another, etc. Otherwise the resampler will get into a big fat mess in the future.

thomas-nicolai-frequenz, 2022-10-13:
> I was leaving the "common object mapping layer" out of this issue to reduce the scope and have a first version that just deals with the microgrid API

NO!

leandro-lucarella-frequenz, 2022-10-13:
But it is transparent to the user, we can add it later, it doesn't affect the design, it's just implementing the same design in steps.

thomas-nicolai-frequenz, 2022-10-13:
> I think we should have a first version that gets the stuff directly from the microgrid API.

I absolutely disagree.

jakub-toptal, 2022-10-13:
> The protobuf offers a "hierarchical" view on the components data, but each leave is a float value (or should be, if you find any that it isn't, please point them out!).

Just checked and it's true for the `data`. The only thing that is non-float is `State`, e.g. for `EVCharger` we already have this in the SDK:
https://github.com/frequenz-floss/frequenz-sdk-python/blob/v0.x.x/src/frequenz/sdk/microgrid/component_data.py#L239
https://github.com/frequenz-floss/frequenz-api-microgrid/blob/v0.x.x/proto/frequenz/api/microgrid/ev_charger.proto#L150

but I am not sure if `state` is relevant in this context. If it is not, then I agree that the underlying type should be `float` and it will also greatly simplify things.

thomas-nicolai-frequenz, 2022-10-13:
> The only thing that is non-float is State, e.g. for EVCharger we already have this in the SDK:

But state will never run through the resampler!

jakub-toptal, 2022-10-13:
> This is what the "data mapper" will have to do through a getter method! Not sure if data mapper is the right name for now to be honest. But the idea is that it extracts the metric of interest from the data object and just passes that on to the resampler.

So how is this Object Mapping Layer supposed to work?

It sits between `MicrogridApiClient` and `ResamplingActor` and it retrieves raw component data (the format is the entire protobuf message) from the `MicrogridApiClient` and then it extracts individual metrics from the entire protobuf messages and sends them downstream to the `ResamplingActor` (the format is a specific metric, e.g. `component_id=1, metric_id=active_power, value=55.0`).

The question is: _For which components should the Object Mapping Layer retrieve the raw component data?_ 

For all of the components in the `Microgrid`? Or perhaps only for those that the `ResamplingActor` is interested in?

If it is the former, then there will be no laziness in retrieving data for all the components.

If it is the latter, then the user of `ResamplingActor` (by user here I mean `BatteryPool` or `LogicalMeter`) needs to `subscribe` for samples of data (e.g. `Meter(id=1, metric_id=active_power)` at some `resampling_frequency`) from the `ResamplingActor` and the `ResamplingActor` needs to `subscribe` for non-resampled data for that `component_id` & `metric_id` from the `ObjectMappingLayer` and the `ObjectMappingLayer` needs to get the data either from the `Microgrid` or parquet.

The way I see it here is that we would need either:

1. An actor in the Object Mapping Layer, e.g. `DataProviderActor` or whatever fancy name we can come up with that would read data from `MicrogridApiClient` (entire protobuf messages) and expose it on per-component & per-metric level, e.g. `Meter(id=5, metric_id=active_power)`. It would also read data from parquet files.

2. Or we would need additional methods in the `MicrogridApiClient`, e.g. instead of having only `meter_data() -> Receiver[MeterData]`, we would need `meter_active_power() -> Receiver[(Timestamp, float)]`. With this approach the `MicrogridApiClient` would become the Object Mapping Layer. But this kind of doesn't make sense, because it should be usable for parquet data too and adding logic to read from parquet files to `MicrogridApiClient` doesn't seem right.

So I guess a separate _thing_ in the Object Mapping Layer for reading the whole protobuf messages either from microgrid or parquet and sending them downstream in `(component_id, metric_id)` format makes sense. What do you think?

leandro-lucarella-frequenz, 2022-10-14:
> The question is: _For which components should the Object Mapping Layer retrieve the raw component data?_

For whatever is requested, it should be on demand. People subscribe to it like it was in `MicrogridData`.

> If it is the latter, then the user of ResamplingActor (by user here I mean BatteryPool or LogicalMeter) needs to subscribe for samples of data (e.g. Meter(id=1, metric_id=active_power) at some [...]

Yes, like this, but no "at some resampling_frequency". The Resampling actor uses only one unique resampling frequency. The BatteryPool can't decide on that. Also no `Meter(...)`, **the resampling actor doesn't know anything about component categories**.

> 1. An actor in the Object Mapping Layer, e.g. `DataProviderActor` or whatever fancy name we can come up with that would read data from `MicrogridApiClient` (entire protobuf messages) and expose it [...]

This one for now. So the `DataSourcingActor` (is the current name) will basically extract individual metrics from the `ComponentData` and send them to a channel. It will also handle the start_time when requesting historical data.

> 2. Or we would need additional methods in the `MicrogridApiClient`, e.g. instead of having only `meter_data() -> Receiver[MeterData]`, we would need `meter_active_power() -> Receiver[(Timestamp, [...]

We discussed this with @tiyash-basu-frequenz and it might happen in the future, but for now we are going with what we have.

leandro-lucarella-frequenz, 2022-10-14:
So there are some more changes coming in, but they are basically on the "data ingestion" layer. So we will not use the `MicrogridApiClient` in the first version but we'll assume the `DataSourcingActor` is ready and use its interface instead. I think this still needs some definition, so for now and for simplicity, pass receivers for metrics to resample in the constructor instead of the `MicrogridApiClient`. I will update the issue description soon and let you know when it is updated.

In the mean time you can start working on all the other classes.

Another heads-up: I find the nomenclature for Metric/Sample/ComponentMetricKey to be confusing, as it is used differently in different contexts (mainly the metric) so there might be some name updates too, but it should be just a search and replace change.

leandro-lucarella-frequenz, 2022-10-14:
I updated the code, this probably needs a bit more thought depending on how the `DataSourcingActor` will work, but if we just take channels for now, we can worry about that later, and it will be very easy to test.

leandro-lucarella-frequenz, 2022-10-17:
After the discussion about terms we had, and looking at the current state of the design, I think we can completely abstract the resampler to be independent from component data, so it could be used with other sources, like weather data. But this means the connection between the resampling actor and the data sourcing actor should be made externally, so I'm not sure we want to go this route.

I was going to update the issue description with this change but I feel it is too soon, so I'm copying the updated description here as a comment, so we can replace it later if we want:

-----

### Classes

#### Utils

```py
# probably in `frequenz.sdk.resampling`, but maybe Sample can go to a more general place

TimeseriesId = str

# Downstream consumers can have a class like this (we can already provide it)
@dataclass
class ComponentMetricId:
    metric_id: str  # Could be an enum or some ID
    component_id: int
    def __str__(self):
        return f"{self.component_id}_{self.metric_id}"

# this should probably now be done by the ChannelRegistry
def channel_name(prefix: str, metric: metric):
    return f'{prefix}_{metric.component_id}_{metric.metric_id}'

@dataclass
class Sample:
	timestamp: datetime
	value: float

ResamplingFunction = Callable[(samples: Sequence[Sample], resampling_period_s: float), float]
```

```py
# probably in `frequenz.sdk.core`

T = TypeVar('T')

# Look if python already offers some ring buffer implementation
class RingBuffer(Generic[T]):

    def __init__(self, size: int):
        ...

    def add(self, item: T) -> None:
        ...

    @property
    def items(self) -> Sequence[T]:
        ...

    # Future
    def resize(self, new_size: int) -> None:
        ...
```

#### `IndividualTimeseriesResampler`

```py
# probably in `frequenz.sdk.resampling`
class IndividualTimeseriesResampler:
    # Future:
    # * Calculate input characteristics: rate, volatily, min, max, etc.
    # * Dynamically decide on:
    #   * Buffer size
    #   * Resampling function to use
    _buffer: RingBuffer[Sample]
    _period_s: float
    _max_data_age_in_periods: float
    _function: ResamplingFunction

    def __init__(self, *,
            period_s: float,
            max_data_age_in_periods: float,
            function: ResamplingFunction,
            ):
        # * calculate buffer size based on the resampling_period_s and
        #   max_data_age_in_periods
        ...

    def add_sample(self, sample: Sample) -> None:
        ...

    def resample(self) -> float:
        ...
```

#### `TimeseriesResampler`

```py
# probably in `frequenz.sdk.resampling`
class TimeseriesResampler:
    _resamplers: Dict[TimeseriesId, IndividualTimeseriesResampler]

    def __init__(self, *,
            period_s: float,
            max_data_age_in_periods: float,
            function: ResamplingFunction,
            ):
        # create the resamplers dict

    def add_metric(self, timeseries_id: TimeseriesId) -> None:
        # create a metric resampler if it doesn't already exist and add it to self.metrics

    def remove_metric(self, timeseries_id: TimeseriesId) -> None:
        # remove a metric from self.resamplers

    def add_sample(self, timeseries_id: TimeseriesId, sample: Sample) -> None:
        # forward to IndividualTimeseriesResampler.add_sample()

    def resample(self)-> dict[str, Sample]: # channel_id, sample
        timestamp = datetime.now()
        # loop over all metrics and call resample() and attach the timestamp to
        # create a new sample for each returned value
```

#### `TimeseriesResamplingActor`

**NOTE:** How we create and pass input and output channels still needs to be decide and might change if we can do dynamic subscriptions.

```py
# probably in `frequenz.sdk.resampling` with an alias in `frequenz.sdk.actor`
class TimeseriesResamplingActor:
    _resampler: TimeseriesResampler
    _sampling_timer: Timer
    _input_receivers: Dict[TimeseriesId, Receiver[Sample]]
    _output_channels: Dict[TimeseriesId, Channel[Sample]]

    def __init__(self, input_receivers: Dict[TimeseriesId, Receiver[Sample]], *,
            start_time: Optional[datetime] = None,  # None means NOW, so only live data
            period_s: float = 0.2,
            max_data_age_in_periods: float = 3,
            function: ResamplingFunction = average,
            ):
        # create timer
        self._resampler = TimeseriesResampler(period_s, function,
                metrics_buffer_size, max_data_age_in_periods)

    def subscribe(self, timeseries_id: TimeseriesId) -> Receiver[Sample]:
        # if metric not in _output_channels:
        #    * (NOT NOW, in the future) create a new receiver from DataSourcingActor for the metric and add it to
        #      _input_receivers.
        #      We will need to keep an index of component_id because metrics for a single
        #      component come packed in one single object in the MicrogridApiClient (and the
        #      gRPC interface), the stream will receive all the metrics for a component_id.
        #    * create a new _output_channels for the metric
        # return a new receiver from _output_channels[metric]

    # Future
    def unsubscribe(self, subscription: ?) -> None:
        # TODO: Can this be done?

    def run(self):
        # for event in selec(timer, metric_receivers):
        # 	if event == timer:
        # 		call self.resampler.resample() and put the output into the output channels
        #   elif event == metric_receiver:
        # 		call self.resampler.add_sample()
```

#### Example usage

This assumes actors don't start when instantiated, but maybe it is possible to do this too even if they are already running.

```py
async def app_run() -> None:

    resampling_actor = ...(input_receivers, ...)

    battery_pool_actor = ...()
    for metric in battery_pool_actor.metrics:
        battery_pool_actor.attach(metric, resampling_actor.subscribe(metric))

    await actors_run(resampling_actor, battery_pool_actor)
```

jakub-toptal, 2022-10-17:
Aren't `TimeSeriesId` and `ComponentMetricId` supposed to be the same thing?

```
# probably in `frequenz.sdk.resampling`, but maybe Sample can go to a more general place

TimeseriesId = str

# Downstream consumers can have a class like this (we can already provide it)
@dataclass
class ComponentMetricId:
    metric_id: str  # Could be an enum or some ID
    component_id: int
    def __str__(self):
        return f"{self.component_id}_{self.metric_id}"
```

thomas-nicolai-frequenz, 2022-10-17:
> I think we can completely abstract the resampler to be independent from component data

YES!

jakub-toptal, 2022-10-17:
I think that a single `ResamplingFunction` here will not do the job, because we have different types of components. 

Unless the `ResamplingFunction` will run different code based on the component type (which is not present in the `TimeSeriesId` at the moment, I mentioned that here https://github.com/frequenz-floss/frequenz-sdk-python/pull/32#issuecomment-1280591465).

```
# probably in `frequenz.sdk.resampling`
class TimeseriesResampler:
    _resamplers: Dict[TimeseriesId, IndividualTimeseriesResampler]

    def __init__(self, *,
            period_s: float,
            max_data_age_in_periods: float,
            function: ResamplingFunction, # <-- single function for all resamplers
            ):
        # create the resamplers dict
```

So it could be something like this:
```
@dataclass(frozen=True)
class TimeSeriesId:
    """ID for a single metric of a specific component, e.g. battery_123_active_power."""

    component_id: int
    metric_id: str
    component_category: ComponentCategory

    def __str__(self) -> str:
        """Return string representation of a time series id.
        Returns:
            string representation of a time series id
        """
        return f"{self.component_category}_{self.component_id}_{self.metric_id}"

ResamplingFunction = Callable[(samples: Sequence[Sample], resampling_period_s: float), float]

def resample(samples: Sequence[Sample], resampling_period_s: float): -> float:
    if samples[0].component_category == BATTERY:
         # resample battery samples, etc.
    ...
```

or

```
# probably in `frequenz.sdk.resampling`
class TimeseriesResampler:
    _resamplers: Dict[TimeseriesId, IndividualTimeseriesResampler]

    def __init__(self, *,
            period_s: float,
            max_data_age_in_periods: float,
            ):
        # create the resamplers dict

    def add_timeseries(self, timeseries_id: TimeseriesId, resampling_function: ResamplingFunction) -> None:
        self._resamplers[timeseries_id] = IndividualTimeseriesResampler(resampling_function=resampling_function, ...)
```

thomas-nicolai-frequenz, 2022-10-17:
> I think that a single ResamplingFunction here will not do the job, because we have different types of components.

It will. Right now we only have one Resampling function and in the future this is either (1) derived from the metric or (2) the characteristics of the values. This has not been yet decided. For now the Resampler should simply take a timestamp and a float value as input. @leandro-lucarella-frequenz please correct me if you had something else in mind.  

A resampling function is tied to the metric and a metric can show different characteristics. Using the metric as an indicator for the choice of the resampling function we however rejected. Thats why in the future we will build something that captures the characteristics of samples coming in through the stream. That will mean that in the beginning we will use the fallback resampling function and once enough samples have come through switch to an auto-detection to choose the correct function.

leandro-lucarella-frequenz, 2022-10-17:
> Aren't `TimeSeriesId` and `ComponentMetricId` supposed to be the same thing?

Yes, `ComponentMetricId` is just a convenient way to construct a `TimeseriesId` to get a timeseries for a component's metric.

leandro-lucarella-frequenz, 2022-10-17:
> It will. Right now we only have one Resampling function and in the future this is either (1) derived from the metric/measurement or (2) the characteristics of the streamed data. For now the Resampler should simply take a timestamp and a float value as input. @leandro-lucarella-frequenz please correct me if you had something else in mind.

No, that's correct and my understanding too. This is specified in the **Next steps**:

> * Infer the data rate of every individual input:
>   * ...
>   * The input data rate could be used also to automatically pick a good resampling function (we can know, for example, if we are up or downsampling).
> ...
> * Automatically chose a good resampling function based on metrics characteristics, like volatility (or input data rate, when available).

leandro-lucarella-frequenz, 2022-10-17:
> component_category: ComponentCategory

No, no, we want to avoid to tie the resampling to component categories (and to components entirely).

thomas-nicolai-frequenz, 2022-10-17:
> Yes, ComponentMetricId is just a convenient way to construct a TimeseriesId to get a timeseries for a component's metric.

@jakub-toptal in database speak `TimeseriesId` the name of the unique key build out of the  `UNIQUE(componentId,metricId)`

jakub-toptal, 2022-10-17:
@thomas-nicolai-frequenz 

I believe you are referring to the resampling function interface here:
```
To summarise:

INPUT: TUPLE(timestamp, float)
OUTPUT: TUPLE(timestamp, float)
````

For input, shouldn't it be `LIST[TUPLE(timestamp, float)]` ? E.g. `resample(input: List[Sample]) -> Sample`, where `Sample => TUPLE(timestamp, float)`

leandro-lucarella-frequenz, 2022-10-17:
> Unless the ResamplingFunction will run different code based on the component type (which is not present in the TimeSeriesId at the moment, I mentioned that here).

Yes, so this is the part where component type (category) gets mixed up with resampling. So this is a path we don't want to take.

thomas-nicolai-frequenz, 2022-10-17:
> I believe you are referring to the resampling function interface here:

no I do talk about the Resampling actor itself.

> For input, shouldn't it be LIST[TUPLE(timestamp, float)] ?

By default its a stream thats is forwarding individual samples not lists!

jakub-toptal, 2022-10-18:
```
# probably in `frequenz.sdk.resampling`
class TimeseriesResampler:

    ....

    def resample(self)-> dict[str, Sample]: # channel_id, sample
        timestamp = datetime.now()
        # loop over all metrics and call resample() and attach the timestamp to
        # create a new sample for each returned value
```

Thinking that setting the `timestamp` like this will not work when reading parquet data, because samples will belong to resampling windows in the past.

This method will be called from the resampling actor like this:
```
async def run(self) -> None:
    """Run the actor."""
    while True:
        select = Select(
            resampling_timer=self._resampling_timer,
            ...
        )
        while await select.ready():
            if _ := select.resampling_timer:
                samples: Dict[TimeSeriesId, Optional[Sample]] = self._resampler.resample()
            ...
```
so we can change the interface to `def resample(self, timestamp: datetime)-> dict[str, Sample]:` instead, but then the `ResamplingActor` will have to keep track of the "current window in the past" and provide an appropriate `timestamp` to the `resample` method.

thomas-nicolai-frequenz, 2022-10-18:
> Thinking that setting the timestamp like this will not work when reading parquet data, because samples will belong to resampling windows in the past.

agreed but please ignore parquet use case for now.

leandro-lucarella-frequenz, 2022-10-23:
I think we might have other fundamental differences when reading historical data that might even force us to have a different resampling strategy for it (for example, the timer doesn't make sense either when resampling historical data, nor the max age of a sample to be considered relevant). For this issue, I would just focus on live data and we'll see how to deal with historical data in a next issue/PR.

thomas-nicolai-frequenz, 2022-10-23:
> for example, the timer doesn't make sense either when resampling historical data, nor the max age of a sample to be considered relevant

yes lets deal with that in the next issue/PR but we could by default deactivate the timer as long as timestamps are older than `NOW`. I feel there will be simple solutions to the problem. Just to keep that in mind and not to worry that much about it right now.

closed 2022-11-01 by jakub-toptal