frequenz-floss/frequenz-sdk-python #19: Incorporate `typing_extensions` into type checkers

jakub-toptal, 2022-10-06:
### What's needed?

Type annotations in some cases are limited and thus it would be good to enhance them with what the `typing-extensions` package has to offer.

One example of this would be `@classmethod`, which always uses `cls` argument as the first argument. Ideally, we would annotate it with the `Self` type from the `typing-extensions` package.

### Proposed solution

Use `typing-extensions` package to enhance type annotations.

### Use cases

This can be applied to `@classmethod`s and perhaps other functions used as decorators. One apparent case where we could use better type checking is the `@actor` decorator, which decorates classes at runtime making it impossible for `pylint` to verify if attributes / methods are defined on the decorated class.

### Alternatives and workarounds

In case of `@classmethod`, we an omit the type annotation for the `cls` argument and it still passes type checking. 

Otherwise, we have to explicitly tell type checkers `pylint` and/or `mypy` that a particular line should be ignored, because there is no correct type annotation available.

leandro-lucarella-frequenz, 2022-10-07:
Some other important clarifications:

* `Self` is introduced in Python 3.11.
* `typing-extensions` is not a random package, is an official Python package maintained by the Python organization and only includes features backported from newer Python versions so they are available for older versions. Thus the use of this package is by definition temporary (until we can depend on a new enough Python minimal version)
* `mypy` doesn't support `Self` yet, this is why we didn't go for it. There is a PR, so support should come fairly soon. https://github.com/python/mypy/pull/11666
* This comes from another PR's discussion: https://github.com/frequenz-floss/frequenz-sdk-python/pull/4#discussion_r986722004

leandro-lucarella-frequenz, 2023-04-21:
We already have `typing_extensions` as a dependency and mypy already supports `Self`, we just need to start using it.

closed 2023-04-21 by leandro-lucarella-frequenz