frequenz-floss/frequenz-sdk-python #24: Creating output channels dynamically for the resampling actor

sahas-subramanian-frequenz, 2022-10-10:
### What's needed?

The `Resampling` actor needs to send out dynamically created channel receivers (corresponding to dynamic component IDs) to use-case actors.  We don't have a way to do this currently.

### Proposed solution

The channel registry that was proposed a few months back would be a good solution to this problem.  It would work as follows:

We need a `ChannelRegistry` class that would create and own channels and provide channel senders and receivers to all actors that ask for them.  It would have `get_sender(chan_name)` and `get_receiver(chan_name)` methods that provide access to an existing channel,  or implicitly create one if a channel with the given name doesn't exist.

In single process contexts,  `ChannelRegistry` would hold a dictionary of channels,  and in multi-process contexts,  it would use an external service like redis or nats,  whichever we use for implementing multi-process channels.

In each process,  the `ChannelRegistry` instance can be accessed through the top level `MicrogridApi` module.

The `Resampling` actor would accept a channel receiver in its constructor,  through which external actors can send requests to it.  Requests would contain a component ID and a metric name.  The resampling actor and the external actor would construct a channel name based on the requested id and metric name,  and go to the `ChannelRegistry` to ask for a sender or receiver respectively,  and thus establish communication using `ChannelRegistry` as a broker.

If the resampling actor is already handling a newly requested metric,  it would discard the request.

### Use cases

This can be used in all cases where actors need to dynamically create channels.

thomas-nicolai-frequenz, 2022-10-10:
> In each process, the ChannelRegistry instance can be accessed through the top level MicrogridApi module.

Can you please explain why MicrogridApi module? Is that the MicrogridApiClient or what is that and what has the ChannelRegistry to do with the MicrogridApi? Keep in mind this might also be used in the Cloud where there is no Microgrid API after all.

thomas-nicolai-frequenz, 2022-10-10:
> ChannelRegistry would hold a dictionary of channels, and in multi-process contexts, it would use an external service like redis or nats

Lets ignore this for now but in the multi-channel environment you'd want this probably to be part of some kind of distributed queue mechanism.

sahas-subramanian-frequenz, 2022-10-10:
> Can you please explain why MicrogridApi module?

`MicrogridApi` is a simple class that maintains per-process singleton objects.  For example,  the `MicrogridApiClient` is actually created and owned by `MicrogridApi`,  and from it, any actor can get a reference to `MicrogridApiClient` whenever they want.  But putting `ChannelRegistry` inside `MicrogridApi`,  we provide a single ready-to-use instance of `ChannelRegistry`, so that each actor doesn't create its own.

We'll probably have to rename it to something else before we start using it in the cloud.

thomas-nicolai-frequenz, 2022-10-10:
> Or maybe we could make a separate singleton instance for ChannelRegistry.

That sounds like a really good plan ;-) Lets do it now @leandro-lucarella-frequenz

ela-kotulska-frequenz, 2022-10-11:
1. So `ChannelRegistry` would create and store channels for each requested pair:  `component_id_&_metric`, yes? It won't be an actor. 

> The Resampling actor would accept a channel receiver in its constructor, through which external actors can send requests to it. Requests would contain a component ID and a metric name.

2. User will subscribe for metric by sending requests using channel to ResamplingActor. ResamplingActor will not respond for that request. Then user should get receiver stream from the ChannelRegistry?
Then ResamplingActor receives requests `component_id_&_metric` and gets sender from ChannelRegistry.

leandro-lucarella-frequenz, 2022-10-11:
> `MicrogridApi` is a simple class that maintains per-process singleton objects

This came up several times when trying to explain something about the current code. I think it is time to rename this class, and probably split too. I'm not sure why we wanted to have a single object to store all "singletons", I don't see anything against having a global `ChannelRegistry` in its own module and that's it, it has nothing to do with the microgrid.

leandro-lucarella-frequenz, 2022-10-11:
> Requests would contain a component ID and a metric name.

I wouldn't tie this registry to the Resampling! I would just take a channel name, the registry doesn't care about how the name was constructed.

leandro-lucarella-frequenz, 2022-10-11:
I don't want to be **that guy** (and maybe having the registry is a good idea anyway) but I think part of the problem comes from the fact that actors are started as soon as they are instantiated. If this was not the case, then the *registry* can be local and stored in the actors, which own their channels.

In the particular case of the resampler, we can have (extract from #5):

```py
class MicrogridDataResamplingActor:
    ...
    _input_receivers: Dict[Metric, Receiver[Sample]]
    _output_channels: Dict[Metric, Channel[Sample]]

    ...

    def subscribe(self, metric: Metric) -> Receiver[Sample]:
        # if metric not in _output_channels:
        #    * create a new receiver from MicrogridApiClient for the metric and add it to
        #      _input_receivers.
        #      We will need to keep an index of component_id because metrics for a single
        #      component come packed in one single object in the MicrogridApiClient (and the
        #      gRPC interface), the stream will receive all the metrics for a component_id.
        #    * create a new _output_channels for the metric
        # return a new receiver from _output_channels[metric]

    ....

async def app_run() -> None:

    resampling_actor = ...(MicrogridApiClient, ...)

    battery_pool_actor = ...()
    for metric in battery_pool_actor.metrics:
        battery_pool_actor.attach(metric, resampling_actor.subscribe(metric))

    await actors_run(resampling_actor, battery_pool_actor)
```

The only issue with this is subscription have to be done at app initialization time. If we want to do subscriptions once actors are running, then yes, the registry would be one option, or another option would be to be able to pass a channel through a channel, but maybe that's more complicated (or will need a registry too).

I wonder if the channel registry shouldn't be done in the channels repo. If we do so, we certainly can implement passing channels through channels, we can just pass a special value in the channel, like "channel:<name>" and then the receiving end looks if it is a channel, then it looks for `<name>` in the global registry and then returns the channel to the user transparently, without knowing there is a registry at all.

I think it's good to have channel ownership instead of having all channels global belonging to nobody, because also otherwise is hard to know when a channel can be destroyed.

**Disclaimer:** I'm not saying we should go for the full blown solution at first. I'm all for going in small steps, so I'm fine if we just implement a channel registry at first. I'm even fine (and might even rather prefer) to just start with the app-initialization-time approach, so not going for the channel registry just yet (I would prioritize decoupling actor instantiation and start for example).

thomas-nicolai-frequenz, 2022-10-11:
> If this was not the case, then the registry can be local and stored in the actors, which own their channels.

Local? The Resampling actor needs to know all channel names from all subscribers (e.g. actors). So why wouldn't the ChannelRegistry being part of the channel implementation itself?

leandro-lucarella-frequenz, 2022-10-11:
I don't get the question (in the context of the quoted text).

To expand on the quoted text and the "Local?" question, what I mean is the Resampling actor can **own** his own channels instead of delegating that to the channel registry, this is what I mean by "local". As I show in the code there, the `_output_channels` attribute is in a way a "local channels registry".

> So why wouldn't the ChannelRegistry being part of the channel implementation itself?

I don't know what you are referring to, what is "channel implementation" here? The `frequenz-channels-python` repo or something else?

I'm suggesting 2 separate things. One is we don't need a registry right now if we limit the "subscription" to be only done at app-initialization-time (this would be just to have an initial version, in the future I think we'll probably need it one way or another). The other is we might not need a global registry (exposed to the user) if we instead create a registry in the `frequenz-channels-python` and use it internally in that repo only to implement passing channels through channels.

Then users will just do a request to subscribe to a metric like `response = resampler_client_handle.send(Subscribe(component_id, metric_id))` and will get the receiver directly from the response: `receiver = response.receiver` instead of having to go through the registry explicitly (like `channel_name = ChannelRegistry.get_receiver(response.channel_name)`).

I hope this makes things more clear and not more obsure :D

thomas-nicolai-frequenz, 2022-10-11:
> I mean is the Resampling actor can own his own channels instead of delegating that to the channel registry, this is what I mean by "local"

got it! makes sense!

sahas-subramanian-frequenz, 2022-10-11:
We haven't explored supporting methods in actors (i.e calling of methods on a running actor from a different actor),  or passing receivers through receivers.  So I think `ChannelRegistry` is the way to go for now.

leandro-lucarella-frequenz, 2022-10-12:
> We haven't explored supporting methods in actors (i.e calling of methods on a running actor from a different actor), or passing receivers through receivers. So I think `ChannelRegistry` is the way to go for now.

I didn't meant to implement calling methods, only passing channels through channels, but nevermind, I think we'll need a channel registry to implement that too, so we need it no matter what!

PR #25 (2022-10-13): Implement ChannelRegistry
Closes #24
commit 05ce8d5e (2022-10-13): Implement ChannelRegistry (#25)

Closes #24
closed 2022-10-13 by sahas-subramanian-frequenz