frequenz-floss/frequenz-sdk-python #30: Implementa a data sourcing actor

tiyash-basu-frequenz, 2022-10-13:
### What's needed?

We need to implement an actor that can expose data from different sources to the downstream layers.

Component data can come from different sources, e.g., the Microgrid API, the Reporting API, and parquet files in local storage.
The data format in these sources can be different.

There may be mutiple downstream actors, each of them reading from multiple sources.
If they all have mechanisms to convert data into one common format for further consumption, then:
* they would each need to understand how all the different data sources work, causing high coupling, and
* they would very likely run similar conversions, wasting CPU cycles

### Proposed solution

The solution would be to create an actor that
* can read multiple data sources, e.g., the Microgrid API, and parquet files in local storage,
* can expose data in a common format that abstracts the underlying data source away,
* can switch between input data streams seamlessly, based upon well-defined conditions (see below for details).

Note that in the first version (v1), this actor needs to consume data from the following sources only:
* Microgrid API
* parquet files in local storage

The top-level building blocks of the actor could be a set of classes as follows
(Note that these are just pseudocoded ideas to specify requirements. The actual implementation may vary):

```py
# This is the class that uniquely represents a metric channel.
@dataclass
class ChannelKey:
    # An enum or a unique ID representing the metric.
    metric_id: str

    # The component ID
    component_id: int

    # The time from which data should be consumed.
    start_ts: datetime
```
See `ComponentMetricKey` in [issue #5](https://github.com/frequenz-floss/frequenz-sdk-python/issues/5) for more details on `ChannelKey`.

```py
# This is the class that consumes and forwards data for one given channel.
# The actor can have a collection of the objects of this class.
class DataIngestor:
  _channel: Option[BroadcastChannel[float]]

  def __init_(self):
    # Initialize `_channel` as `None`
    self._channel = None
    ...

  def get_channel(self, data_key: ChannelKey) -> BroadcastChannel[float]:
    # This method is key here.
    #
    # Consume data from different sources, and stream them downstream via an
    # output channel. The user has to input the details of which
    # metric has to be consumed and from what time. The user gets a channel with
    # the desired data in return.
    #
    # If `data_key.start_ts` is `now()`, then read directly from the
    # Microgrid API.
    #
    # If `data_key.start_ts` is in the past, e.g., `now() - 30m`, then read the
    # parquet files until there is no more data there. Then read from the
    # microgrid API.
    #
    # If `data_key.start_ts` is in the future, e.g., `now() + 30m`, then wait
    # until the given timestamp is reached. Then read from the microgrid API.

    # Return `self._channel` if it already exists, otherwise create a new one.
    ...
```

The actor itself could be defined as follows:

```py
@dataclass
class DataRequestKey:
  # The details of the metric to read data for, and the timesamo from which the
  # data should be read.
  data_key: ChannelKey

  # Can have more elements as required.

class DataSourcingActor:
  # The idea here is to store previously returned channels, so that we reuse them
  # insted of creating new ones every time. The data structure can be a `dict`, or
  # can be some other custom data structure, depending upon the following
  # requirements:
  #
  # Two `ChannelKey` objects with `start_ts >= now()` are
  # effectively the same, since they consume live data from the Microgrid API.
  #
  # However, two `ChannelKey` objects that need historical data from the
  # same `start_ts` timestamp may not be the same. Assuming the two
  # `ChannelKey` objects are received at times 00:00:00 and 01:00:00,
  # then the 2nd one will still require to start from the the first historical
  # data sample, whereas the 1st one could be several data samples ahead or
  # reading live data by then.
  _data_ingestors: Dict[DataRequestKey, DataIngestor]

  def __init__(self):
    # Initialize the `_data_ingestors` object.
    self._data_ingestors = {}
    ...

  def get_channel(self, data_key: ChannelKey):
    data_req_key = DataRequestKey(data_key)

    # If a `DataIngestor`1 object has not yet been created, then create it.
    if not self._data_ingestors.contains(data_req_key):
      self._data_ingestors[data_req_key] = DataIngestor()

    # Return the channel from the `DataNgestor` instance.
    return self._data_ingestors[data_req_key].get_channel()
```

thomas-nicolai-frequenz, 2022-10-13:
@sahas-subramanian-frequenz can you please add the details with regards to reading from the parquet files and the buffers we talked about so that we get a fair distribution of load across different reading processes and smartly yield.

thomas-nicolai-frequenz, 2022-10-13:
First problem is reading from parquet needs to happen in small chunks if we are reading things in parallel to avoid one reading "process" to be faster and to block all others. So this needs to be non-blocking as possible and avoiding feeding one channel faster than others.

Second issue is if the a buffer is full the reading process from parquet might pause but this is not possible with the Microgrid API. There we have two choices: (1) throw away records (2) push them to disk and consumer later if buffer gets empty again. Point 2 has the problem that for realtime cases like grid fuse protection etc. you are probably not interest in the oldest records from disk but these might be needed for e.g. building the averages for the past 15min. So we would need to make sure that priority does have current data while non priority older data or we need to be able to define in the channel properties that in one channel there is no disk overflow and in others there is. Thoughts?

All problems are also marked with a red exclamation mark in the slide:

[image]

thomas-nicolai-frequenz, 2022-10-13:
> properties that in one channel there is no disk overflow and in others there are

assuming that use-case actors know best which data can be thrown away like the ev actor or circuit breaker while others like pv optimization need it the argument might be to support it in the channel properties. I'm just not sure if this is v1 or v2. Feels more like v2 imho but open to discuss.

thomas-nicolai-frequenz, 2022-10-14:
> I think ordering is cheap. A simple (maybe slightly adjusted) bubble sort on the time stamps should be very efficient given the invariants we have in the buffer, i.e. it's sorted after each operation. My bet is that such an algorithm has an average constant runtime.

Some comment from @matthias-wende-frequenz with regards to sorting incoming samples in the LogicalMeter/*Pool actors if they are being out of order. If there is no problem in a single process environment fine but I don't think we should bet on that.

leandro-lucarella-frequenz, 2022-10-14:
A few questions:

1. > If the start timestamp is in past, then the local parquet files should be read first. Once there is no more data in the parquet files, then data from the live data source (Microgrid API) should be streamed.

   When does reading from the Report API comes into play?

2. Silly bike-shedding: I think we can use `DataSource` instead of `DataIngestor`.

3. > def get_channel(self, data_key: ChannelKey) -> BroadcastChannel[float]:

   Shouldn't this be `get_receiver(...) -> Receiver[float]:` instead?

4. > If `data_key.start_ts` is `now()`, then read directly from the
    # Microgrid API.

   I know is pseudocode but I would say `None` instead of `now()` as we can't possible compare `now() == now()` as they will be always different :D

5. > # If `data_key.start_ts` is in the future, e.g., `now() + 30m`, then wait
   > # until the given timestamp is reached. Then read from the microgrid API.

   Do we really have an use case for this?

6. So, if we need to have multiple channels per (component_id, metric_id, start_time), then we need to add some sort of UUID to the channel name. If that is the case, then we don't have anymore a way to infer a channel name based on the data requested to the data source, so I think we need to have this in mind in general.

7. The `DataSourcingActor` is missing the `run()` method. That method will have to do the conversion of data (in the microgrid API converting a `ComponentData` object into individual float values). But I think this conversion should be done in the `DataSource` class. Even more, I think we should probably have a `MicrogridDataSource`, `LocalStorageDataSource`, etc. classes that are in charge of the particular conversions, so we can easily extend it to add more sources without making the `DataSource` a spaghetti.

8. I think `DataSourcingActor.get_channel()` shouldn't be a method but a message passed via a control channel to the actor (or the method could be preserved to call it at initialization-time). If I didn't understand everything wrong, that was the whole point of the `ChannelRegistry`, to be able to dynamically request for channels?

   So we should probably have a `DataSourcingActor.SubscribeRequestMessage` like:

   ```py
   @dataclass
   class SubscribeRequestMessage:
       metric_id: str
       component_id: int
       start_ts: datetime
   ```

   And a `DataSourcingActor.SubscribeResponseMessage` like:
   ```py
   @dataclass
   SubscribeResponseMessage:
       channel_name: str
   ```

   Then the actor making the request can go to the `ChannelRegistry` and do `.get_receiver(response_message.channel_name)` to get the receiver.

jakub-toptal, 2022-10-14:
Thinking that reusing channels makes sense only for streaming live data. Consider this:
- User 1 (`U1`) requests the following data `category=METER, id=5, start_ts=100` at time `200`
- User 2 (`U2`) requests the following data `category=METER, id=5, start_ts=100` at time `300`

When `U2` requests the same data at time `300`, `U1` might have already consumed something from that channel i.e. `category=METER, id=5` from interval `<100,300>`, so if we have a shared channel for `U1` and `U2`, `U1` will receive duplicated data points from the interval in the past that `U2` needs data from. 

It is a broadcast channel so whoever subscribes for data in the past, they will cause that data to be sent in a fan-out mode to all subscribers of that `DataRequestKey`, who might have consumed some parts of it already.

jakub-toptal, 2022-10-14:
Also, thinking that multiple request for the same `component_id`, but different `metric_id` currently mean separate reads from the parquet file. If there are 4 data requests like so:
- `category=BATTERY, id=5, metric_id=soc, start_ts=100 `
- `category=BATTERY, id=5, metric_id=capacity, start_ts=100 `
- `category=BATTERY, id=5, metric_id=active_power_lower_bound, start_ts=100 `
- `category=BATTERY, id=5, metric_id=active_power_upper_bound, start_ts=100 `

these could be read from the very same parquet file, instead of doing a separate read per `metric_id`.

thomas-nicolai-frequenz, 2022-10-14:
> When does reading from the Report API comes into play?

@leandro-lucarella-frequenz once the data pipeline exists and model training is being done by us in the cloud. timeline as always, yesterday ;-) this is mainly related to training and once we use the SDK for trading purposes on the cloud side.

thomas-nicolai-frequenz, 2022-10-14:
> Also, thinking that multiple requests for the same component_id, but different metric_id currently mean separate reads from the parquet file. 

Thats fine and can happen and might happen. We should not prevent a user to do less optimal things if he wants to. However, the optimization of that will happen on two levels: using the channel broadcasting and the LogicMeter/*Pool actors will take care of it as shared in the sdk slack channel:

- If two channels share the same properties we should use the broadcast feature in the channel implementation; auto-detection as subscription time? 
[@Sahas](https://frequenz-io.slack.com/team/U0267KXBT55)
- We need a buffer for the LogicMeter (e.g. max 5 items ... if buffer runs full the output will be NaN) ... this will also need sorting the buffer when a new sample arrives. Keep in mind two metrics that might need aggregation can arrive through different channels and this arriving at different points in time but carrying the same resampling time bucket timestamp. 
[@luca](https://frequenz-io.slack.com/team/U03LRUFHKN0)
 can you create an issue for this?
- LogicalMeter should support channel subscription optimisation to turn below into just one channel (version 2)
channel_1 = meter_1 -> metric_1
channel_2 = meter_2 -> metric_1

sahas-subramanian-frequenz, 2022-10-14:
@jakub-toptal and I were discussing and noticed that,  because our parquet files are created once every 5 minutes,  when somebody requests for data from 30 minutes ago,  and it is 3 minutes since the last file,  if we finish streaming the existing files in a few seconds and switch to the live stream immediately,  the stream will have a few minutes of missing data.

The easiest solution is that we wait for the next file while caching live data,  so that we can transition smoothly.

I discussed with @tiyash-basu-frequenz and he suggested that we could create a new parquet writer instance that writes files with a higher frequency, and not sync its data to the cloud.  And once we have the reporting API,  this will be the only version of parquet writer that runs on the devices.

thomas-nicolai-frequenz, 2022-10-14:
> I discussed with @tiyash-basu-frequenz and he suggested that we could create a new parquet writer instance that writes files with a higher frequency, and not sync its data to the cloud

Lets go with the other solution we discussed @sahas-subramanian-frequenz where we simply wait to close the max 5-min gap.

sahas-subramanian-frequenz, 2022-10-14:
> Stream from parquet until end of data ( NOW() - X )

Yes,  but just to expand a bit on step 1:
- 1.1 Stream until end of parquet data.
- 1.2 check if we can continue with the Microgrid API buffer without missing data, if so, continue to step 2, discarding any overlapping messages.
- 1.3 if missing data, wait for another parquet file,  stream it and continue to step 2, discarding any overlapping messages.

leandro-lucarella-frequenz, 2022-10-14:
> @leandro-lucarella-frequenz once the data pipeline exists and model training is being done by us in the cloud. timeline as always, yesterday ;-) this is mainly related to training and once we use the SDK for trading purposes on the cloud side.

I mean functionally, how does the data sourcing actor knows when to read from the reporting api?

leandro-lucarella-frequenz, 2022-10-14:
> We need a buffer for the LogicMeter (e.g. max 5 items ... if buffer runs full the output will be NaN) ... this will also need sorting the buffer when a new sample arrives. Keep in mind two metrics [...]

#31

leandro-lucarella-frequenz, 2022-10-18:
@sahas-subramanian-frequenz can you maybe add an example of how do you see the actor being used by someone that needs to consume resampled data? For example, if a `LogicalMeter` would want to subscribe to resampled data for `component_id=1` and `metric_id=active_power` (not sure if that exists, whatever metric that makes sense). Can you sketch an usage example? I think it will be useful to have a clearer picture.

sahas-subramanian-frequenz, 2022-10-18:
Here's how it is coming up:

``` python
@dataclass
class ComponentMetricRequest:
    # a uuid to identify the output channel from the channel registry.
    channel_uuid: str
    # id of the requested component
    component_id: int
    # `metric_name` would be attribute names on BatteryData, etc.  Example
    # value: `soc`, `current_phase_1`, etc.  For individual phase current and
    # voltage values, we need to add additional attributes to `MeterData` and
    # `EVChargerData`.
    metric_name: str
    # the start time from which data is required.
    start_time: Optional[datetime]
```

And the actor can be started with something like:

``` python
data_sourcing_req_chan = Broadcast[ComponentMetricRequest]("data_sourcing_req")
data_sourcing_actor = DataSourcingActor(data_sourcing_req_chan.get_receiver())
resampling_actor = ResamplingActor(
    data_sourcing_req_chan.get_sender(), 
    # receiver for resampling requests
 )
```

We can chat about specifics tomorrow morning if you want.  I should have an implementation and some tests by EOD tomorrow.

thomas-nicolai-frequenz, 2022-10-18:
> metric_name: str

are we sure this to be a string rather than an enum?

leandro-lucarella-frequenz, 2022-10-19:
> We can chat about specifics tomorrow morning if you want.

Alright. I will write some code here with what I had in mind so we can have something more concrete to discuss, but we can discuss in a call to make it more dynamic.

So I guess a potential `LogicalMeter` will have to do something like this:

```py
data_sourcing_req_chan = Broadcast[ComponentMetricRequest]("data_sourcing_req")
data_sourcing_actor = DataSourcingActor(data_sourcing_req_chan.get_receiver())

resampling_req_chan = Broadcast[ComponentMetricRequest]("resampling_req")
resampling_actor = ResamplingActor(
    data_sourcing_req_chan.get_sender(), 
    resampling_req_chan.get_receiver(),
 )

logical_meter = LogicalMeter(resampling_req_chan.get_sender())

# And then logical meter at some point will have to subscribe to some metric, for example:

for (component_id, metric_id) in info_we_want:
    req = ComponentMetricRequest(uuid, component_id, metric_id)
    resp = self.resampling_req_sender.send(req)
    receivers.append(channel_registry.get_receiver(resp.channel_name))
```

(also I guess channels for request will be `Bidirectional`, not `Broadcast`, right?)

I'm also not sure what the uuid in the `ComponentMetricRequest` is for and what the response will be (I'm just assuming it will have at least a channel name that is found on the channel registry from which I can get a receiver).

What it is not entirely convincing for me about this approach is we are coupling the resampling actor to one only data source, and a data source of very particular characteristics (component metrics). I'm not sure if we will have other use cases for the resampling actor, but it seems to me like it would be a good idea to break that coupling. What I had in mind might be a bit more annoying to use (but we can encapsulate it in some utility function), but is decoupled:

```py
data_sourcing_req_chan = Broadcast[ComponentMetricRequest]("data_sourcing_req")
data_sourcing_actor = DataSourcingActor(data_sourcing_req_chan.get_receiver())

resampling_req_chan = Broadcast[ComponentMetricRequest]("resampling_req")
resampling_actor = ResamplingActor(
    # NO data_sourcing_req_chan here!
    resampling_req_chan.get_receiver(),
 )

logical_meter = LogicalMeter(
    data_sourcing_req_chan.get_sender(),  # each user decides where to attach the resampling
    resampling_req_chan.get_sender(),
)

# And then the resampling actor takes this request instead:

@dataclass
class TimeseriesResamplingRequest:
    data_source_channel_name: str

# So you ask to resample an arbitrary channel data

# And then logical meter at some point will have to subscribe to some metric, for example:

for (component_id, metric_id) in info_we_want:
    data_sourcing_req = ComponentMetricRequest(uuid, component_id, metric_id)
    data_sourcing_resp = self.data_sourcing_req_sender.send(data_sourcing_req)
    resampling_req = TimeseriesResamplingRequest(resp.channel_name)
    resampling_resp = self.resampling_req_sender.send(resampling_req)
    receivers.append(channel_registry.get_receiver(resampling_resp.channel_name))
```

I'm not sure if we need this kind of flexibility, but what I like about this is the resampling actor only do resampling and doesn't know **anything** about the source of data it is resampling, whatever you throw at it, as long as it is a channel of `Sample`s, it can resample it.

leandro-lucarella-frequenz, 2022-10-19:
As a side note (and not something to care about for this first implementation), since when we are wiring all the channels at the start of the app, I wonder if actors should create their own requests/control channels (and now we can even register them into the channel registry). So we could do (in my example):

```py
# New name for generating a unique channel, which could be registered to the channel registry
# as f"{self.__class__.__name__}_{name}" for example, we could even just create a UUID if no
# name is provided. We could get more crazy at some point and even create the name from the
# initialization arguments and fail if the same actor was created twice, or ensuring some actor
# is a singleton (global to the whole system), by letting it have only one channel name.
data_sourcing_actor = DataSourcingActor(name="components")
resampling_actor = ResamplingActor(name="_1s", resampling_period_s=1)

logical_meter = LogicalMeter(
    # We get the senders from the instance
    data_sourcing_actor.get_requests_sender(),
    resampling_req_chan.get_requests_sender(),
)
```

An advantage about this is person doing the composition doesn't need to know that much about how channels need to be created, and if it connects channels wrongly, the type system should make it fail anyway. The disadvantage is it could lead to some dependency issues if there are circular dependencies between actors, but if we go ahead with #36, things can be much better.

sahas-subramanian-frequenz, 2022-10-19:
> (also I guess channels for request will be Bidirectional, not Broadcast, right?)

true,  that's the only way to report back errors like unknown component_id, etc.  This would mean the request object won't need a channel uuid as well.

> but what I like about this is the resampling actor only do resampling and doesn't know anything about the source of data it is resampling

that makes sense as well,  and I think the extra compositional cost is definitely worth the flexibility.

> I wonder if actors should create their own requests/control channels (and now we can even register them into the channel registry)

At some point,  we'll want to start deploying actors independantly,  and then we won't have to restart all the actors when we just want to restart one.  If they are deployed separately,  then actors can't own the channels themselves,  but should use the registry.  I'm not sure we can get actors to create their own channels and still support the scenario that I mentioned.

thomas-nicolai-frequenz, 2022-10-19:
> since when we are wiring all the channels at the start of the app

Is that true? I would not bet on that. People might open and close channels dynamically.

thomas-nicolai-frequenz, 2022-10-19:
> true, that's the only way to report back errors like unknown component_id, etc. This would mean the request object won't need a channel uuid as well.

Is that needed? An unknown component_id just means not data is arriving but I don't think it needs a seperate back channel. I would rather suggest to implement this check on a higher level directly checking against the component graph.

thomas-nicolai-frequenz, 2022-10-19:
> but what I like about this is the resampling actor only do resampling and doesn't know anything about the source of data it is resampling

YES! BIG LIKE!

leandro-lucarella-frequenz, 2022-10-19:
> **EDIT** Actually this is not possible with the current `Bidirectional` implementation, because `Bidirectional` can have only two ends. If there is a single service end and many client ends, everything the service sends will be received by all the clients

Se we need to keep a list of all clients, like the power distributor, right? Maybe we need to support class to handle this if it will become a common patter used by many actors.

> At some point, we'll want to start deploying actors independently, and then we won't have to restart all the actors when we just want to restart one. If they are deployed separately, then actors can't own the channels themselves, but should use the registry. I'm not sure we can get actors to create their own channels and still support the scenario that I mentioned.

Yeah, is a good point about the restart, but if the actor uses the registry, then we should be fine. The only drawback is we can't use the registry to ensure singleton actors are singletons, because in this case we would need to just use an existing channel in the registry if there is already one there.

We can take more time to think about this one, I like that actors handle their own stuff as much as possible, but we need to be careful not to lose flexibility, I agree on that.

>> since when we are wiring all the channels at the start of the app
> Is that true? I would not bet on that. People might open and close channels dynamically.

Yeah, I meant "control" or "requests" channels. A channel you need to use to send requests to an actor, like "create new dynamic channels" :D

> I would highly recommend to NOT using backchannels (bidirectional) at all. Its not needed and if a use case comes up where its needed lets brainstorm how to avoid it or how to support it and where.

I'm not sure if we are talking about the same thing here, Tom. We are talking about control channels, so I want to tell the data sourcing actor it should open a new dynamic channel to send data for component C, metric M. If the actor can't fulfill that request, it would be nice to be able to return an error to the requester (for example, a logical meter). Otherwise the requester will have to wait for a while polling the channel registry to see if some pre-agreed channel exists or not. If we get response for requests also the actor creating the dynamic channels (like the data sourcing actor) can report back the name of the channel it created, so there is no need for pre-agreed channel names.

We are **NOT** talking about channels sending sample data, that is definitely unidirectional.

thomas-nicolai-frequenz, 2022-10-19:
> If the actor can't fulfill that request, it would be nice to be able to return an error to the requester (for example, a logical meter).

Why? What for?

leandro-lucarella-frequenz, 2022-10-19:
BTW @sahas-subramanian-frequenz I think now I understand why there is a `channel_uuid`, this is because you weren't thinking about responses, so the requester should pick a unique channel name and ask in the request to put the requested data in that particular channel, right?

thomas-nicolai-frequenz, 2022-10-19:
> Otherwise the requester will have to wait for a while polling the channel registry to see if some pre-agreed channel exists or not. If we get response for requests also the actor creating the [...]

So you mean the channel name is not derived from the channel properties? So the subscriber wouldn't set the channel name? Why would the sender having the say how the channel is named?

leandro-lucarella-frequenz, 2022-10-19:
> So you mean the channel name is not derived from the channel properties?

No, I never said that, I'm not talking at all about how channels names should be constructed.

> So the subscriber wouldn't set the channel name?

To clarify nomenclature, I assume "subscriber" is the actor sending a `ComponentMetricRequest` to the `DataSourcingActor`, what I called the "requester", and I assume it is also the "sender".

So yes, I'm saying the subscriber doesn't set the channel name. Just requests to subscribe to some component metric data and waits for the channel as a response to that request.

> Why would the sender having the say how the channel is named?

Why would the receiver make that decision? I think the sender is in a much better position to make the call. In the future if we want to do optimizations in terms of channels, and want to reuse channels when they reach the "live" state when they originally had a `start_date`, this can only be done and decided by the sender. I find it much harder to do any reuse channel optimization in the future if the receiver is the one in charge of deciding the channel name, because it doesn't have the whole picture.

thomas-nicolai-frequenz, 2022-10-19:
> To clarify nomenclature, I assume "subscriber" is the actor sending a ComponentMetricRequest to the DataSourcingActor, what I called the "requester", and I assume it is also the "sender".

oh to me thats the receiver of the channels data lol So the sender is the one that sends the data stream. *confused*

leandro-lucarella-frequenz, 2022-10-19:
Sorry, yeah, I meant the "receiver".

thomas-nicolai-frequenz, 2022-10-19:
> I find it much harder to do any reuse channel optimization in the future if the receiver is the one in charge of deciding the channel name, because it doesn't have the whole picture.

well it depends on if we know what the receiver wants. maybe he wants to seperate streams. I kinda see it the opposite. You're proposal is very centralised in a way. Thats why I proposed the optimization (combination of channels) on the LogicalMeter side and not the side of the DataSourceActor.

leandro-lucarella-frequenz, 2022-10-24:
Update from our last chat with @thomas-nicolai-frequenz and @sahas-subramanian-frequenz:

There was some misunderstanding about what I mean by "but what I like about this is the resampling actor only do resampling and doesn't know anything about the source of data it is resampling" and Tom screamed "YES! BIG LIKE!", so we will not go for this and have the Resampler tied to the data sourcing actor, and thus, the resampling actor will know (and also be coupled) to the concepts of *component ID* and *metric ID*.

So far we don't see a use case for a generic timeseries resampler, so it will be a component metric resampler instead.

The existence of the data source actor will be then sort of hidden to the users, which will only need to deal with the resampler. A use might be like this (no more `ResamplingRequest`):

```py
@dataclass
class ComponentMetricRequest:
    # a uuid to identify the output channel from the channel registry.
    channel_namespace: str
    # id of the requested component
    component_id: int
    # `metric_name` would be attribute names on BatteryData, etc.  Example
    # value: `soc`, `current_phase_1`, etc.  For individual phase current and
    # voltage values, we need to add additional attributes to `MeterData` and
    # `EVChargerData`.
    metric_id: str
    # the start time from which data is required.
    start_time: Optional[datetime]

    @property
    def channel_name(self) -> str:
        return f"{self.channel_namespace}:{self.component_id}:..."

# Logical meter at some point will have to subscribe to some metric, for example:

for (component_id, metric_id) in info_we_want:
    subscribe_req = ComponentMetricRequest("channel_namespace", component_id, metric_id)
    resampling_receiver = channel_registry.get_receiver(subscribe_req.channel_name)
    self.resampling_req_sender.send(subscribe_req)  # No response here

    receivers.append(resampling_receiver)
```

The resampling actor then will have to forward the subscription request to the data sourcing actor (changing the namespace so the input channel doesn't collide with the output channel).

thomas-nicolai-frequenz, 2022-10-24:
> Resampler tied to the data sourcing actor, and thus, the resampling actor will know (and also be coupled) to the concepts of component ID and metric ID.

Its my suggestion but just wanna make sure we could also choose the alternative approach to make things more modular. 

> So far we don't see a use case for a generic timeseries resampler, so it will be a component metric resampler instead.

Thats an interesting point. In your approach the `Resampler` wouldn't know anything about the `DataSource` actor. So it the very end move the code from left to right. @matthias-wende-frequenz seemed also be more in favour of would you suggested @leandro-lucarella-frequenz while @sahas-subramanian-frequenz and I where more coming from the use of use perspective. Not sure if @ela-kotulska-frequenz has an opinion about that as she's working on the `LogicalMeter` actor. @ela-kotulska-frequenz can you share your thoughts?

The main point though is that we might want to resample trading data at some point (BIG might). @leandro-lucarella-frequenz you really made me uncertain lol

leandro-lucarella-frequenz, 2022-10-25:
> @leandro-lucarella-frequenz you really made me uncertain lol I can see the beauty in your approach but it also makes the usage a little more complicated but also way more flexible. If we go for your approach the DataSource actor would also needing to be renamed to MicrogridDataSource actor as we might also have a TradingDataSource actor.

Yeah, this is exactly the architecture I had in mind, "Data source" actors would produce timeseries (or pseudo-timeseries if we allow them to not be strictly in order, which could make sense to do things a bit more real-time and remove the need to do buffering in the data sources, as the resampling actor will need to do buffering anyway), and thus the resampling actor would be completely data source agnostic. We just care about input and output produced in channels.

About the usage, it can always be hidden under a utility function that can make all the setup for you, but yeah, that still means more code needs to be written, even when the end user experience can be greatly improved.

That said, **I would prefer to continue with this approach now** so we can move forward and merge the close to ready PRs. The current approach is simpler in terms of coding needed and fulfills the current requirements, so we can move faster. After we have **something**, I can submit a draft PR with the architecture I have in mind, and it will be much easier to compare and understand that something just on paper. This way we are also parallelizing work, as we don't block everything until we come up with the best final architecture to rule them all.

What do you think?

PR #52 (2022-10-28): Add DataSourcingActor
Closes #30
commit e6b4b892 (2022-10-28): Add DataSourcingActor (#52)

Closes #30
closed 2022-10-28 by sahas-subramanian-frequenz