frequenz-floss/frequenz-sdk-python #31: Add a class that manages a buffer to get all the needed resampled data

leandro-lucarella-frequenz, 2022-10-14:
### What's needed?

We need to be able to wait until all metrics for a particular timestamp of resampled data have arrived. We know they will eventually arrive, but we need to store the metrics somewhere until we have all to do the needed calculation.

### Proposed solution

Add a class that manages this buffer. This class should own the buffer that holds all the data points that arrived so far for a particular timestamp, and have a method to tell when all the expected data points arrived for this particular timestamp, so whatever calculation needs to be made with the data can be done.

### Use cases

Many users of the resampling actor, like the `LogicalMeter` or `BatteryPool` will need to do this.

### Additional context

Example draft implementation:

```py
@dataclass
class Sample:
    timestamp: datetime
    value: float

class ResampledTimeSeriesWindow:
    _samples: Dict[TimeSeriesId, Sample]

    def __init__(self):
        self._samples = {}

    def process_sample(self, timeseries_id: TimeSeriesId, sample: Sample) -> Optional[Dict[TimeSeriesId, Sample]]:
        result = None
        if self._samples and self._samples[0].timestamp != sample.timestamp:
            result = self._samples
            self._samples = {}
        self._samples[timeseries_id] = sample
        return result

window = ResampledTimeSeriesWindow()

while True:
    ts, sample = get_sample()
    results = window.process_sample(ts, sample)
    if results:
        # We have a complete window
        process_window(results)
```

leandro-lucarella-frequenz, 2022-11-16:
FYI @sahas-subramanian-frequenz in case you can come up with something while addressing #66

sahas-subramanian-frequenz, 2022-11-16:
I already have a solution for doing that in #66:  https://github.com/frequenz-floss/frequenz-sdk-python/pull/66/commits/0b613ae7c325acac917396f7c9612ee202e6be74#diff-a1aa5eb2cd80b9e0c2cc053be5369a7e7abd327594ebec482827295f54534997R67-R87

It is very modular,  but also specific to scenarios where multiple metrics are needed for a single formula,  so we should be able to reuse it in BatteryPool also.  I will write up some documentation for it soon.

leandro-lucarella-frequenz, 2022-12-05:
I don't think that's what this issue is about. This issue is about basically passing a set of sample sources and get one resample window of data at the time.

leandro-lucarella-frequenz, 2022-12-05:
I think a simpler interface would be:

```py
async resampled_window(sources: Set[Source]) -> Map[Source, Sample]:
    ...
```

So this will take all the sources you want to group and wait for a sample in each (make sure all the timestamps match) and return a map with the resulting samples for a particular window.

This is at least what I had in mind when creating the issue and I found would have really useful to have when writing the 2nd stage resampling example.

leandro-lucarella-frequenz, 2023-04-21:
Now this is done in one place, the `FormulaEngine`, and we don't have any other use cases not covered by it for now, so closing.

closed 2023-04-21 by leandro-lucarella-frequenz