frequenz-floss/frequenz-sdk-python #33: Implement BatteryStatusActor to track what batteries are working

ela-kotulska-frequenz, 2022-10-18:
### What's needed?

We need external instance that would track waht batteries are working. 
It should be possible to get subset of working batteries anytime.
First it should be used by BatteryPool and PowerDistributor.

### Proposed solution

BatteryStatusActor should be Singleton that stores status of all battery-inverter pairs in the microgrid.

class BatteryStatusActor:
def __init__(self, api: MicrogridApi):
        # Subscribe for the metrics below for internal use
        metrics = [".result.inverter.state.componentState",  ".result.battery.state.componentState", ".result.battery.state.relayState" ]
        metric_channels = # For each battery-inverter pair in Microgrid: Subscribe for `self.metrics`, each channel should be peekable. 

def get_working_batteries(self, batteries_pool: Set[int]) -> Set[battery_id]:
   # Returns set of working batteries

   # Battery is consider as broken if:
   #    * inverter state in [UNSPECIFIED, UNAVAILABLE]
   #    * battery relay in [CLOSED]
   #    * battery state in [ERROR]
   #    * component didn't send any data for some time ( how long? it should be configurable )
   #    * latest error level returned from `set_power` method is Critical.
   #    * battery-inverter didn't respond for the latest `set_power` call
   # If battery is considered as NOT working, then change it to working if:
   #    * state of the inverter or battery changed - valid state
   #    * RelayState or state of the battery changed
   #    * some time has passed (increase exponentially until reach 30 sec - this should be configurable)
   #    * component started to send data
  
   # If all batteries given in argument are considered as broken, then relax the conditions to below:
   # Battery is consider as broken if:
   #    * inverter state in [UNSPECIFIED, UNAVAILABLE]
   #    * battery relay in [CLOSED]
   #    * battery state in [ERROR]
   #    * component didn't send any data for some time

### Use cases

BatteryPool need subset of working batteries to return data (capacity, active_power) for working batteries only.
PowerDistributor needs subset of working batteries to charge/discharge only working batteries. 

### Alternatives and workarounds

BatteryStatus (poor version) is not part of PowerDistirbutor. But it has bugs and only PowerDistributor can use it.

ela-kotulska-frequenz, 2022-10-19:
Unfortuatelly these ideas didn't work out :/ 
New propostion:
[image]

This is how the program initialization would look like.
[image]

leandro-lucarella-frequenz, 2022-10-19:
Without being super familiar with the fine details of how batteries should be handled and after having a quick call with Ela, it sounds reasonable, so let's see if this work.

closed 2022-12-28 by ela-kotulska-frequenz