frequenz-floss/frequenz-sdk-python #37: BMS temperature data support

idlir-shkurti-frequenz, 2022-10-19:
### What's needed?

Battery temperature data is needed for lynch in order to decide whether to (dis)charge early within a 15min window

### Proposed solution

Temperature for **each** battery id within the microgrid. 

### Use cases

Peak shaving (GMM)

idlir-shkurti-frequenz, 2022-10-20:
If `max` is the maximum temperature of all batteries at the current time then I need `max`.

ela-kotulska-frequenz, 2022-10-20:
No no xD

Protobuf BatteryData has 3 values:
raw.battery.data.temperature.max
raw.battery.data.temperature.min
raw.battery.data.temperature.avg

Which one you need? Definitions in the API:
https://github.com/frequenz-floss/frequenz-api-microgrid/blob/v0.x.x/proto/frequenz/api/microgrid/battery.proto#L178-L180
https://github.com/frequenz-floss/frequenz-api-microgrid/blob/v0.x.x/proto/frequenz/api/microgrid/common.proto#L48-L56

idlir-shkurti-frequenz, 2022-10-20:
Ah ok I see. For this usecase we need the `max` value.

sahas-subramanian-frequenz, 2022-10-20:
@ela-kotulska-frequenz the max/min from protobuf are for individual blocks.  We need to expose only the `avg`,  because that's the temperature of each bms.

closed 2022-11-23 by sahas-subramanian-frequenz