frequenz-floss/frequenz-sdk-python #39: Improve `@property` documentation

leandro-lucarella-frequenz, 2022-10-19:
### What's needed?

Right now we are documenting properties as if they were functions (a description and a `Return` section are both required), but they are clearly not.

Google style says properties should be documented as attributes (in the class documentation) but darglint doesn't support this.

### Proposed solution

So far the only option seems to be fixing the issue in the upstream `darglint` tool.

### Alternatives and workarounds

We tried to use `# noqa` as suggested in the darglint issue but according to @jakub-toptal it doesn't work:

Before:

    @property
    def microgrid_api_client(self) -> MicrogridApiClient:
        """Get MicrogridApiClient.

        Returns:
            api client
        """
        return self._api

After:
```py
    @property  # noqa
    def microgrid_api_client(self) -> MicrogridApiClient:  # noqa
        """Get MicrogridApiClient."""  # noqa
        return self._api  # noqa
```

Still fails with darglint code: DAR201

### Additional context

There is a darglint issue for this, and at some point we might want to give it a shot to improve property handling in darglint:

* https://github.com/terrencepreilly/darglint/issues/94

leandro-lucarella-frequenz, 2022-11-04:
We will generate the documentation with `mkdocstrings`, which also doesn't support properties super gracefully. It doesn't work well with `Attributes:` either.

To produce a decent result, the minimum documentation needed is:

```py
class AClass:
    """A class."""

    @property
    def a_prop(self) -> int:
        """
        Returns:
            The a prop of the class.
        """
        return 1

    @a_prop.setter
    def a_prop(self, a: int) -> None:
        pass
```

(note the `Returns:` can be next to the `"""`, otherwise it doesn't work either, and if the `Returns:` is not included, then the type of the property is not included either)

It produces this results:

[image]

(so it's not ideal that it is listed as a function, but it is nicely annotated as a writable property)

But `pydocstyle` complains about this:
```
in private method `a_prop`:
        D205: 1 blank line required between summary line and description (found 0)
in private method `a_prop`:
        D400: First line should end with a period (not ':')
in private method `a_prop`:
        D401: First line should be in imperative mood (perhaps 'Return', not 'Returns')
```

So it basically demands a text besides the `Returns:` section. darglint doesn't complain with this.

So we can explore if we could disable those `pydocstyle` somehow, or just live with the redundancy of having the properties documented twice. This passes all tests:

```py
class AClass:
    """A class."""

    @property
    def a_prop(self) -> int:
        """Get the a prop.

        Returns:
            The a prop of the class.
        """
        return 1

    @a_prop.setter
    def a_prop(self, a: int) -> None:
        pass
```

leandro-lucarella-frequenz, 2022-11-04:
Given the amount of work that could be involved in making sure all the tools we use work well with one particular way to documenting properties, I'm starting to lean more towards to just drop this issue and bite the bullet.

leandro-lucarella-frequenz, 2022-11-22:
Adding the proposed solution to the issue body.

leandro-lucarella-frequenz, 2023-06-02:
Doing some tests with `pydoclint` / `pydocstyle` / `mkdocstrings` it seems like now we should be able to just write:

```py
@property
def prop(self) -> T:
    """The property."""
```

And it should work fine, so this might get a lot easier soon.

llucax, 2023-07-06:
Blocking on adopting pydoclint

llucax, 2023-08-24:
For progress status on `pydoclint`: There were 3 main issues still blocking the adoption:

- [x] https://github.com/jsh9/pydoclint/issues/55 (PR accepted and released already)
- [x] https://github.com/jsh9/pydoclint/issues/54 (not accepted, so we are going with using `flake8` and adding `# noqa` for now)
- [ ] https://github.com/jsh9/pydoclint/issues/76 (just reported, waiting for an answer to see what we are doing with it)

pawamoy, 2023-08-24:
mkdocstrings can now show the return annotation in property headings, as well as in the separate signature if the corresponding option is enabled. The next version of mkdocstrings-python Insiders will also add cross-references to properties signatures, given they are separated from the heading :slightly_smiling_face: Future versions might allow cross-references in headings themselves.

llucax, 2023-08-25:
Nice, thanks!