frequenz-floss/frequenz-sdk-python #53: Generate API documentation

leandro-lucarella-frequenz, 2022-10-27:
### What's needed?

We need to generate the API documentation so it can be consulted more easily.

### Proposed solution

Use a tool to generate the API documentation by extracting it from the *docstrings*.

### Alternatives and workarounds

Look at the documentation in the source code.

leandro-lucarella-frequenz, 2022-10-27:
So far I've been playing with all of them and my conclusions so far are:

* Sphinx is discarded
* pdoc3 is the easiest to setup for API docs, but it is very limited for high level documentation.
* MkDocs seems like the best alternative because of the good combination between simplicity, modernness, flexibility and power.

I'm mainly considering now:

* Using a combination of MkDocs for high-level docs + pdoc3 for API, so it would be like 2 separate websites

* Considering writing a simple tool to generate the hierarchical structure for docs from the source.

  I succeeded with a very simple script:

  ```sh
  for f in $(find src -name '*.py' -not -name '__init__.py')
  do
    basename=$(echo $f | sed 's|src/frequenz/sdk/\(.*\)\.py|\1|')
    m=$(echo $basename | tr / .)
    mkdir -p docs/api/$(dirname $basename)
    echo "::: frequenz.sdk.$m" > docs/api/$basename.md
  done
  ```

  Which is a good start, although it is a bit annoying that we'll need to either put this in every project or create a tool and add it as a dependency for all projects.

leandro-lucarella-frequenz, 2022-10-28:
I decided to go with `MkDocs` even if it is a little bit more work to get the API docs running because I feel on the long term using pdoc3 will just not be enough.

The results are pretty neat, I will create a PR next week.

pawamoy, 2022-10-29:
You might be interested in this recipe: https://mkdocstrings.github.io/recipes/#automatic-code-reference-pages (if you didn't see it already). It allows to automatically build you code references pages, one page per module :slightly_smiling_face:

leandro-lucarella-frequenz, 2022-10-31:
Preliminary results: https://leandro-lucarella-frequenz.github.io/frequenz-sdk-python/reference/frequenz/sdk/

pawamoy, 2022-10-31:
I see some of your `Returns` tables have multiple entries instead of one: it's because mkdocstrings supports multiple items for returns section. It's kinda confusing for users so we will add an option to disable it. In the mean time you can fix that formatting by indenting your continuation lines.

From:

```python
"""
...

Returns:
    Some item description
    that spans multiple
    lines.
"""
```

To:

```python
"""
...

Returns:
    Some item description
        that spans multiple
        lines.
"""
```

leandro-lucarella-frequenz, 2022-10-31:
> You might be interested in this recipe: [mkdocstrings.github.io/recipes/#automatic-code-reference-pages](https://mkdocstrings.github.io/recipes/#automatic-code-reference-pages) (if you didn't see it already). It allows to automatically build you code references pages, one page per module slightly_smiling_face

Hi @pawamoy, I haven't seen that, I will take a look, thanks for pointing it out and for the great tool!

> I see some of your Returns tables have multiple entries instead of one: it's because mkdocstrings supports multiple items for returns section. It's kinda confusing for users so we will add an option to disable it. In the mean time you can fix that formatting by indenting your continuation lines.

Thanks for having a look at this too! Yeah, some errors are to be expected for sure, we never generated any docs from the docstrings before, so we'll have to go through the docs and make sure to fix some of these issues.

sahas-subramanian-frequenz, 2022-10-31:
@leandro-lucarella-frequenz this PR indents the return value descriptions:  https://github.com/frequenz-floss/frequenz-sdk-python/pull/54

leandro-lucarella-frequenz, 2022-10-31:
Just for the records, I'm including my script to generate the stubs for the reference documentation, which grew quite a bit from the PoC above.

But I will probably switch to the recipe in mkdocstrings recipe, it is both simpler and more correct (for example I was looking at the wrong edit links when I saw @pawamoy comment).

I used [`mkdocs-awesome-pages-plugin`](https://pypi.org/project/mkdocs-awesome-pages-plugin/) to customize the navigation menu instead, I see [`mkdocs-literate-nav`](https://github.com/oprypin/mkdocs-literate-nav) presents itself as a replacement for it. Did you have a particular reasons to use `mkdocs-literate-nav` in the recipe instead of `mkdocs-awesome-pages-plugin`?

<details>
<summary>`autoapi` script</summary>

```sh
#!/bin/sh
set -ue

src_dir=src
doc_dir=docs/reference

for f in $(find "$src_dir" -name '*.py')
do
	basepath=$(echo $f | sed 's|^'"$src_dir"'/\(.*\)\.py$|\1|')
	basedir=$(dirname $basepath)
	basename="$(basename "$basepath")"
	if echo "$f" | grep -q '/_'
	then
		if echo "$f" | grep -q '/_.*/'
		then
			continue
		elif echo "$f" | grep -q '/__init__.py$'
		then
			package=$(echo $basedir | tr / .)
			doc_file="$basedir/index"
		else
			continue
		fi
	else
		package=$(echo $basepath | tr / .)
		doc_file="$basepath"
	fi
	mkdir -p $doc_dir/$basedir
	echo "::: $package" > $doc_dir/$doc_file.md
done

for d in $(find "$doc_dir" -type d)
do
	pkgpath=$(echo $d | sed 's|^'"$doc_dir"'/\?||')
	package=$(echo $pkgpath | tr / .)
	awesome_pages_conf="$d/awesome-pages.yml"
	echo "nav:" > "$awesome_pages_conf"
	if test -f "$d/index.md"
	then
		echo "  - $package: index.md" >> "$awesome_pages_conf"
	fi
	for f in "$d"/*
	do
		echo "f: $f"
		if test "$(basename $f)" = "index.md"
		then
			echo "excl: $f"
			continue
		elif test -d "$f"
		then
			dirname=$(basename "$f")
			echo "  - ${package:+$package.}$dirname: $dirname" >> "$awesome_pages_conf"
		elif echo "$f" | grep -q ".md$"
		then
			basename=$(basename "$f" .md)
			echo "  - .$basename: $basename.md" >> "$awesome_pages_conf"
		fi
	done
done
```

</details>

I ended up baking my own solution for the *autoapi*

pawamoy, 2022-10-31:
mkdocs-literate-nav is simply more focused on that very specific task, and integrates well with mkdocs-gen-files. mkdocs-awesole-pages is probably overpowered for the recipe in mkdocstrings docs, and might not allow to dynamically build the nav as easily.

leandro-lucarella-frequenz, 2022-10-31:
> mkdocs-literate-nav is simply more focused on that very specific task, and integrates well with mkdocs-gen-files. mkdocs-awesole-pages is probably overpowered for the recipe in mkdocstrings docs, and might not allow to dynamically build the nav as easily.

Yeah, I finally got time to get into it using the recipe and I saw the integration between `mkdocs-literate-nav` and `mkdocs-gen-files`, so I finally just use the recipe basically. Thanks again for the recommendation and clarifications.

pawamoy, 2022-10-31:
Glad I could help!

leandro-lucarella-frequenz, 2022-11-02:
For the records, PR for the channels repo: https://github.com/frequenz-floss/frequenz-channels-python/pull/25

After that's merged the same should be done here.

closed 2022-11-22 by leandro-lucarella-frequenz