frequenz-floss/frequenz-sdk-python #58: Implement an efficient ring buffer

leandro-lucarella-frequenz, 2022-11-02:
### What's needed?

We need a [ring buffer](https://en.wikipedia.org/wiki/Circular_buffer) that is efficient, memory and time wise.

### Time requirements

Ring buffers should be able to:

* Push data in O(1)
* Pop data in O(1)
* Random access in O(1)

Sometimes we will need to keep these buffers sorted (if data can come out-of-order), a sorted ring buffer should be able to:

* Push or Pop data preserving order in O(log N)
* The other operation should still be O(1)
* Ideally it should be possible to select which operation is O(1), so users can pick the best implementation to be used when it is forseen to have more pushes than pops or vice-versa.

### Memory requirements

* Ideally the data in the buffer is not scanned by the GC (this might be hard if the data that is stored is not plain data, like `str`, `int`, `float`, `tuple`).

* It should be memory-safe even if objects (non-plain data) is stored in them.

* We'll need to be careful that the data stored in the buffer is plain data (for example store a `tuple` instead of a class, as classes are tracked by the GC).

* Ideally it should be possible to copy the buffers using a simple memory copy.

### Proposed solution

Do some benchmarks testing these underlying data structures to store ring buffers:

* `list`: This is an *array* in Python. The problem is, it is tracked by the GC, and for sorted buffers we need to copy data around to insert sorted, so it might not be suitable for that

* `deque`: This is a doubly linked list in Python, [implemented](https://github.com/python/cpython/blob/75a6fadf369315b27e12f670e6295cf2c2cf7d7e/Modules/_collectionsmodule.c#L78-L82) as a list of ([64](https://github.com/python/cpython/blob/75a6fadf369315b27e12f670e6295cf2c2cf7d7e/Modules/_collectionsmodule.c#L27)) blocks with free lists, so for small buffers it should be pretty similar to a `list` and for bigger buffers it should be worse to do random access but better to do random insertions.

* `numpy.array`: This is an array that escapes the GC, but we need to depend on `numpy` to use it. Also we need to check if it couldn't be error prone if we store non-plain data in them, since they are not scanned by the GC, if we store objects only there, they might be potentially freed by the GC, leading to memory issues.

### Use cases

A lot of parts in the system will use ring buffers, some of them might be pretty big, like actors storing some window of historical data.

### Additional context

Since Python uses a generational GC, and we expect buffers to be long lived, the buffers should be promoted to older generations so they should be rarely scanned after a while. Also absolutely all other objects in the program will still be handled normally by the GC, so unless ring buffers are a considerable portion of the total memory consumed by programs, the impact of optimizing the ring buffer might not be relevant anyway.

So maybe escaping the GC is not that important. This is why we need **realistic** benchmarks before worrying too much about this.

Here is a nice introduction about how Python's GC works: https://rushter.com/blog/python-garbage-collector/

thomas-nicolai-frequenz, 2022-11-11:
We do have an additional requirement:

As the ring buffer might e.g. store the last 15-min worth of data with 1 sample per second resulting in 900 entries we need a way to keep track of a certain entries position inside the buffer to be able to effeciently seek to that position. Otherwise we'd always need to iterate 899 entries in the worse case to find the entry we are interested in. The current use case for that is in Lynch where @idlir-shkurti-frequenz uses [numpy.where ](https://numpy.org/doc/stable/reference/generated/numpy.where.html) which does in fact do a for loop through an entire list to find the matching entry. If we do this once a second this feels very inefficient. The reason we do this is that Lynch is only interested into all samples from the current 15-min time window. So if we are at 00:15:01 we are not interested in the remaining 898 entries. Thus it would be good to know the position in the buffer where the record for 00:15:00 is stored so we could jump straight to that entry.

@sahas-subramanian-frequenz suggested the following things:

We can of course still try to keep ring buffer implementation simple by allowing external code to
1. place and remove bookmarks based on whatever criteria, or
2. even simpler, external code can just keep track of the position where the window changed,  and request records from that point

The key requirement is to be able to set a pointer to e.g. the first record with a timestamp >= 00:15:00 and with every new sample coming in the bookmark has to be incremented until the entry is dropped from the buffer. However, we could also implement something more dynamic where we just say we need to keep a bookmark pointer to the start of each e.g. 1min or 15min window. So the developer would define a duration and not an absolute timestamp.

thomas-nicolai-frequenz, 2022-11-11:
The other core requirement is of course an efficient means of iterating the buffer by reference and avoiding any type of copying but I guess thats clear.

leandro-lucarella-frequenz, 2022-11-15:
> The key requirement is to be able to set a pointer to e.g. the first record with a timestamp >= 00:15:00 and with every new sample coming in the bookmark has to be incremented until the entry is [...]

It looks like we need some sort of indexing (as some sort of DB index). I will try to keep the ring buffer implementation simple and like it is suggested there try to build the complexity on top, by adding maybe some indexing capabilities on top of the ring buffer. But in essence what I see is needed is random access to the ring buffer, which gives us O(1) *pointers* that can be kept in a separate index for example.

thomas-nicolai-frequenz, 2022-11-15:
> It looks like we need some sort of indexing (as some sort of DB index

If we don't have that the peak shaving application will need to do 900 iterations every second. I agree going simple first but we should also aim for a simple implementation of this pointer. I feel like calling it a database index is way too much because that would assume we need to know every data point in the ring buffer which is not what we need. We only need to keep track of one data point at a time and should not aim to go overboard with the most complex solution possible. To me its more like a piggyback solution rather than a database. Its like an anchor. Do not abstract this too much from the use case we need. The use case is in fact quite simple. Yes, it can and should be an addition to the circular buffer. So maybe we should create another issue and split this topic out.

thomas-nicolai-frequenz, 2022-11-15:
> But in essence what I see is needed is random access to the ring buffer, which gives us O(1) pointers that can be kept in a separate index for example.

Just a single one. There is just a use case where we only need to keep track of this for a single data sample in time. There is no use case for multiple.

leandro-lucarella-frequenz, 2022-12-27:
Just for the records, I tried to implement an optimized ring buffer for the resampler but the results weren't good (2x slower than using a `deque`): https://github.com/frequenz-floss/frequenz-sdk-python/pull/130.

At some point it might be still worth implementing the pointer to the different resampling windows, so we don't need to bisect to find them on each resampling operation.

thomas-nicolai-frequenz, 2022-12-28:
> Just for the records, I tried to implement an optimized ring buffer for the resampler but the results weren't good (2x slower than using a deque)

Did you also try to limit the memory and see how often the GC would kick in in both cases? I think we should be careful just looking at the speed. Also did you consider using a numpy based data structure? From what "people" say this lives outside the Python GC.

PR #139 (2023-01-26): Sorted Ringbuffer
fixes #58
commit 13455af2 (2023-01-26): Sorted Ringbuffer (#139)

fixes #58
closed 2023-01-26 by mathias-baumann-frequenz