frequenz-floss/frequenz-sdk-python #61: A `LogicalMeter` implementation

shsms, 2022-11-07:
### What's needed?

A `LogicalMeter` implementation that streams:

1. grid power
2. ev power
3. client load
4. battery power
5. pv power

Their default formulas will be inferred from the component graph,  and the formulas can be overridden from the UI.

It should connect to the `ResamplingActor` to get resampled data for individual components.

thomas-nicolai-frequenz, 2022-11-07:
> client load

we will need a clear definition for this. @sahas-subramanian-frequenz and I discussed this and we did not have a good solution. it essentially means the product or consumption excluding EV, battery, pv power and everything else we can measure. Literally spoken its the `remaining power` and it computes by `grid power - everything that is measurable`.

shsms, 2022-11-07:
After an offline discussion with @thomas-nicolai-frequenz , we decided to make not expose a "client load" formula from the `LogicalMeter`, and instead provide two approaches for users to manipulate the formulas in specific ways:

1. a way to combine (add/subtract) the outputs of formulas, to create higher-order formulas
2. a way to exclude components of a specific type from a formula.

and apps can calculate client load this way,  by excluding batteries from the grid power, for example.

thomas-nicolai-frequenz, 2022-11-08:
> apps can calculate client load this way, by excluding batteries from the grid power

@idlir-shkurti-frequenz any objections?

thomas-nicolai-frequenz, 2022-11-08:
> a way to exclude components of a specific type from a formula.

... either by giving the ID of a component or excluding an entire component types like batteries, ev charger, ....

idlir-shkurti-frequenz, 2022-11-08:
LGTM!

shsms, 2022-11-11:
I was discussing with @tiyash-basu-frequenz about how to identify PV meters,  in order to automatically generate a formula for PV production.

The strategy is to check if a meter has a PV inverter as a successor. This works if the component graph has a PV inverter component,  but there might not always be a PV inverter component.  

There are locations where PV meters are dangling, without any PV inverter as successor,  especially if the PV inverter at the location doesn't have a communication interface.  Users would be able to add dummy PV inverters,  but they might not do it.  And in these locations,  we can't automatically generate a formula for PV power,  but users will have to provide a formula on the UI.

This also means that when apps want to calculate client load,  they will not be able to exclude PV by component type,  but will have to exclude the PV meter by component ID.

thomas-nicolai-frequenz, 2022-11-11:
> There are locations where PV meters are dangling, without any PV inverter as successor, especially if the PV inverter at the location doesn't have a communication interface. Users would be able to add dummy PV inverters, but they might not do it.

If the user doesn't do that its not the job of the LogicalMeter to fix it. If there is no PV inverter then this is not consider a PV meter! Please keep things simple! Nothing the LogicalMeter should fix!

thomas-nicolai-frequenz, 2022-11-11:
> but users will have to provide a formula on the UI.

That would be the alternative, sure.

thomas-nicolai-frequenz, 2022-11-11:
> This also means that when apps want to calculate client load, they will not be able to exclude PV by component type, but will have to exclude the PV meter by component ID.

There is no such thing as client load!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

shsms, 2022-11-11:
> There is no such thing as client load!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! :-)

yes, according to the SDK,  but @idlir-shkurti-frequenz would still have to know that this is the strategy,  because we also have locations where there are meters measuring PV power,  but without a PV inverter component in the graph.

thomas-nicolai-frequenz, 2022-11-11:
> but @idlir-shkurti-frequenz would still have to know that this is the strategy, because we also have locations where there are meters measuring PV power, but without a PV inverter component in the graph.

then we should fix this by (order in priority):
1. Adding the PV inverter to the UI.
2. If 1 does not work which it should but just in case:
    - set custom formula for PV power computation in the UI
    - use `net_power = logical_meter.grid_power() - logical_meter.pv_power()`

shsms, 2022-11-11:
> Adding the PV inverter to the UI.

@tiyash-basu-frequenz this should be possible for our locations right?

thomas-nicolai-frequenz, 2022-11-11:
> net_power = logical_meter.grid_power() - logical_meter.pv_power()

And this should be our default way to compute the net_power aka client load ;-) If we do this we have two ways of fixing the problem in no time, in both ways through the UI ;-) @idlir-shkurti-frequenz shout if this doesn't work!

shsms, 2022-11-11:
> > net_power = logical_meter.grid_power() - logical_meter.pv_power()
>
> And this should be our default way to compute the net_power aka client load ;-)

yes,  that would be a very cool interface,  considering that that would be applying the `-` operation on two streams,  and produces a new stream that returns the calculated value.  It should be easy to achieve.  

I have turned the issue into a tracking issue,  with a number of sub-tasks listed (at the top of this page), to track all the desired features.

matthias-wende-frequenz, 2022-11-17:
> There are locations where PV meters are dangling, without any PV inverter as successor, especially if the PV inverter at the location doesn't have a communication interface. Users would be able to [...]

A grid that has dangling meters would be rendered incomplete. I suggest to have the possibility to always have the possibility to add PV Inverters, even if they either don't have a communication interface or it's not used!

In the future having dangling meters could be a sanity check, i.e. if there is a meter node without any successor the micro-grid is not fully configured.

sahas-subramanian-frequenz, 2023-01-02:
A basic implementation of the logical meter is now complete.  A few additional formulas and support for 3-phase current formulas need to be added,  but those are being tracked in other issues.

closed 2023-01-02 by sahas-subramanian-frequenz