frequenz-floss/frequenz-sdk-python #63: MicrogridGrpcClient.set_power does not respect system power bounds

david-natingga-frequenz, 2022-11-08:
### What happened?

Trying to charge `set_power` at rates exceeding the system bounds results in an error.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/frequenz/sdk/microgrid/client.py", line 510, in set_power
    result: Empty = await self.api.Charge(
  File "/usr/local/lib/python3.8/site-packages/grpc/aio/_call.py", line 290, in __await__
    raise _create_rpc_error(self._cython_call._initial_metadata,
grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
	status = StatusCode.OUT_OF_RANGE
	details = "Command Charge(14031) failed. Charge power 14031W is outside battery's DC power bounds { system_bounds: Some(Bounds { lower: -145000.0, upper: 8163.381 }), component_bounds: Some(Bounds { lower: -161400.0, upper: 117400.0 }), rated_bounds: Some(Bounds { lower: -145000.0, upper: 92000.0 }) }"
	debug_error_string = "UNKNOWN:Error received from peer ipv6:%5B::%5D:62060 {created_time:"2022-11-08T10:57:40.514861837+00:00", grpc_status:11, grpc_message:"Command Charge(14031) failed. Charge power 14031W is outside battery\'s DC power bounds { system_bounds: Some(Bounds { lower: -145000.0, upper: 8163.381 }), component_bounds: Some(Bounds { lower: -161400.0, upper: 117400.0 }), rated_bounds: Some(Bounds { lower: -145000.0, upper: 92000.0 }) }"}"
```

### What did you expect instead?

The `set_power` function should be bound aware and should trim any power outside of the bounds with warning and charge/discharge at the maximum rate possible instead of failing.

### Affected version(s)

>= 0.11.0

### Affected part(s)

Microgrid (API, component graph, etc.) (part:microgrid)

david-natingga-frequenz, 2022-11-14:
I am closing this as using `PowerDistributor` we can overcome the issue.

sahas-subramanian-frequenz, 2022-11-14:
Thanks David!

closed 2025-05-16 by llucax