frequenz-floss/frequenz-sdk-python #65: Make `ConfigManager` read configuration from multiple files

leandro-lucarella-frequenz, 2022-11-15:
### What's needed?

Right now the `ConfigManager` is reading the configuration only from one file, but we need a way to set local config defaults and local config overrides.

### Proposed solution

The following config files should be read, by order of precedence:

1. **Local config defaults** will work as per-site constants, something to override the defaults provided by the code itself.
2. **Cloud sync config** is the configuration file written by the the Cloud Sync service with values taken from the UI.
3. **Local config overrides** will work as an emergency override file, to be able to *lock* some configuration in case some particular installation doesn't want to allow configuring some stuff through the UI for safety or other reasons.

thomas-nicolai-frequenz, 2022-11-15:
> Local config defaults will work as per-site constants, something to override the defaults provided by the code itself.

I suggest these to be defined as constants in the code.

leandro-lucarella-frequenz, 2022-11-15:
They will be too, this is a way to override the constants per location.

PR #1091 (2024-10-25): config: Allow reading from multiple files
The `ConfigManagingActor` can now take more than one config file, and will read multiple files one after the other, overriding the previous values.

Fixes #65.
commit ff786cd0 (2024-10-25): config: Allow reading from multiple files (#1091)

The `ConfigManagingActor` can now take more than one config file, and
will read multiple files one after the other, overriding the previous
values.

Fixes #65.
closed 2024-10-25 by llucax