frequenz-floss/frequenz-sdk-python #100: Make `nones_are_zeros` a keyword-only parameter in `FormulaEngine`

sahas-subramanian-frequenz, 2022-12-02:
Reading this, I recommend making the `nones_are_zeros` arguments a keyword-only argument (`def f(..., *, nones_are_zeros: bool`), otherwise is very hard to know what's going on on the caller site without having to go see the function definition.

_Originally posted by @leandro-lucarella-frequenz in https://github.com/frequenz-floss/frequenz-sdk-python/pull/95#discussion_r1038235546_

leandro-lucarella-frequenz, 2022-12-03:
If this argument is only a temporary hack and will be removed in a future release, then I don't think this is worth it.

llucax, 2023-08-10:
It seems like this argument didn't go away and it feels like it is not temporary anymore. We should decide what to do about it.

commit 7f647fad (2023-08-29): Make nones_are_zeros a keyword-only parameter (#611)

Accept keyword-only argument `nones_are_zeros` in `FormulaEngine` and
related classes and methods in order to clarify behavior at the caller's
end, eliminating the need to inspect function definitions for
understanding.

Fixes #100