frequenz-floss/frequenz-sdk-python #99: Handle `NaN` values in FormulaEngine

sahas-subramanian-frequenz, 2022-12-02:
While you are at this, it should probably be a good idea to also treat *NaN*'s specially (probably the same as `None`). You can check using `math.isnan()`.

(same for all steps)

**OR**, maybe replace `None` with *NaN* and it should have the same effect of propagating `None` manually as you are doing here, but it will be done automatically. Then you only need to convert the final *NaN* in the stack to `None` and you are set. This option is easier to write (as you don't have to touch every step) but it might be harder read (to figure out what's going on).

_Originally posted by @leandro-lucarella-frequenz in https://github.com/frequenz-floss/frequenz-sdk-python/pull/95#discussion_r1038228733_

daniel-zullo-frequenz, 2023-01-03:
@sahas-subramanian-frequenz what is the rationale to treat float `nan` as `None`. In any case if we try `nan` in a special way, should we do the same for `inf`?

daniel-zullo-frequenz, 2023-01-04:
I'd like to check with you if this is what you have in mind to handle None/NaN, please see the example below

```python
from math import isnan, isinf

class Adder(FormulaStep):
    """A formula step for adding two values."""

    def __repr__(self) -> str:
        """Return a string representation of the step.

        Returns:
            A string representation of the step.
        """
        return "+"

    def apply(self, eval_stack: List[Optional[float]]) -> None:
        """Extract two values from the stack, add them, push the result back in.

        Args:
            eval_stack: An evaluation stack, to apply the formula step on.
        """
        val2 = float("NaN") if (val := eval_stack.pop()) is None else val
        val1 = float("NaN") if (val := eval_stack.pop()) is None else val
        res = val1 + val2
        if isnan(res) or isinf(res):
            res = None
        eval_stack.append(res)
  ```

shsms, 2023-01-04:
Hey @daniel-zullo-frequenz ,  I think that makes sense,  but I wonder what kind of perf cost this would have.  Maybe we should write some benchmark scripts for the data pipeline first.  We can talk about that this afternoon.

But another option for implementing this,  is to just check incoming values in the `MetricFetcher` if they are `NaN` or `INF`, and replace those with `None` in that step alone.

shsms, 2023-01-04:
Or like Luca originally mentioned,  we could replace `None` values with `NaN`, or `0` in the `MetricFetcher`, and remove all the special `None` handling in the other steps,  and in the engine's `apply` function,  replace the output with `None` if it was `NaN`.  Maybe that's be most efficient approach.

daniel-zullo-frequenz, 2023-01-04:
That makes sense. I'll work out the changes and push a MR. Thanks for the clarification!