frequenz-floss/frequenz-sdk-python #113: Define a data structure for storing data batches

matthias-wende-frequenz, 2022-12-13:
### What's needed?

We need to store batches of data that are received from the `LogicalMeter` and can be used by downstream users, usually trained machine learning models.

This should use a ring buffer as an underlying data structure
* can access data by time windows
* can create an ordered copy of a slice of the ring buffer as numpy array

### Use cases

This will be used in the forecasting pipeline as the central data storage for

* storing time series data that can be piped into ml predictions,
* calculating features as
	* daily averages
	* energy consumption/production in a certain time window
	* mean and standard deviation
	* polynomial coefficients (for power curves)

### Alternatives and workarounds

In the past pandas dataframes have been used. It turned out as too inefficient due to

* memory allocation when operating (e.g. (joining, resampling,...),
* creation of new dataframes on new incoming data.

matthias-wende-frequenz, 2022-12-19:
This data structure could furthermore take care of loading historic data, as an alternative approach to the seamless transition between disk data and streaming data. This could be simplified if we would cache transformed logical meter data on disk.

ela-kotulska-frequenz, 2022-12-20:
How frequently would the mean be updated and send?

matthias-wende-frequenz, 2022-12-21:
> How frequently would the mean be updated and send?
Does this question belong to [this](https://github.com/frequenz-floss/frequenz-sdk-python/issues/120) issue? This module is just about batching data from a stream.

matthias-wende-frequenz, 2022-12-21:
For resampling we might consider using the python wrapper around [libsamplerate](https://pypi.org/project/samplerate/)

matthias-wende-frequenz, 2022-12-27:
Possible Names:

* StreamBuffer
* SlidingWindow
* SlidingBucket

thomas-nicolai-frequenz, 2022-12-27:
> I'd favor SlidingWindowOperator as it's an operator that operates on a data stream and creates a sliding window from it.

What about something more simple like `StreamBucket`? I think the term `Stream` should be certainly part of the name. Not sure if `window` is the right term when thinking about a stream of samples. There is either pools or buckets that can capture things for a certain period of time. A window can't hold anything. A window is something to look through.

thomas-nicolai-frequenz, 2022-12-27:
> This should use a ring buffer as an underlying data structure

Shall this be called RingBuffer or [CircularBuffer](https://en.wikipedia.org/wiki/Circular_buffer)?

thomas-nicolai-frequenz, 2022-12-27:
> How frequently would the mean be updated and send?

@ela-kotulska-frequenz let me define the scope of this issue:

The scope of the `StreamBucketOperator` aka `SlidingWindowOperator` is to capture the timeseries samples being streamed by the LogicalMeter and store them in a `RingBuffer`. There are no operations being done on the timeseries samples rather just storing a predefined number of samples.
* Holding n-number of past samples that arrived from the LogicalMeter
* Provide an iterator that allows iterating all samples stored in the underlying `RingBuffer`
* Provide the ability to set a bookmark to a certain samples timestamp so that iterating all samples to seek for a specific sample can be avoided. 
* Dump RingBuffer to disk on regular interval and app shutdown so that a fast restart is supported and the last state can be recreated if possible

@matthias-wende-frequenz anything I'm missing in terms of core features? Surely, we will add more over time but I think we should keep things rather narrowly focussed and simple.

thomas-nicolai-frequenz, 2022-12-27:
> For resampling we might consider using the python wrapper around [libsamplerate](https://pypi.org/project/samplerate/)

@matthias-wende-frequenz can you please elaborate on that? How is that different to our own Resampler @leandro-lucarella-frequenz has been working on? The idea was to reuse our own Resampler but it seems you're suggesting a different approach now?

`LogicalMeter - Resampler (Pre) -> StreamBucketOperator (RingBuffer) -> Resampler (Post) -> ...`

matthias-wende-frequenz, 2022-12-29:
> Shall this be called RingBuffer or [CircularBuffer](https://en.wikipedia.org/wiki/Circular_buffer)?
That's the same data structure ;).

> @matthias-wende-frequenz anything I'm missing in terms of core features? Surely, we will add more over time but I think we should keep things rather narrowly focussed and simple.

No think these are the core features.

> What about something more simple like StreamBucket? I think the term Stream should be certainly part of the name. Not sure if window is the right term when thinking about a stream of samples. There [...]

I choose term the `Window` because it's been in use since many years in research of stream processing systems (c.f. [Cutty: Aggregate Sharing for User-Defined Windows](http://www.redaktion.tu-berlin.de/fileadmin/fg131/Publikation/Papers/Carbone_Cutty-2016_preprint.pdf). (Note that the oldest paper where I've seen is from 2003). There are different versions of Windows and since the one we are using gets updated frequently I choose Sliding as the prefix. We could consider leaving the `Operator` out of the name and make it implicit, since it's indeed an Operator that operates on a stream of data.
Btw. the window term is also used more abstractly, e.g. in time window, as something that gives a view on a certain detail of something and in that sense the window one can look through also shows a part of the landscape ;).

> Of course there could be questions around how and why the StreamBucketOperator aka SlidingWindowOperator is different from the RingBuffer but the RingBuffer should be used in many places in different way thus we should decouple those two things imho but we can discuss that if needed.

The main difference is that the `SlidingWindowOperator` acts on events and updates the underlying data structure, which in this case is supposed to be a ring buffer (but could also be another data structure). In that sense it's an independent module and deserves it's own name.

matthias-wende-frequenz, 2022-12-29:
> @matthias-wende-frequenz can you please elaborate on that? How is that different to our own Resampler @leandro-lucarella-frequenz has been working on?

[libsamplerate](http://libsndfile.github.io/libsamplerate/) is an optimized library, written in C, for resampling audio data. It supports up and downsampling and is one of the best resamplers around. At the time when luca started writing the resampler we simply didn't think of using this library until I stumbled upon the python bindings recently.

The idea is that our data is not much different from audio data, as both signals are voltages or currents sampled from an ADC and thus it's likely possible to use libsamplerate.

> The idea was to reuse our own Resampler but it seems you're suggesting a different approach? 

The idea is still the same at the end we want to reuse the same `Resampler`. So this comment can be considered as, questioning the status quo and explore alternative approaches.

thomas-nicolai-frequenz, 2022-12-30:
> libsamplerate is an optimized library, written in C, for resampling audio data. It supports up and downsampling and is one of the best resamplers around.

OK, this is outside the scope of this issue. @matthias-wende-frequenz please create a seperate issue for this so we can investigate that upon @leandro-lucarella-frequenz return.

thomas-nicolai-frequenz, 2022-12-30:
> We also want to resample an array. Passing the array into an AsyncIterator is certainly possible but has surely bad performance.

In what cases would we want to resample an array? Is there a specific case for that right now? If so please also create a seperate issue for that as this is also outside the scope of this issue.

matthias-wende-frequenz, 2022-12-30:
> In what cases would we want to resample an array? Is there a specific case for that right now? If so please also create a seperate issue for that as well this is also outside the scope of this issue.

The use case is when we need the same data at least two times with different sample rate. We have this case e.g. in our PV optimization app, with two different forecasts.
We could either create two pipes and create then two windows that are storing the same data with different sample rates or we store the data with the higher rate and then have the ability to resample sub windows.

matthias-wende-frequenz, 2022-12-30:
> OK, this is outside the scope of this issue. @matthias-wende-frequenz please create a seperate issue for this so we can investigate that upon @leandro-lucarella-frequenz return.

New issue for further discussion -> https://github.com/frequenz-floss/frequenz-sdk-python/issues/143

thomas-nicolai-frequenz, 2022-12-30:
> We could either create two pipes and create then two windows that are storing the same data with different sample rates or we store the data with the higher rate and then have the ability to resample sub windows.

The idea was always to have two pipes. Otherwise if there is multiple processes each of them might end up doing the same array resampling all over again. Resampling arrays feels like batch processing all over again and it would impact any downstream process that would also just work in a batched like manner.

thomas-nicolai-frequenz, 2022-12-30:
> or we store the data with the higher rate and then have the ability to resample sub windows.

Please also see the concept below where the orange line indicates that the high res 1s resampled data is consumed directly by the use case actor. So there is no need to store high res data in any buffer.

[image]

matthias-wende-frequenz, 2023-01-03:
I changed the batch resampling to an optional requirement and we can add it later if required or requested.

tiyash-basu-frequenz, 2023-01-09:
Looking at the example in the description in https://github.com/frequenz-floss/frequenz-sdk-python/issues/120, I think we need to address how we define the term `distance`.

According to the current definition (example_1):
```
timeseries: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]

window_size: 3
distance: 1

window_1: [1, 2, 3]
window_2: [5, 6, 10]
```

However, this is a bit confusing in the context of sliding windows. Conventionally, in sliding windows, samples are repeated in subsequent windows, e.g. (example_2):
```
timeseries: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]

window_size: 3
next_window_start_offset: 1

window_1: [1, 2, 3]
window_2: [2, 3, 4]
window_3: [3, 4, 5]
window_4: [4, 5, 6]
...
```

So, what is the intended behaviour here (example_1, or example_2)?

matthias-wende-frequenz, 2023-01-09:
> Looking at the example in the description in https://github.com/frequenz-floss/frequenz-sdk-python/issues/120, I think we need to address how we define the term distance.

In that module the term distance doesn't exist at all. This is specific to the average builder.

> However, this is a bit confusing in the context of sliding windows. Conventionally, in sliding windows, samples are repeated in subsequent windows, e.g. (example_2):

You are right, we are not fulfilling the "classic" definition of different kinds of windows.
In that sense neither sliding nor tumbling fits. What distinguishes our approach is that our window is not an once created immutable window, but get's constantly updated.
We are archiving that by using a ring buffer as underlying data structure and the main motivation behind this decision is efficient memory usage, without the need to allocate, copy and free repeatedly. The later approach might work efficiently in cloud systems where horizontal scaling is possible but we are also aiming for low cost and low powered systems.

`MovingWindow` might be a better name for that module.

thomas-nicolai-frequenz, 2023-01-09:
This discussion reminds me of something: https://pandas.pydata.org/docs/reference/api/pandas.DataFrame.rolling.html

> Size of the moving window. If an integer, the fixed number of observations used for each window.

Are you sure `sliding` is better then `moving`? If pandas introduces this term already we should stick to that as close as possible imho

thomas-nicolai-frequenz, 2023-01-09:
> I think we need to address how we define the term distance

`step` as pandas is using it?

ela-kotulska-frequenz, 2023-01-09:
I am wondering about this point in the description:
* can create an ordered copy of a slice of the ring buffer as numpy array

Consider actor that needs: Every 5 seconds last 1 hour data resampled every 1 sec.
Would that feature copy whole 1 hour array every 5 seconds?
Or would it return only the latest 5 seconds data to be appended to the recently received data?

matthias-wende-frequenz, 2023-01-10:
>> I think we need to address how we define the term distance

> step as pandas is using it?

That's not needed as mentioned [here](https://github.com/frequenz-floss/frequenz-sdk-python/issues/113#issuecomment-1375583627).

matthias-wende-frequenz, 2023-01-10:
> I am wondering how the data would be returned to the user.
> If I understand correctly it will copy the ringbuffer (to not override it by mistake).
> 
> If yes, then for example:
> Consider actor that every 5 seconds needs last 1 hour data resampled every 1 sec.
> Would it copy whole 1 hour array every 5 seconds?
> Or would it return only the latest 5 seconds data to be appended to the previously received data?

That's indeed a very good question.

The ringbuffer is designed to copy only if

* The requested time period is crossing the start/end of the buffer.
* The requested time period contains missing entries.
* The force_copy parameter was set to True (default False)

As we try to avoid coping as much as possible I think we should inherit from the default ring buffer behavior.

idlir-shkurti-frequenz, 2023-01-10:
> What distinguishes our approach is that our window is not an once created immutable series of windows, as in your second example, but instead is one window that gets constantly updated.

Just to make sure I understand this properly. It seems to me that the second example is a bit more general if we need the classic sliding window functionality. I understand that if we want these windows to represent things like days or weeks then the windows will not overlap but isn't it better to generalise this by applying the second example?

idlir-shkurti-frequenz, 2023-01-10:
> The ringbuffer is designed to copy only if

By copy do you mean load the entire window again instead of appending the ringbuffer? If so then I agree with the bullet points.

mathias-baumann-frequenz, 2023-01-23:
> By copy do you mean load the entire window again instead of appending the ringbuffer? 

There is no "appending", only overwriting already allocated memory and possibly old out-of-date data.

You can request any datetime range from the ringbuffer `buffer.window(start, end)` and the returned array/list will be a view to the actual raw buffer except for the cases @matthias-wende-frequenz outlined above. 
In those cases it will be a newly allocated buffer.

thomas-nicolai-frequenz, 2023-01-24:
> There is no "appending"

@idlir-shkurti-frequenz this is the main reason for the ring buffer. its has a fixed sizes and thus there is no such things as appending. appending always means new memory allocation and that in turn may invoke the garbage collector.

> In those cases it will be a newly allocated buffer.

Please try to use that feature carefully. There has to be a really good reason using that.

matthias-wende-frequenz, 2023-01-25:
> Just to make sure I understand this properly. It seems to me that the second example is a bit more general if we need the classic sliding window functionality. I understand that if we want these [...]

There is no overlap as it's just one window that constantly moves with the data flow. So it'll always have the latest data and with each new incoming data point the oldest data point is removed.

There has been a confusion between the `TimeseriesAvg` and the `MovingWindow`, that is sketched here so please forget the two examples Tiyash gave.

matthias-wende-frequenz, 2023-01-25:
>> In those cases it will be a newly allocated buffer.

> Please try to use that feature carefully. There has to be a really good reason using that.

The underlying ringbuffer already minimizes coping to the cases described above.
Additionally I'll add examples howto arrange the buffer when initializing to further minimize the likelihood of copying.

closed 2023-02-08 by matthias-wende-frequenz