frequenz-floss/frequenz-sdk-python #118: Add modules for each component component in frequenz.sdk.microgrid

ela-kotulska-frequenz, 2022-12-14:
### What's needed?

Now in frequenz.sdk.microgrid we have modules like: *component_errors*, *component_states*, *component_type*.

Instead of we could have module for each component category. The same as in frequenz-api
It would:
* Simplify findings features of the component. Now it is not clear what is where.
* Instead of implementing class BatteryErrorCode it would be possible to implement class ErrorCode in battery submodule and import it as battery.ErrorCode
* It would be more consistent with the api.

### Use cases

Easier to implement attributes of the component message.
Easier for user to find correct type.
Easier to import code.

llucax, 2023-08-10:
Blocked on:

* #575

llucax, 2024-07-29:
This is very outdated, now the client has its own repo and the structure will be changed to migrate to v0.17 anyways.

closed 2024-07-29 by llucax