frequenz-floss/frequenz-sdk-python #124: Simple forecast pipeline example

matthias-wende-frequenz, 2022-12-19:
### What's needed?

Write a simple example that shows how sdk internal
tools can be used to create a simple forecasting pipeline.

A forecasting pipeline is a data pipeline with the goal
to transform data, such that it can be directly passed
into a ml models prediction function.

As starting point of the pipeline we define a logical meter, that
means the data itself has already been processed, i.e. re-sampled and
possible combined from other sources into the desired source.

### Proposed solution

For now our anticipated features are single feature values, derived from
time series data, such as current, voltage or power, as well as a bounded
subset of certain time series, again we can think of current, voltage or power.
Note, that the logical meter presents us an unbounded time series.
However one could also think of price data for price predictions just to name
a further example of input data.

## Example

In this example we sketch how the average current of a certain weekday
can be obtained.

```mermaid
graph LR;
    LogicalMeter-->DataBatcher;
    DataBatcher-->TimeseriesAvg;
    TimeseriesAvg-->Mean;
    Mean-->predict();
```

As a first step, we can create a bounded subset by using a
[data batcher](https://github.com/frequenz-floss/frequenz-sdk-python/issues/113).
This data batch can then used to create an averaged weekly profile for current data using
the [`TimeseriesAvg`](https://github.com/frequenz-floss/frequenz-sdk-python/issues/120)
and as the last step calculate the mean of this averaged load profile. As a result
that gives as the average current that flows through the observed point on a specific weekday.

The code might look as follows.

```python
# Create an instance of the logical meter and aquire the first phase of the current
# note that formulas for power and current are the same
lm = LogicalMeter(channel_registry, resampler_subscription_sender)
current_ph1_rx = lm.start_formula(consumption_load_formula, ComponentMetricId.CURRENT_PHASE_1)

# update the ring buffer with the latest streamed data from the microgrid
data_batcher = DataBatcher(size = n_days * 3600 * 24, current_ph1_rx)

# get an averaged day of the last n weeks with start day yesterday
time_window = 3600 * 24 # one day of data with sample rate 1Hz
window_distance = 3600 * 24 * 6 # 6 days distance between the days
offset = 3600 * 24 + no_seconds_today # start day is yesterday

ts_avg = TimeseriesAvg(
    time_window_s = time_window,
    window_distance_s = window_distance,
    offset_s = offset
)

# iterate over the timeseries and calculate the mean
mean = ts_avg.mean()
```

david-natingga-frequenz, 2023-01-04:
Please, note that also frequency data is needed for forecasts relevant to FCR actor.

matthias-wende-frequenz, 2023-01-04:
> Please, note that also frequency data is needed for forecasts relevant to FCR actor.

This is supported out-of-the-box. We can construct such forecast pipelines with everything that can be provided by the `LogicalMeter`.