frequenz-floss/frequenz-sdk-python #132: Handle data from disk

matthias-wende-frequenz, 2022-12-23:
### What's needed?

We want an simple and efficient way to handle dumping and loading data from disk.

### Proposed solution

Add dump the full buffer to disk and reload the buffer from disk to the (ring buffer)[https://github.com/frequenz-floss/frequenz-sdk-python/pull/116].

### Use cases

Let's consider an application that creates one feature from timeseries data. Usually we would think of creating a buffer, e.g. using a (ring buffer)[https://github.com/frequenz-floss/frequenz-sdk-python/issues/113], that stores data for a certain time window, e.g. 15 minutes or 28 days.

When the application restarts, if there is no data on disk, it would take the time of the window size to rebuild this buffer and hence create this feature.

mathias-baumann-frequenz, 2023-01-26:
A question in relation to this: Should we check for problems like full disk? I know from past experience that this this type of error can cause all kinds of strange behavior if not handled properly

matthias-wende-frequenz, 2023-01-27:
I think that's a good idea :)

mathias-baumann-frequenz, 2023-01-27:
Next question is then, should the class check that or something external to it

closed 2023-02-14 by mathias-baumann-frequenz