frequenz-floss/frequenz-sdk-python #140: Implement BatteryPool

ela-kotulska-frequenz, 2022-12-29:
### What's needed?

User should be able to subscribe for data from some pool of Batteries.

## What data are needed:
* average SoC 
* sum of SoP  - charge/discharge power bounds (from batteries and adjacent inverters)
* sum of capacity
* min/max of SoC bounds

## What should not be in the batteryPool:
* active power - user should use Logical meter to get active power from batteries

### Proposed solution

BatteryPool will use:
* BatteryPoolStatus - to know what batteries are working
* ComponentMetricResamplingActor - to subscribe for the metric data and receive data with some frequency. This frequency should be similar or the same to logical meter frequency. 
    *  Interpolation methods for the resampler:
          * Capacity, SoC bounds : Return last value -  This goes down very very slowly so returning last value seems to be correct.
          * SoP, SoC - linear interpolation based on first and last value in Resampler. Both SoC and SoP don't change the direction frequently. They either  increases  or decreases for some time.

## Open questions ##
How long time window we should considered to store in order to interpolate SoC/SoP?

thomas-nicolai-frequenz, 2022-12-29:
> min/max of SoC bounds (maybe)

I think we should add this right away. Peak shaving I believe needs  to know the bounds @idlir-shkurti-frequenz 

I would also suggest to provide the **min** and **max** SoP bounds. @tiyash-basu-frequenz The min bound will be a new upcoming feature by the Microgrid API that the `PowerManager` will need to support. Peak shaving shaving has already been tested with that feature by @idlir-shkurti-frequenz . @tiyash-basu-frequenz can shed more light onto that @ela-kotulska-frequenz Its quite an important feature we still need to test before the winter ends ;-) Its needed by the DC heating core app, however the SDK will have to support these min and max SoP bounds.

thomas-nicolai-frequenz, 2022-12-29:
@ela-kotulska-frequenz when touching the BatteryPool I think we should start working / designing the PowerManager right away so that users would not directly need to be dealing with SoC and SoP bounds as well as talking care about the PowerDistributor. Of course up to you to decide. Just throwing that in as a point of discussion.

thomas-nicolai-frequenz, 2022-12-29:
> Should active_power be returned from BatteryPool or LogicalMeter.

Depending on whether the metric changes if a battery fails is one way to decide. I would tend to say to place it in the BatteryPool for consistency and/or offer both options. If a user wants to include the ID of a battery in a LogicMeter formula there is nothing we can prevent the user of doing so. Thats why I'd say probably giving the a consistent option via the BatteryPool might not hurt even if it feels a bit redundant. However, key point is that the LogicalMeter does not know when a battery goes offline and the battery needs to be removed from the calculation. I think this logic should always stay inside the BatteryPool. Just my few thoughts about it .... This discussion will also come back with EvChargerPool or the InverterPool etc.

ela-kotulska-frequenz, 2023-01-04:
@thomas-nicolai-frequenz  
I discussed min/max bounds with Tiyash. I also need to add it to PowerDistributingActor.

I will focus on BatteryStatus first. The next important task would be to implement min bounds in PowerDistributingActor.
I will start PowerManager after that.

thomas-nicolai-frequenz, 2023-01-04:
> I discussed min/max bounds with Tiyash. I also need to add it to PowerDistributingActor. I will focus on BatteryStatus first. The next important task would be to implement min bounds in PowerDistributingActor. I will start PowerManager after that.

Indeed, that makes a lot of sense. Thank you very much! I simply missed that, sorry for that. Good catch!

idlir-shkurti-frequenz, 2023-01-15:
> max_soc: sum(battery.capacity*battery.max_soc for each working battery in battery pool)/total_capacity

What exactly is the `battery.max_soc` in this formula?

thomas-nicolai-frequenz, 2023-01-16:
> User should be able to subscribe for data from some pool of Batteries.

I struggled when I read the word "some" but it totally makes sense. A user might want to split up the entire pool of batteries in any given microgrid into two pools. That means a user should be able to:
1. Provide a list of battery ids that belong to a pool
2. Provide no list and the pool will by default use all available batteries

closed 2023-03-09 by ela-kotulska-frequenz