frequenz-floss/frequenz-sdk-python #148: Move to python3.10

ela-kotulska-frequenz, 2023-01-03:
### What's needed?

In README there is sentence:
`For arm64 only Python 3.8 is supported (due to some dependencies that only support 3.8).`

These dependencies was an xgboost package that I've build locally, since we wanted to have gpu support for our nvidia jetson xaviar nx edge controllers
Actually these dependencies were never used anywhere.
It is not build and needed now. 

### Proposed solution

We should drop this  until we have more time or the gpu priority increases.

Then we could move to python 3.10.
Python 3.10 is the latest python supported by tensorflow.

christianparpart, 2023-01-03:
Actually, locally, on all my Linux systems, I am having Python 3.11 installed. Despite a few issues today I had in getting this repo built, I managed to succeed, using Python 3.11. Maybe we can run the tests against 3.10 as well as 3.11 and support both?

shsms, 2023-01-04:
Oh,  I thought we were already supporting python 3.11.  I guess we need to add that to the CI first, then.

christianparpart, 2023-01-06:
> Oh, I thought we were already supporting python 3.11. I guess we need to add that to the CI first, then.

NB: It's on CI now.

leandro-lucarella-frequenz, 2023-02-07:
I renamed this issue because I think it was not about supporting Python 3.10 (which was already supported) but to make it the minimum supported Python version (so we can use more modern features in the SDK itself). I also updated it to Python 3.11, as eventually Tensorflow should support it.

Still, I would be very careful about doing this move, as once it is done there is no (easy) way back. Things I would have in mind before doing this:

* Check we'll really never, ever need to use a compiled package that it is only supported in Python 3.8 (either because we won't use the package or because it can be compiled with Python 3.10 / 3.11 too).

* Check the current supported version in the latest Ubuntu LTS (and probably other popular distros). This is a nice to have, but I think making it work out of the box in a Ubuntu LTS is a nice *feature*. If we decide against this, we have to be conscious people might need to compile their own Python to use the SDK.

I also don't think we need to do this in v0.18.0, so we might consider moving it to a future milestone.

leandro-lucarella-frequenz, 2023-03-06:
I updated this issue again, adding current blockers to move to Python 3.11. I removed moving to Python 3.10 as an option because I think is too much effort to move to 3.10 first and shortly after to 3.11 again (I assume the blockers will be lifted sooner than later, but we can reconsider if that doesn't end up happening).

leandro-lucarella-frequenz, 2023-03-15:
The current tensorflow is compatible with Python3.11, at least it is installable in Python 3.11. It seems like some issues remain with supporting CUDA.

Can we do some testing to see if it is enough for what we need? It would be awesome to make the jump to 3.11, also for all the background task stuff having task groups and exception groups would make everything much easier.

leandro-lucarella-frequenz, 2023-04-03:
Now the issue in tensorflow is closed, there is one comment saying cuda is supported. Someone complains there are no binary wheels for mac M1, but I guess that shouldn't be a stopper for us.

leandro-lucarella-frequenz, 2023-04-03:
I added a link a web that shows major libraries compatibility with 3.11. It's not perfect because then a project specifies an open ended python version support (like we do), it's not 100% sure it really works, but in general it should: https://observablehq.com/d/eb336fc983fcdf52

commit 0408bd8c (2023-05-23): Drop support for Python versions older than 3.11 (#401)

- Reformat `pyproject.toml` using `taplo`
- Drop support for Python versions older than 3.11
- Improve typing annotations
- Remove Sample comparison hack

Fixes #148, fixes #399.
PR #401 (2023-05-23): Drop support for Python versions older than 3.11
- Reformat `pyproject.toml` using `taplo`
- Drop support for Python versions older than 3.11
- Improve typing annotations
- Remove Sample comparison hack

Fixes #148, fixes #399.
closed 2023-05-23 by leandro-lucarella-frequenz