frequenz-floss/frequenz-sdk-python #152: PowerDistributingActor should use min SoP

ela-kotulska-frequenz, 2023-01-04:
### What's needed?

Microgrid will expose min bounds for charge/discharge power soon.
After that we should update PowerDistributing algorithm to look at it.

### Proposed solution

Implementation should be in old MOS repository, because in past we had this min bounds.
 To be checked

leandro-lucarella-frequenz, 2023-04-21:
Blocking because we need a new version of the microgrid API service to be released for being able to merge this.

thomas-nicolai-frequenz, 2023-07-26:
@sahas-subramanian-frequenz the min and max SoP bounds should be first exposed via the BatteryPool. Second step would also be to look at the bounds if someones sends a power request thats out of bounds.

commit 6258c76e (2023-08-09): Support exclusion bounds in power distributor (#562)

To be merged after #537

Closes #152
PR #562 (2023-08-09): Support exclusion bounds in power distributor
To be merged after #537 

Closes #152
closed 2023-08-09 by llucax