frequenz-floss/frequenz-sdk-python #161: PowerManager

ela-kotulska-frequenz, 2023-01-10:
### What's needed?

PowerManager is a tool for controlling micro-grid from the sdk side.

### Type of requests:
* Charge/Discharge batteries
* **Set bounds for batteries**
* Charge/Discharge EvChargers
* Set bounds for EvChargers (Power can only be adjusted if it has been invoked by a car, however this might change in the future with hybrid charging and is part of the VW project)
* Charge/Discharge ChPools (Discharge only for natural gas power plants but also charge with hydrogen electrolyzers which will also be something upcoming)

### Request structure:
```python
class Priority:
    Normal = 0
    High = 1
    

class Request:
    duration_sec: float  # how long this request would block other requests. No other requests 

class Lock(Request):
    actor_id # id of the actor that asks for lock.
    request_type # For which type of request this lock is assumed.

class BatteryPower(Request):
  * priority: Priority
  * batteries_ids :  Set[int]  # For what batteries we need to set power
  * power_w: float #  How much power to set (total power to be distributed and set between all batteries)
  * adjust_power # Adjust given power if outside or below the SoP bounds
```

### Requirements:
* Exclusive lock, separate for each type of requests above.
     * The lock should be for some `duration time`, not forever
     * No two actors can have exclusive lock for the same type of request. 
     * Other actors when trying to engage might get a response telling that them how long the lock is still to last. The lock would always require a duration to be given.
     * If above is correct, then we have to distinguish who (what actor) sends the request.
* Each request should has priority. Priority defines how important is that request. Requests with highest priority will be executed first. 
    * Should we consider case like `starving`? When requests with bigger priority will be executed first and the `less important` requests won't be executed at all. Actor will wait for response for `less important` request forever.
* New request of the same type shouldn't be executed unless previous request is finished.
* Track stream of requests to detect any invalid operations like:
      * Actor is sending charge 0, discharge 5000 over and over in the loop. Or two actors can do it by overriding one another?
      * Charging and discharging with high values over and over

### Cases:
* Two lock requests arrived for the same request type:
    * We should execute first request and ignore older requests
* Two requests of the same type arrived, for the same set of components. What should happen:
    * We should execute only the latest (newest) request and ignore the older request.
* Actor would like to charge/discharge but the bounds has been set to 0 by previous actor (Check if there is a need to set bounds to the power from sdk side).
* Should we override executed request with the request with higher priority?
     * We set the ev bounds to be 0 for 30 sec - the request priority NORMAL
     * 5 seconds later there comes a request to set ev bounds to 1000 - the request priority HIGH.

### Proposed solution

TBD

matthias-wende-frequenz, 2023-02-17:
> class BatteryPowerRequest:
>     priority: float # Actors priority
>     batteries: Set[int] # What batteries should be charged/discharged
>     power: float # Total power recommended by actor to be distributed between given batteries. Negative value would discharge batteries, positive value would charge them.This value should be between min_power and max_power.
>     min_power: float # min power bound (described below)
>     max_power: float # max power bound (described below)
>     duration_sec: float # how long the request should stay.

I suggest to use the type `datetime.timedelta` for `duration_sec` and rename this field to `duration` as Marenz did in the ringbuffer.

thomas-nicolai-frequenz, 2023-03-28:
I suggest to design something that allows plugging in different consensus building algorithms. It would be great if we could abstract it in a way that it would allow to replace the `matrjoschka / russian doll algorithm` with something else like a voting-based mechanism.

thomas-nicolai-frequenz, 2023-03-29:
> In this example, actors rate each option for charging power using integers between -100 and 100, with negative values indicating a desire to discharge the battery. As before, ratings are multiplied by the actor's weight to give actors with more influence more impact on the decision. The average rating for each option is calculated by taking the weighted average of the ratings provided by each actor. Finally, the option with the highest average rating is chosen as the winner.

Down below is an example of what a voting based-approach could look like. In this example multiple actor negotiate about the (dis)charging power of a battery.

```Python
options = ['-100kW', '-50kW', '0kW', '50kW', '100kW']
num_actors = 3

# Initialize a dictionary to store the ratings for each option
ratings = {option: [] for option in options}

# Assign weights to each actor
weights = {1: 1, 2: 2, 3: 1}

# Prompt each actor to rate each option
for i in range(num_actors):
    print("Actor", i+1, "please rate each option for charging power (-100kW to 100kW):")
    for option in options:
        rating = int(input(option + ": "))
        # Multiply the rating by the actor's weight
        rating *= weights[i+1]
        if i == 2: # if actor is discharging, then invert the rating
            rating *= -1
        ratings[option].append(rating)

# Calculate the average rating for each option
total_weight = sum(weights.values())
averages = {}
for option in options:
    # Compute the weighted average of the ratings for each option
    rating_avg = sum(ratings[option]) / total_weight
    averages[option] = rating_avg

# Find the option with the highest average rating
winner = max(averages, key=averages.get)

print("The chosen option is:", winner)
```

thomas-nicolai-frequenz, 2023-03-29:
What I would like to achieve is to create an interface for the PowerManager that allows both approaches to work; the russian doll priority-based approach as well as the voting based one.

llucax, 2023-08-10:
Related issue:

* https://github.com/frequenz-floss/frequenz-sdk-python/issues/293

closed 2023-10-13 by llucax