frequenz-floss/frequenz-sdk-python #165: Investigate using Sample type in Ringbuffer

matthias-wende-frequenz, 2023-01-18:
### What's needed?

The `OrderedRingBuffer`s update method takes a timestamp and a value as parameters.
The default use case for this class is to store data coming from the logical meter, which sends data of type `Sample` which members are a timestamp and a value.

### Proposed solution

Move the `Sample` class from the `timestamp` path to e.g. `util` and use it in the RingBuffer.

shsms, 2023-01-18:
> Move the Sample class from the timestamp path to e.g. util and use it in the RingBuffer.

why do we have to move it?  you could just get it with: `from timeseries import Sample`

matthias-wende-frequenz, 2023-01-19:
> why do we have to move it? you could just get it with: from timeseries import Sample

I think there is need to move it out any longer.

My concern was that when it's used outside of `timeseries` it becomes a more generic type and thus deserves it's place outside of `timeseries`.

But since we decided to move the ringbuffer to timeseries, this concern isn't valid any more.

PR #186 (2023-02-01): Use `Sample` type with the RingBuffer
fixes #165
commit 30e3a730 (2023-02-01): Use `Sample` type with the RingBuffer (#186)

fixes #165
closed 2023-02-01 by mathias-baumann-frequenz