frequenz-floss/frequenz-sdk-python #182: There is a lack of clear standards and documentation on actor development

david-natingga-frequenz, 2023-01-26:
### What happened?

I wanted to add nox checks to my FCR actor and I combined the nox file from peak shaving actor and isort feature from EVC actor. And this failed because the isort rule is inconsistent with the black rules.
- black wants:
```
from frequenz.actor.fcr.charge_discharge_management import (
    get_charge_discharge_management,
)
```
- isort wants:
```
from frequenz.actor.fcr.charge_discharge_management import \
    get_charge_discharge_management
```

### What did you expect instead?

Instead I would expect there is a well-documented standard for actors including the nox configuration that for any actor can be just copied and trusted and it will work straight away.

### Affected version(s)

All versions

### Affected part(s)

Documentation (part:docs), Build script, CI, dependencies, etc. (part:tooling)

### Extra information

We have to automate this and it should be part of every release. Otherwise we cannot expect that the actors would follow good standard if any of the following is true:
- the standards or documentation on them does not exist,
- there is no mechanism or process to keep the standards and their documentation updated,
- the standards work only partially or for some actors only.

I think the problem is **having good standards** seems not a requirement, but this is in contradiction with requiring that **actors follow good standards**.

david-natingga-frequenz, 2023-01-26:
It is important that these standards are clear, documented, up-to-date, and _written down in a single location_ - not a word of mouth or a meddle of things from across multiple repositories.

leandro-lucarella-frequenz, 2023-04-14:
The new repo https://github.com/frequenz-floss/frequenz-repo-config-python should take over as the single source of truth for this.

About your particular problem, yes, `black`, `pylint` and `isort` have some conflicts, we have some configuration in the `pyproject.toml` file in the SDK to solve this:

```toml

[tool.pylint.messages_control]
disable = [
    # ...
    # disabled because it conflicts with isort
    "wrong-import-order",
    "ungrouped-imports"
]

[tool.isort]
profile = "black"
line_length = 88
src_paths = ["src", "examples", "tests"]
```