frequenz-floss/frequenz-sdk-python #188: Use concrete types for power, energy, time and where-ever it's suitable

mathias-baumann-frequenz, 2023-01-31:
### What's needed?

Currently in most places where we expect seconds it is implicitly in the documentation or variable name, e.g. `sampling_rate_s` for seconds.

### Proposed solution

I propose we use for cases like this the native python type `timedelta` which can be initialized with any time unit (years, months, days, seconds, etc) and internally is represented by microseconds. It would be used for example like this: `timedelta(seconds=3, milliseconds=21, microseconds=20)`

Similar, for places where we expect kilowatthours, we could create a new type `energy` that we would use like this `energy(kwh=30, wh=100, mwh=3)` (all but one parameter are optional)

Maybe further types, depending on what we use and need...

This can further be enhanced adding operations between the types (so that they make sense and result in the correct units)

leandro-lucarella-frequenz, 2023-02-03:
Related (past) discussion and proposal:
* https://github.com/frequenz-floss/frequenz-sdk-python/discussions/6

llucax, 2023-08-10:
There was a lot of progress about this, I think this is taken care of in most places, but we should check if there are any places where this is missing.

commit 1fe5cb8c (2023-08-28): Update high-level public interfaces to concrete types (#607)

Fixes #188
PR #607 (2023-08-28): Update high-level public interfaces to concrete types
Fixes #188
closed 2023-08-28 by daniel-zullo-frequenz