frequenz-floss/frequenz-sdk-python #196: Don't start actor automatically in the __init__()

leandro-lucarella-frequenz, 2023-02-08:
### What's needed?

Probably all applications will need to start some actors and then wait for them to finish indefinitely. So far this is on the SDK users, but it could be handled by SDK. Also this allows us to find better methods to wait for all actors (or tasks) to finish.

In this case the SDK can also provide sanity checks, like the `microgrid_api` is properly initialized, before starting all the actors, or even do the initialization itself if we can provide the run function some config.

This issue is complementary to:
* #36

### Proposed solution

Stop auto-starting actors in the init, and provide a `start()` method instead. Then we can have an even more simplified run function:

Taking this more to the extreme, if we get all the initialization parameters from
This makes me think we should probably make an `initialize()` function at the `sdk` level too).

```py
def run(config: Config, *actors: Actor) -> None:
    async def _run() -> None:
        await microgrid_api.initialize(config.microgrid.host, config.microgrid.port)
        await asyncio.gather(actor.start() for actor in actors)
        await asyncio.gather(actor.join() for actor in actors)
    asyncio.run(_run())
```

Then clients will be like:

```python
from frequenz import sdk

config = sdk.Config(...)

def main() -> None:
    actor1 = Actor1()
    actor2 = Actor2()
    actor3 = Actor3()

    sdk.run(config, actor1, actor2, actor3)

if __name__ == '__main__':
    main()
```

We should probably allow the user to initialize the `microgrid_api` itself, in case they need to do some extra config using the `micrgrid_api` before calling the `run` method, so at the end we can also provide smaller, more specific functions, that the user can use, and a very simple `run()` that should cover 90% of the use cases.

### Use cases

We have a few scenarios where not making the actors auto-start could lead to better patterns, in particular actors could own their own channels without the risk of circular dependencies, or initialization order issues.

### Alternatives and workarounds

For the auto-start dependencies issue, we are just creating all channels outside of the actors first, and then passing the channels to the actors.

shsms, 2023-02-08:
>     asyncio.run(_run())

Tiny implementation note:  `asyncio.run()` would fail if there is an event loop running already,  which there will be when we start the data pipeline, for example.

Instead we should check if there is an existing loop `asyncio.get_running_loop()` or `asyncio.create_new_loop` and then `loop.run_until_complete()`.

leandro-lucarella-frequenz, 2023-02-08:
Ah, damn, I forgot to update the proposal, I think the `run()` should be async and the user should call `asyncio.run()` explicitly, so we don't mess with creating the loop ourselves. Is a minor annoyance for the user having to call `asyncio.run()` explicitly but it makes the flow and initialization much clearer.

I'll update the proposal now.

christianparpart, 2023-03-15:
@leandro-lucarella-frequenz I think it would also be really helpful to allow an actor to **orderly** shut down. So some `Actor.shutdown()` method would be really handy to make actors fully testable (I need that in my own actor I am working on currently, and I want to make that actor fully automatic-testable on CI).

leandro-lucarella-frequenz, 2023-03-16:
Yeah, that's the idea, they will have s `stop()` which by default will just cancel the tasks, but you could override it to do something else like sending some stop event if you run method is basically an event loop.

leandro-lucarella-frequenz, 2023-04-11:
This should be fixed by making `Actor` a [`BackgroundService`](https://github.com/frequenz-floss/frequenz-sdk-python/issues/240). Blocking for now until the `BackgroundService` is implemented.

llucax, 2023-08-10:
This is being worked on as part of #240.

closed 2023-08-25 by llucax