frequenz-floss/frequenz-sdk-python #210: Add an EVChargerPool

shsms, 2023-02-19:
### What's needed?

`EVChargerPool` instances enable high-level interactions with ev chargers.  

### First version
In their most basic form, they should provide resampled power or current streams for individual ev chargers, or aggregated streams covering all ev chargers in the pool,  and state change updates for the ev chargers in the pool.  They should also provide interfaces for limiting max charge power for individual ev chargers in the pool.

Instances would by default cover all ev chargers available in the component graph,  but they can also be instantiated with a subset of ev charger component ids,  and then they would provide state change updates only for the subset of component IDs,  and the total power and current values will be the total of the subset of components.

### EV Chargers as Batteries
After the first version is stabilized,  it will be extended with the same interface as the `BatteryPool`, to allow for the use of EV batteries directly.

### Public methods in the first version

- **states** :: returns a receiver that streams ev charger state change updates.
- **total_current** :: a receiver that streams resampled total 3-phase current of all ev chargers in the pool.
- **total_power** :: a receiver that streams resampled total power of all ev chargers in the pool.
- **current** :: takes a component id and returns a receiver that streams that ev charger's current values.
- **power** :: takes a component id and returns a receiver that streams that ev charger's power values.
- **set_bounds** :: takes a component id and new current bounds, and forwards them to the microgrid.

shsms, 2023-02-19:
After #201 is merged,  there will be two pending items:
- [ ] updating of component states, on missing data.
- [ ] the set_bounds method.

sahas-subramanian-frequenz, 2023-04-14:
Completed in https://github.com/frequenz-floss/frequenz-sdk-python/pull/297 and https://github.com/frequenz-floss/frequenz-sdk-python/pull/266.