frequenz-floss/frequenz-sdk-python #212: Support Stream Moving Average

matthias-wende-frequenz, 2023-02-20:
### What's needed?

A weighted moving average implementation that updates the avg whenever a new data sample arrives from the stream.

### Proposed solution

We can use the [`MovingWindow`](https://github.com/frequenz-floss/frequenz-sdk-python/blob/v0.x.x/src/frequenz/sdk/timeseries/_moving_window.py) class. Whenever the `MovingWindow` gets updated it triggers the update of the mean.

Right now, the `MovingWindow` doesn't support sending events on update.

We propose two different implementation approaches.

* Add a output trigger channel that informs downstream actors about an update of the window.
* The moving average is implemented as part of the `MovingWindow` and downstream users can request an output channel.

The weights can be user defined by an iterator of weights. When no weight is provided the average is supposed to return the arithmetic average. We might also want to provide a couple of defaults, e.g. exponential.

**Example**

```python
window = MovingWindow(100, resampled_data_recv)
# use the latest three values for the weighted average
ma_recv = window.moving_avg_recv(weights=[0.2, 0.3, 0.5])
...
```

### Use cases

The main use case is feature extraction for ML applications.

matthias-wende-frequenz, 2023-03-21:
This can be done using numpy. The numpy average method accepts weights. In order to use it we need a wrapped array from Moving Windows underlying Ring Buffer, such that we can write `moving_window[-10:]` to address the e.g. the latest 10 entries.
If that can be done then we can use the latest `n` entries for calculating an weighted average as follows:

```
np.average(moving_window[-n:], weights)
```
where `weights` is an arbitrary array of length `n`.

matthias-wende-frequenz, 2023-07-14:
We need to make sure that the MW returns a sorted array! Add priority URGENT as its already used by @cwasicki.

llucax, 2023-08-08:
Blocked because we need to fix some bugs in the ring buffer for this to be useful.

cwasicki, 2023-08-24:
> We need to make sure that the MW returns a sorted array! Add priority URGENT as its already used by @cwasicki.

If the purpose of this issue is to ensure that the array returned by the moving window is sorted by timestamp I agree that this is important (for v1). This will be addressed by https://github.com/frequenz-floss/frequenz-sdk-python/issues/214.

The stream weighted average mentioned in the title and the description can be calculated with a sorted array on our side or could still be implemented in the SDK later if required, but I think the priority and milestones should be adjusted accordingly. @matthias-wende-frequenz  @llucax

llucax, 2023-08-24:
Leaving this for @matthias-wende-frequenz to answer, I'm too far away from the problem to have a meaningful opinion.