frequenz-floss/frequenz-sdk-python #213: Handle gaps in the `MovingWindow` data

matthias-wende-frequenz, 2023-02-21:
# Provide an interface for handling gaps

* the user should be able to tell what to do with gaps, e.g. replace with constant number, bfill ffill, interpolate
* should be able to get a list of all gaps

# Accessing behavior of uninitialized or missing data

We need to check if data is potentially uninitialized in the `__getitem__`
method and define what to do in that case.
The check can be performed by using the ringbuffers gaps. Thus we'd need to
check if the requested data subwindow intersects with any interval stored as gap.

## possible actions when uninitialized data is accessed:

* raise an exception
* return the biggest subset

Marenz, 2023-06-16:
Gaps here refers to the gaps found in the OrderedRingBuffer class and how they can be handled

matthias-wende-frequenz, 2023-10-10:
@cwasicki can we close this issue?

cwasicki, 2023-10-16:
Do you remember who requested these features? 

>the user should be able to tell what to do with gaps, e.g. replace with constant number, 

This is available now.

> bfill ffill, interpolate

This is not supported but could be done by the user on the array.

> should be able to get a list of all gaps

This request was just closed: https://github.com/frequenz-floss/frequenz-sdk-python/issues/558

matthias-wende-frequenz, 2023-10-17:
> Do you remember who requested these features?

I can't remember but might come from myself :). I'll close the issue now.

closed 2023-10-30 by llucax