frequenz-floss/frequenz-sdk-python #214: Create Interface for handling the data copy behaviour

matthias-wende-frequenz, 2023-02-21:
When asking the `MovingWindow` for a subwindow by indexing using a slice
we will return a slice of the buffer, that means a reference to the underlying
numpy based ringbuffer.

In some cases, e.g. when dealing with small window sizes and high sample rates,
the data might get updated from the data stream before the requested subwindow
has been processed.

As a solution we want to tell the `MovingWindow` to return a copy of the data
instead of a reference.
In order to do that we need to add an interface, such that users can control copy behavior.

Furthermore we add explicit documentation and examples on when data is copied and how to reduce the copying to a minimum.

matthias-wende-frequenz, 2023-07-21:
FYI @cwasicki @thomas-nicolai-frequenz

thomas-nicolai-frequenz, 2023-07-21:
I don't understand why we could not design it in a way where the user passes his numpy array by reference the moving window is writing to. This way we could avoid the need for the fill_gaps parameter. In that case the numpy array could be initialised with whatever values the someone wants.

cwasicki, 2023-07-21:
To me passing in an array feels less like typical behavior of other functions in numpy or pandas. 

> 	fill_gaps: Optional[Quantity] = None,

Since we should not have gaps in the moving window but only NaNs or uninitialized in the beginning, I would name this differently.

>start: datetime,
>end: datetime,

That should also work with index, not only datetime.

matthias-wende-frequenz, 2023-07-24:
> I don't understand why we could not design it in a way where the user passes his numpy array by reference the moving window is writing to. This way we could avoid the need for the fill_gaps parameter. In that case the numpy array could be initialised with whatever values the someone wants.

This is about the gaps that are coming from the resampler, e.g. due to faulty hardware.

Passing the np array is also planned but an independent issue (https://github.com/frequenz-floss/frequenz-sdk-python/issues/530).

thomas-nicolai-frequenz, 2023-07-24:
> This is about the gaps that are coming from the resampler, e.g. due to faulty hardware.

Yes but in this case we would copy the NaN's that are coming from the resampler.

matthias-wende-frequenz, 2023-07-26:
> Yes but in this case we would copy the NaN's that are coming from the resampler. This has to be treated differently I think but not 100% sure.

You are right, we shouldn't copy NaN's. If `fill_gaps` is set to `None` and there are gaps then we should raise an exception.

cwasicki, 2023-08-09:
> Blocked because we need to fix some bugs in the ring buffer for this to be useful.

@llucax Which bugs are those?

cwasicki, 2023-08-10:
> @llucax Which bugs are those?

Found it: https://github.com/frequenz-floss/frequenz-sdk-python/pull/578

llucax, 2023-08-10:
OK, if you need any further info about this, ask @matthias-wende-frequenz

thomas-nicolai-frequenz, 2023-08-22:
@leandro-lucarella-frequenz but this bug is fixed and merged. Can we unblock the issue?

llucax, 2023-08-22:
Done.

cwasicki, 2023-09-13:
@llucax As discussed summary of the (ongoing) work related to this and what I see essential for v1.0.  

Essential for this issue:
* For v1.0:
  * https://github.com/frequenz-floss/frequenz-sdk-python/pull/641
  * https://github.com/frequenz-floss/frequenz-sdk-python/pull/598
  * https://github.com/frequenz-floss/frequenz-sdk-python/pull/638
  * https://github.com/frequenz-floss/frequenz-sdk-python/pull/668
  * Support element access in moving window (not draftable yet)
  * Remove unsupported getitem and setitem magic depending on previous two PRs.
* Post v1.0?
  * https://github.com/frequenz-floss/frequenz-sdk-python/pull/669.
  * https://github.com/frequenz-floss/frequenz-sdk-python/issues/646

Related to moving window:
* For v1.0
  * https://github.com/frequenz-floss/frequenz-sdk-python/pull/664
* Post v1.0?
  * https://github.com/frequenz-floss/frequenz-sdk-python/issues/558

cwasicki, 2023-10-02:
Apart from a very special bug (#646) all associated work has been merged.

closed 2023-10-02 by cwasicki