frequenz-floss/frequenz-sdk-python #249: Some Formula tests are flaky

ela-kotulska-frequenz, 2023-03-10:
### What happened?

```
    async def test_3_phase_formulas(self, mocker: MockerFixture) -> None:
        """Test 3 phase formulas current formulas and their composition."""
        mockgrid = MockMicrogrid(grid_side_meter=False)
        mockgrid.add_batteries(3)
        mockgrid.add_ev_chargers(1)
        request_chan, channel_registry = await mockgrid.start(mocker)
        logical_meter = LogicalMeter(
            channel_registry,
            request_chan.new_sender(),
            microgrid.get().component_graph,
        )
        grid_current_recv = await logical_meter.grid_current()
        pool = EVChargerPool(channel_registry, request_chan.new_sender())
        ev_current_recv = await pool.total_current()
    
        engine = (grid_current_recv.clone() - ev_current_recv.clone()).build(
            "net_current"
        )
        net_current_recv = engine.new_receiver()
    
        await synchronize_receivers(
            [grid_current_recv, ev_current_recv, net_current_recv]
        )
    
        for _ in range(10):
            grid_amps = await grid_current_recv.receive()
            ev_amps = await ev_current_recv.receive()
            net_amps = await net_current_recv.receive()
    
>           assert grid_amps.value_p1 is not None and grid_amps.value_p1 > 0.0
E           assert (None is not None)
E            +  where None = Sample3Phase(timestamp=datetime.datetime(2023, 3, 10, 7, 9, 39, 78987, tzinfo=datetime.timezone.utc), value_p1=None, value_p2=None, value_p3=None).value_p1
```

### What did you expect instead?

We should fix them

### Affected part(s)

Unit, integration and performance tests (part:tests)

ela-kotulska-frequenz, 2023-03-10:
Increase the sampling period rate to make it less flaky: https://github.com/frequenz-floss/frequenz-sdk-python/pull/251

mathias-baumann-frequenz, 2023-04-12:
This test is also flaky: 

=================================== FAILURES ===================================
_____________________ TestEVChargerPool.test_state_updates _____________________

self = <tests.timeseries.test_ev_charger_pool.TestEVChargerPool object at 0x7ff695e1c990>
mocker = <pytest_mock.plugin.MockerFixture object at 0x7ff68f98a510>

    async def test_state_updates(self, mocker: MockerFixture) -> None:
        """Test ev charger state updates are visible."""
    
        mockgrid = MockMicrogrid(grid_side_meter=False, sample_rate_s=0.01)
        mockgrid.add_ev_chargers(5)
        await mockgrid.start(mocker)
    
        state_tracker = StateTracker(set(mockgrid.evc_ids))
    
        async def check_states(
            expected: dict[int, EVChargerState],
        ) -> None:
            await asyncio.sleep(0.02)
            for comp_id, exp_state in expected.items():
                assert state_tracker.get(comp_id) == exp_state
    
        ## check that all chargers are in idle state.
        expected_states = {evc_id: EVChargerState.IDLE for evc_id in mockgrid.evc_ids}
        assert len(expected_states) == 5
>       await check_states(expected_states)

tests/timeseries/test_ev_charger_pool.py:56: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

expected = {6: <EVChargerState.IDLE: 'IDLE'>, 16: <EVChargerState.IDLE: 'IDLE'>, 26: <EVChargerState.IDLE: 'IDLE'>, 36: <EVChargerState.IDLE: 'IDLE'>, ...}

    async def check_states(
        expected: dict[int, EVChargerState],
    ) -> None:
        await asyncio.sleep(0.02)
        for comp_id, exp_state in expected.items():
>           assert state_tracker.get(comp_id) == exp_state
E           AssertionError: assert <EVChargerState.MISSING: 'MISSING'> == <EVChargerState.IDLE: 'IDLE'>
E            +  where <EVChargerState.MISSING: 'MISSING'> = <bound method StateTracker.get of <frequenz.sdk.timeseries.ev_charger_pool._state_tracker.StateTracker object at 0x7ff68fcc1cd0>>(6)
E            +    where <bound method StateTracker.get of <frequenz.sdk.timeseries.ev_charger_pool._state_tracker.StateTracker object at 0x7ff68fcc1cd0>> = <frequenz.sdk.timeseries.ev_charger_pool._state_tracker.StateTracker object at 0x7ff68fcc1cd0>.get

tests/timeseries/test_ev_charger_pool.py:51: AssertionError
------------------------------ Captured log call -------------------------------

leandro-lucarella-frequenz, 2023-04-19:
> tests/timeseries/test_ev_charger_pool.py:51: AssertionError

I'm hitting this one very often in #335.

@sahas-subramanian-frequenz can you have a look at this to see if we can fix the root cause?

leandro-lucarella-frequenz, 2023-04-19:
This might need to be moved to v0.20.0 if we can't find a proper workaround before the release.

sahas-subramanian-frequenz, 2023-04-19:
For now,  we can just increase the sample rate for the failing test like this:  https://github.com/frequenz-floss/frequenz-sdk-python/pull/336/commits/b91da340c403c5385601b52e6385a323aa4cba1c

For permanent fix,  I have already made a few tests by mocking the resampling actor's output when testing formulas.  I will work on a generic way to do this, so that all the tests can be updated.  I will get to that this week or early next week.

closed 2023-05-23 by daniel-zullo-frequenz