frequenz-floss/frequenz-sdk-python #250: Add new `net_consumption` formula to the `LogicalMeter

leandro-lucarella-frequenz, 2023-03-10:
### What's needed?

We need to provide a `net_consumption` metric in the `LogicalMeter`.

The name is still to be decided, so the issue is blocked until then. In your hands @matthias-wende-frequenz @frequenz-floss/datasci-team 

### Proposed solution

The `LogicalMeter` should get a new formula/metric `net_consumption` would calculate the client-side consumption of the microgrid. This should also include consumptions of EV chargers (but of course exclude PVs, batteries, etc.).

### Alternatives and workarounds

Right now users need to look at the microgrid topology themselves to make this calculation, excluding batteries, PVs, etc.

ela-kotulska-frequenz, 2023-03-10:
Is this what you need?
https://github.com/frequenz-floss/frequenz-sdk-python/blob/v0.x.x/tests/timeseries/_formula_engine/test_formula_composition.py#L147-L154

thomas-nicolai-frequenz, 2023-03-10:
> would calculate the client/market-side consumption of the microgrid

Its important to correctly define of what net consumption means. In the past we usually referred to it as market load, however net consumption is the consumption of none-flexible consumers that can't be shut down. EV chargers are obviously  a hybrid but we agreed to include them into the net consumption for now, opposite to PVs, CHPs and batteries. The main goal is to:

1. Have one place where net_consumption is being defined
2. Have a common understanding of what `net_consumption` means (definition)

cwasicki, 2023-03-13:
These are the definitions we agreed on:

[image]

$P_S$  : Supply power
$P_L$  : Net Consumption power
$P_G$ : Grid power
$P_B$  : Battery power
$P_L+P_S$ : Consumption excess power (supply excess also possible, but value will be negative if supply exceeds consumption)

Note:
* This uses Passive Sign Convention.
* For now EV is part of consumption only, in future EV will be treated as a battery.

shsms, 2023-03-23:
So the logical meter will eventually provide streams for:

1. `grid_power`
2. `grid_current` (used by the ev charging actor,  probably also in other safety scenarios)
3. `net_consumption`

Does that sound accurate?

I can get used to it if it makes sense to every one else,  but to have types of metrics (current and power) in the names for some streams,  and the word "consumption" in a similar context for another stream, feels a bit inconsistent and awkward to me.

thomas-nicolai-frequenz, 2023-03-23:
> and the word "consumption" in a similar context for another stream, feels a bit inconsistent and awkward to me.

Well the problem is net_consumption is `>0` but grid_power can be `<>=0`.

shsms, 2023-03-27:
I think the word `power` signifies that it is a derived quantity measured in `kW`,  not that it is flowing in both directions.  Even when it is a single direction,  it is the `power` value measured in `kW`.  

`consumption` provides a direction,  but takes away info about what quantity is being measured.  It could mean both power or current.

Also, later on, if we want to introduce a `net_consumption_current` data stream,  with 3-phase current values, would we then rename `net_consumption` to `net_consumption_power`, to disambiguate?

thomas-nicolai-frequenz, 2023-03-27:
> consumption provides a direction, but takes away info about what quantity is being measured. It could be either power or current.

true. good point.

cwasicki, 2023-03-27:
@shsms  Right, it should be `net_consumption_power`. Net consumption refers to the component, not the quantity. See https://github.com/frequenz-floss/frequenz-sdk-python/issues/250#issuecomment-1466946742 for the proposed definitions. But I see that this can be confusing. Do you have a better suggestion?

leandro-lucarella-frequenz, 2023-03-27:
Maybe `local_production` and `local_consumption`? (then with `_power` or `_current` or whatever suffix)?

shsms, 2023-03-27:
@cwasicki no, my problem was that the ticket just said `net_consumption`.  If we can say `net_consumption_power`, then I don't think it is confusing anymore.

cwasicki, 2023-03-28:
@shsms @leandro-lucarella-frequenz 
@thomas-nicolai-frequenz and I had another discussion about the terminology, apparently there was a misunderstanding. I have update comment https://github.com/frequenz-floss/frequenz-sdk-python/issues/250#issuecomment-1466946742 to reflect the new terminology. Notably:
* Net consumption is the combination of Production and Consumption. This is also in line with the traditional terminology in this field.
* We think that the term `production` instead of `supply` has lower chances of confusion (also w.r.t common terminology in the energy market).

Let me know in case you still have doubts.

thomas-nicolai-frequenz, 2023-03-28:
and `consumption` is `net_consumption + PV + ...`

thomas-nicolai-frequenz, 2023-03-28:
> So the logical meter will eventually provide streams for: grid_power, grid_current (used by the ev charging actor, probably also in other safety scenarios), net_consumption

1. grid_power
2. grid_current
3. consumption_power

And in case we want to mirror the DSO meter output we could also expose:

4. net_consumption_power (if grid_power > 0)
5. net_production_power (if grid_power <0)

shsms, 2023-03-28:
> And in case we want to mirror the DSO meter (one meter in each direction) output we could also expose:
>
>    net_consumption_power (if grid_power > 0)
>    net_production_power (if grid_power <0)

So these methods just split the `grid` power into two based on direction?  Do we need to make them consistent by doing one of:

1. `grid_power`, `grid_consumption_power`, `grid_production_power`
2. `net_power`, `net_consumption_power`, `net_production_power`

thomas-nicolai-frequenz, 2023-03-28:
> 1. grid_power, grid_consumption_power, grid_production_power

ok.

> net_power, net_consumption_power, net_production_power

no. all gone. we've revised the [comment](https://github.com/frequenz-floss/frequenz-sdk-python/issues/250#issuecomment-1486995731) above

cwasicki, 2023-03-30:
Blocking again for now for feedback from @tiyash-basu-frequenz before moving forward.

leandro-lucarella-frequenz, 2023-03-31:
Now `grid_production_power` sounds weirder than `grid_supply_power` but I guess consistency across names is more important than ultimate precision. Besides this i like the split. It makes names a bit longer and adds some boilerplate to the SDK but I think is much clearer what's going on when reading it, and reason about it.

thomas-nicolai-frequenz, 2023-04-02:
> Now grid_production_power sounds weirder than grid_supply_power

Well in which direction does the supply go? out or in? Supply is an ambiguous term that does not contain the direction thus @cwasicki proposal above is well suited imho.

tiyash-basu-frequenz, 2023-04-03:
> Well in which direction does the supply go? IN or OUT? Supply is an ambiguous term that does not contain the direction (source supplies destination)

True. The same argument also applies to production, IMO. After reading this, I suggest we highlight that `grid_production_x` refers to the _local_ production (and not by the grid). I still prefer the term `grid_production_x` for better consistency.

tiyash-basu-frequenz, 2023-04-03:
> * consumption_power: Passive-only components. Non-negative value, which is the gross consumption of the client (e.g. supermarket) without taking into account any renewable generation. (alternative terms: load, demand, sink)
> * production_power: Active-only components (ignoring self-consumption). Non-negative value, which is the power from the renewable generation only (alternative terms: generation, supply, source).

Just wanted to clarify a few points:
* Passive-only does not mean it will only ever consume. Inductors e.g. can produce massive back currents on sudden voltage drops. Then they could produce energy (according to our terminology). A similar thing can happen with capacitors too.
* We cannot know all the time if a measurement is of a passive component, because we don't always know what are behind dangling meters. There could be a generator behind a meter that we are not aware of. So, we don't exactly know if a dangling meter is measuring a purely passive load or not. Hence, the above definition of `consumption_power` has to be fixed.
* We know with 100% certainty if a measurement is of an active component or not, because we track placements of active components in microgrids.

I think the essence of the issue is to highlight the consumption of the components that are pre-declared to be just consumers (including charging-only EV chargers). So I think names like `consumer_consumption_power` and `consumer_production_power` could make sense.

tiyash-basu-frequenz, 2023-04-03:
> And in case we want to mirror the DSO meter (one meter in each direction) output we could also expose:
>
> grid_consumption_power (if grid_power > 0)
> grid_production_power (if grid_power <0)

Do we actually have a use-case for this? Asking because we did this initially in the microgrid API, but it did not prove to be as helpful as we thought (and rather problematic). A single signed value conveys the same information, and can be clipped pretty easily.

thomas-nicolai-frequenz, 2023-04-03:
> Do we actually have a use-case for this? Asking because we did this initially in the microgrid API

DSOs measure and report consumption and feed-in separately. To be able to easily compare the data these shortcuts might become handy. This is just a convenience feature and is independent of the API. That goes along the lines of the discussion of battery.charge(10) and battery.discharge(10) vs. battery.set_power(10).

thomas-nicolai-frequenz, 2023-04-03:
> True. The same argument also applies to production

Not sure I necessarily agree. What is the opposite of supply? For production its easy, its consumption and these are terms that everyone understands. Keep in mind we want to open up electricity to a wider audience and the terminology in this space has been set up a long time ago and thus many terms are way passed their time. We should be mindful not to repeat the mistakes of the past.

cwasicki, 2023-04-05:
Updated proposal:
* Terminology follows `{component}_{quantity}` and `{component}_{consumption,production}_{quantity}`.
* Components are: `grid`, `battery`, `consumer` and individual active components `pv`, `chp`, `wind`.

 
Using the example of power:
* **grid_power**: Can be positive (consumption from the grid) or negative (production into the grid).
* **battery_power**: Can be positive (charge battery) or negative (discharge battery).
* **{pv,chp,wind}_power**: Active components, the power from local (renewable) generation. Negative if electric power is generated, otherwise zero or positive in case of self-consumption of the generator. (alternative terms: generation, supply, source, active)
* **consumer_power**: This aggregates the remaining parts not covered by the above metrics. Under normal circumstances this is expected to correspond to the gross consumption of the site excluding active parts and battery. The implementation should still support negative values to also handle exotic site topologies (e.g. unidentified generators) and sudden short-term effects. (alternative terms: load, demand, sink, passive)

In addition to that, for each of the above there will two additional metrics:
 * **{component}_consumption_power**: Positive power value if `{component}_power` is positive, otherwise zero.
  * **{component}_production_power**:  Positive power value if `{component}_power` is negative, otherwise zero.

Remarks:
* Gross consumption: Consumption before accounting for any local generation from solar, wind or CHP.
* Net consumption/load: This term traditionally refers to the difference between the gross electricity consumption and the local generation (like PV production). This is the electricity consumption that needs to be met by the battery or from the main grid. This means `consumer_power + {pv,chp,wind}_power` (note that latter is negative when electricity is produced). Since this term caused confusion in the past, it's not included in this proposal. 
* Residual consumption/load: In microgrid context sometimes used as the remaining difference between the net consumption and the battery power, i.e. what we define as grid power.   
* Consumer: Term deliberately chosen to indicate that this component is expected to predominantly consume electric power. Under normal circumstances there should be no active sources in this metric and the metrics `consumer_{consumption,production}_power` would not be used.

@thomas-nicolai-frequenz @shsms @leandro-lucarella-frequenz @tiyash-basu-frequenz

thomas-nicolai-frequenz, 2023-04-06:
Just to add some more context to the term `risidual`. There are a few more definitions that we should be aware off:
- Residual energy
- Residual power
- Residual capacity

@leandro-lucarella-frequenz we need to find a good place in the SDK wiki where we explain each of the terms. I feel its quite important to clearly define and explain it.

david-natingga-frequenz, 2023-04-06:
What about EV charging stations? It should be a component in the same position like a battery: `evc_station`.

cwasicki, 2023-04-06:
> What about EV charging stations? It should be a component in the same position like a battery: `evc_station`.

Good point, for now we cover it in `consumer` since these are currently only consuming. In future they would be treated as a battery. But not sure if we will need a dedicated `ev_battery` component for this?

> Just to add some more context to the term `residual`

Residual load is mostly used in context of the large grid where it is the electricity demand minus fluctuating renewable generation (https://de.wikipedia.org/wiki/Residuallast). On microgrid-level the usage seemed a bit vague so I would avoid a strict definition for us. Just added it for context since we discussed about it.

thomas-nicolai-frequenz, 2023-04-06:
> But not sure if we will need a dedicated ev_battery component for this?

No, EV chargers will stay being their own component category but will most likely derived from the same interface as the battery component. @sahas-subramanian-frequenz is aware of that. We did already discuss that.

thomas-nicolai-frequenz, 2023-04-06:
> Good point, for now we cover it in consumer since these are currently only consuming.

@cwasicki I don't mind carving them out to be honest. It wouldn't make much difference I guess and might avoid needing to restructure the code in the future if we want to keep things more clean. Your call.

thomas-nicolai-frequenz, 2023-04-06:
> On microgrid-level the usage seemed a bit vague so I would avoid a strict definition for us. 

I think we should define it rather not defining it in the context of microgrid if we can. It opens up the possibility that we can shape the usage of this term in a way we feel its correctly applied. I think it would be dangerous to not talk about it.

cwasicki, 2023-04-06:
> @cwasicki I don't mind carving them out to be honest. It wouldn't make much difference I guess and might avoid needing to restructure the code in the future if we want to keep things more clean. Your call.

Agree, update the proposal to include `ev_power` equivalent to `battery_power`. 

> I think we should define it rather not defining it in the context of microgrids if we can. It opens up the possibility that we can shape the usage of this term in a way we feel its correctly applied. I think it would be dangerous to not talk about it.

I agree that we should document it, but wouldn't encourage its usage or include the term in the proposal since it does not describe a new concept but would be just another term for `grid_power`. Of course we could swap the two terms since latter is also not ideal, but overall I would stick to one term only.

thomas-nicolai-frequenz, 2023-04-08:
> but would be just another term for grid_power

lets just mentioned it when we define grid_power. lets add this to the public wiki.

leandro-lucarella-frequenz, 2023-04-11:
> @leandro-lucarella-frequenz we need to find a good place in the SDK wiki where we explain each of the terms. I feel its quite important to clearly define and explain it.

We could put it in the https://github.com/frequenz-floss/frequenz-sdk-python/wiki/Glossary, or it's own Wiki page for now. In the end I think it should be in the SDK documentation website (https://frequenz-floss.github.io/frequenz-sdk-python/v0.19/), but for now a Wiki page should be enough.

But I think it would be best if this doc is started by the people creating these definitions.

cwasicki, 2023-04-11:
Unblocking this issue again.

> We could put it in the https://github.com/frequenz-floss/frequenz-sdk-python/wiki/Glossary, or it's own Wiki page for now. In the end I think it should be in the SDK documentation website [...]

Moved this to #316