frequenz-floss/frequenz-sdk-python #291: Add a `DispatchForwardingActor`

leandro-lucarella-frequenz, 2023-03-29:
### What's needed?

We need an actor that is in charge of abstracting the access to the dispatch API for receiving dispatches, and that can act as a local cache for dispatches, so if the connection to the dispatch API is lost other actors can still get whatever contents could be fetched in the past.

### Proposed solution

Add a `DispatchForwardingActor` that will poll the dispatch API and save currently active dispatches (start time in the present or future) and forward them to other actors.

When an actor first subscribe to the dispatch channel, they will get the whole list of active dispatches and then when new dispatches are found in the dispatch API, they will be sent to the channel too.

This could be done in a class or actor, but an actor is preferred because we only need one cache for a whole microgrid location, not one per process/controller.

### Additional context

We have an internal actor, we can start moving this actor to the SDK or build it from scratch depending on how much will change with the dispatch API client.

leandro-lucarella-frequenz, 2023-03-29:
FYI @thomas-nicolai-frequenz @stefan-brus-frequenz @christianparpart @sahas-subramanian-frequenz

thomas-nicolai-frequenz, 2023-03-30:
I was wondering if the actor should more be called along the lines of a "dispatch request broadcaster" or "dispatch request router"?

leandro-lucarella-frequenz, 2023-03-31:
OK, renamed to `DispatchBroadcastingActor`.

llucax, 2024-06-18:
Done: https://github.com/frequenz-floss/frequenz-dispatch-python

closed 2024-06-18 by llucax