frequenz-floss/frequenz-sdk-python #314: Use a better name for methods creating a new receiver

leandro-lucarella-frequenz, 2023-04-11:
### What's needed?

Right now the `LogicalMeter`, `BatteryPool`, and `EVChargerPool` name functions returning a receiver after the data that comes from the receiver.

Examples:

```py
grid_power_receiver = logical_meter.grid_power()
power_bounds_receiver = battery_pool.power_bounds()
```

This can be a bit obscure for people that are new to programming with channels, senders and receivers in general and specifically new to the SDK.

People might think that the method (which sounds like a regular getter) might be idempotent so calling:

```py
grid_power = await logical_meter.grid_power().receive()
# do something
grid_power = await logical_meter.grid_power().receive()
```

is OK, while in reality is creating multiple receivers, which adds overhead to the channel processing. Also if the channel has the option to always return the last value, it will receive a repeated value if there was nothing new between the first `receive()` and the second.

### Proposed solution

Make the metric attributes an object that can be asked for a new receiver instead.

In the example above, this would be:

```py
grid_power_receiver = logical_meter.grid_power.new_receiver()
power_bounds_receiver = battery_pool.power_bounds.new_receiver()
```

This makes it much more obvious that a new receiver is being created, so this code will look wrong more naturally:

```py
grid_power = await logical_meter.grid_power.new_receiver().receive()
# do something
grid_power = await logical_meter.grid_power.new_receiver().receive()
```

### Alternatives and workarounds

Some alternatives were discussed in:

* https://github.com/frequenz-floss/frequenz-sdk-python/discussions/215

### Additional context

### Extensions

Using this scheme also enables the possibility to add other ways to query a metric in the future. For example there could be a way to query the current or last value:

```py
current = await p.soc_capacity.current_value()  # Queries the microgrid API (if it's even possible :D)
last = p.soc_capacity.last_value  # If we have a channel getting this, we could peek into it
```

Also we could use it to compose metrics, for example:

```py
composed_receiver = (
    logical_meter.grid_power + battery_pool.active_power
    ).new_receiver()
```

### Notes

* If getting the wrapper is not trivial, then it should be a function call instead of a property:
  ```py
  grid_power_receiver = logical_meter.grid_power().new_receiver()
  ```
  But we should try to cache these wrappers to allow for a property syntax if possible.

sahas-subramanian-frequenz, 2023-04-13:
>     grid_power = await logical_meter.grid_power().receive()
>     # do something
>     grid_power = await logical_meter.grid_power().receive()

It is actually `await (await logical_meter.grid_power()).receive()`.  Creating these formula objects is async,  because it involves sending subscription objects to the resampling actor.

So the interface will be:

``` python
grid_power_receiver = (await logical_meter.grid_power()).new_receiver()
power_bounds_receiver = (await battery_pool.power_bounds()).new_receiver()

grid_power = await (await logical_meter.grid_power()).new_receiver().receive()
# do something
grid_power = await (await logical_meter.grid_power()).new_receiver().receive()

composed_receiver = (
    await logical_meter.grid_power() + await battery_pool.active_power()
    ).new_receiver()
```

That means we will will have to do it in a few steps, with some tmp variables,  but it is still a nice approach,  and doing it also simplifies the formula engine a lot.

leandro-lucarella-frequenz, 2023-04-13:
Oh, damn, that looks quite inconvenient though :(

Can't we at least make the wrapper object lazy and do the subscription when a `new_receiver()` is requested?

This would at least simplify the interface a bit:

```py
grid_power_receiver = await logical_meter.grid_power().new_receiver()
power_bounds_receiver = await battery_pool.power_bounds().new_receiver()

grid_power = grid_power_receiver.receive()
# do something
grid_power = grid_power_receiver.receive()

composed_receiver = await (
    logical_meter.grid_power() + battery_pool.active_power()
    ).new_receiver()
```

Another way would be to make it even more lazy and make `new_receiver()` return a receiver wrapper, that will do the subscription the first time `receive()` is called. It is quite a bit of work, but it would make the interface much simpler.

And I guess we only need to await because we are creating a new subscription each time someone asks for the `grid_power`, right? If we pre-create the subscriptions and have only one channel per metric, then we wouldn't need any `await`s at all to get a new receiver (at the expense of having all the metrics streamed even if nobody asked for them).

sahas-subramanian-frequenz, 2023-04-13:
It was possible to eliminate all `await`s except for `receive`, by doing some lazy initialization.  I've checked in the code in: https://github.com/frequenz-floss/frequenz-sdk-python/pull/327

I will try and squash some commits,  update the PR description, etc,  and maybe it will become a bit easier to review.

closed 2023-04-19 by sahas-subramanian-frequenz