frequenz-floss/frequenz-sdk-python #319: Document the terminology used around consumption and production

leandro-lucarella-frequenz, 2023-04-12:
### What's needed?

We need to document the terminology used to talk about production and consumption (of power, current, etc.).

The final proposal was posted in this discussion:
* https://github.com/frequenz-floss/frequenz-sdk-python/discussions/318?converting=1#discussioncomment-5589177	

### Proposed solution

Move the documentation to the Wiki, to its own page or the [Glossary](https://github.com/frequenz-floss/frequenz-sdk-python/wiki/Glossary) if that's a good fit.

### Alternatives and workarounds

The individual definitions should be also included in the code documentation, in the logical meter, battery pool, etc.

We could also include a summary of this terminology in the `frequenz.sdk.microgrid` module documentation.

cwasicki, 2023-05-03:
Added a dedicated section to the glossary https://github.com/frequenz-floss/frequenz-sdk-python/wiki/Glossary#microgrid-power-terminology

leandro-lucarella-frequenz, 2023-05-04:
Great, thanks! Closing. Feel free to reopen there was anything else to be done.

closed 2023-05-04 by leandro-lucarella-frequenz