frequenz-floss/frequenz-sdk-python #352: Add a weather API client

leandro-lucarella-frequenz, 2023-04-21:
### What's needed?

We need to provide a weather API client.

### Proposed solution

Write a client based on what we have for the microgrid API.

This should live in `frequenz.sdk.weather.client` to also follow the microgrid API.

### Additional context

Dispatching will be a core part of building actors and apps, so we need to include support for it in the SDK.

FYI @sahas-subramanian-frequenz @thomas-nicolai-frequenz

cwasicki, 2023-07-12:
@llucax Is this essential if we want to use weather forecasts in our forecasting actor?

llucax, 2024-05-10:
This is being done as a separate repo.

closed 2024-05-10 by llucax