frequenz-floss/frequenz-sdk-python #360: Expose grid connection limits

leandro-lucarella-frequenz, 2023-04-26:
### What's needed?

We need to expose the new [microgrid grid connection rated fuse maximum current](https://github.com/frequenz-floss/frequenz-api-microgrid/blob/2beb6745b2e0be82f12e8ef171bec0ed1c51362a/proto/frequenz/api/microgrid/grid.proto#L25).

For now this limit can be static (queried from the microgrid API and kept immutable). A change in the grid connection limit will need a SDK restart. In the future we want to support dynamically changing this limit.

### Proposed solution

### Low-level

Add a new `Component` that wraps the `Grid` object and its metadata, as it is done with other microgrid components.

### High-level

1. Add an object to represent the `GridConnection` and a read-only `rated_fuse_current` property
2. Add a global instance exposed via `frequenz.sdk.microgrid`
3. Initialize the global instance in the `initialize()` function
   a. Until the new version of the microgrid API is deployed, just fill the instance `rated_fuse_current` with `None`

We should probably also add some utility methods for checking and adjusting a current value to the accepted limits. We could accept multiple current values so it is easy to use even with 3-phase samples.

```py
class GridConnection:

    @property
    def rated_fuse_current(self) -> float:
        ...

    def is_current_within_bounds(self, *current: float) -> bool:
        ...

    @overload
    def adjust_current(self, /, current: float) -> float:
        ...

    @overload
    def adjust_current(self, /, current1: float, current2: float, *extra_currents: float) -> list[float]:
        ...

    def adjust_current(self, *current: float) -> float | list[float]:
        """Truncate any current above the rated_fuse_current."""
```

### Use cases

Probably the `PowerManager`/`PowerArbitrator` (#161) or the should check these limits when applying charge/discharge commands, but others might as well, like the `EVChargerPool` might want to check the limit before increasing the bounds to allow vehicles to charge with high power.

### Additional context

Design discussion:
* https://github.com/frequenz-floss/frequenz-sdk-python/discussions/322#discussioncomment-5718348

Future work:

- [ ] Initialize with the proper microgrid API calls after the new microgrid API exposing the limit is deployed
- [ ] Support dynamically changing this limit.

  This means we need a streaming interface from the microgrid API and we should make the `rated_fuse_current` a channel that streams changes in this property, like the `battery_pool()`, etc.

thomas-nicolai-frequenz, 2023-04-26:
> This means we need a streaming interface from the microgrid API

Can't we go with polling first if thats easier then implementing a streaming interface in the API?

leandro-lucarella-frequenz, 2023-04-26:
Let's keep discussing in https://github.com/frequenz-floss/frequenz-sdk-python/discussions/322 so we keep the issue clean.

llucax, 2023-07-07:
Blocked on #412

llucax, 2023-08-09:
Unblocked because #412 was merged.

llucax, 2023-08-24:
I'm not sure, I don't have a strong opinion but will play devil's advocate here, where the devil is @thomas-nicolai-frequenz :D. He said several times that we shouldn't couple the SDK with the microgrid API. IMHO in the context of the SDK the most clear term is `GridConnection`, the `Point` doesn't seem to be adding much and making a long symbol name even longer.

That aside, the class will be mostly hidden to the user, the public interface will be `microgrid.grid_connection `, but I guess the discussion applies to it too.

@frequenz-floss/python-sdk-team not sure if anybody else have something to share about this.

shsms, 2023-08-24:
I would even shorten it and say `Grid` and `microgrid.grid.fuse`, etc,  so it is easy to type and read, etc.

llucax, 2023-08-24:
For me `microgrid.grid` looks a bit weird. Is not a *grid* inside the microgrid (as with most other elements, like the `battery_pool`, which is something inside the microgrid), but a connection to the external world, so IMHO the `connection` part is important. But I can live with just `grid` too if everybody else thinks is clear enough.

tiyash-basu-frequenz, 2023-08-24:
To me, `Grid`/`grid` is a clearer term than `GridConnection` :)

thomas-nicolai-frequenz, 2023-08-24:
Please be aware that a microgrid can have:
1. One grid connection (usually the default)
2. Multiple grid connections
3. No grid connection at all (island)

Usually the grid connection is also referenced as the metering point. Not sure if that helps but I think Luca has a point when saying a grid within the microgrid sounds a bit odd. I also feel just referencing it as grid might be too ambiguous.

tiyash-basu-frequenz, 2023-08-24:
Since we are adding a singleton here, we want to support just one grid connection point here, right? `grid_connections_count != 1` would therefore an error.

thomas-nicolai-frequenz, 2023-08-24:
> Also, should we error out when there is no grid connection in a microgrid?

No we can't do that. You can't kill the island scenario.

tiyash-basu-frequenz, 2023-08-24:
> No we can't do that. You can't kill the island scenario. Or am I misunderstanding something here?

I will include the island scenario.

llucax, 2023-08-24:
> Please be aware that a microgrid can have:
>
 >   One grid connection (usually the default)
 >   Multiple grid connections
 >   No grid connection at all (island)

Here we are talking about an abstraction, like the battery pool. It will be a "virtual" connection, for example when we need to check if we are surpassing the allowed current accepted buy the fuses of multiple grid connections, we don't want the user to know the topology and have to do the math themselves.

llucax, 2023-08-24:
About the island scenario, shouldn't in that case just be `microgrid.grid == None`?

tiyash-basu-frequenz, 2023-08-25:
Does it make sense to abstract `Fuse` further? Note that fuses will not report current metrics live. The only property it has is a rated max current, and this property is a static value.

One thing that we could do right now is add a limit per-phase to the `Fuse` class, so the upper-layers remain unaffected when we start supporting fuses in the Microgrid API.

closed 2023-08-30 by tiyash-basu-frequenz