frequenz-floss/frequenz-sdk-python #389: Add initial implementation of unit types

leandro-lucarella-frequenz, 2023-05-16:
### What's needed?

We need to have types to represent units like power, current, voltage, etc. to avoid common mistakes of using different units, like using kilowatt or watt.

### Proposed solution

Add the most basic representation for the most basic quantities we have to try out the idea.

### Additional context

Discussion with more details on a possible implementation:

* https://github.com/frequenz-floss/frequenz-sdk-python/discussions/6

christianparpart, 2023-05-17:
NB: I was adding a helper function to the FCR prequalification actor to cast a `float` into a human readable representation for Watt (with the best fitting unit scale chosen) plus some tests for it. The thought then was to probably have such a function in the SDK. And I see a good fit for that along with a unit type for that: https://github.com/frequenz-io/frequenz-actor-fcr-prequalifying/pull/153#discussion_r1196314193

leandro-lucarella-frequenz, 2023-05-22:
Interesting. It would be good to bring all kind of things, that could be generally useful for other actors to the SDK as early as possible, so we can agree on an interface or at least a broad direction we want to make, so then we can either use what you implemented or it is easy to migrate when we incorporate something in the SDK.  Opening a new ["idea discussion"](https://github.com/frequenz-floss/frequenz-sdk-python/discussions/new?category=ideas) is probably the best place for this.