frequenz-floss/frequenz-sdk-python #430: The package `networkx-stubs` has been archived

llucax, 2023-06-09:
### What happened?

We are using the package https://github.com/eggplants/networkx-stubs, but this repository has been archived, so type check for `networkx` might break soon.

### What did you expect instead?

`nextworkx` type check should not break. We should eventually add it to `mypy` ignore list and monitor https://github.com/networkx/networkx/issues/3988 to see if type hints are eventually added natively in the project to remove it from the `mypy` ignores.

### Affected part(s)

Build script, CI, dependencies, etc. (part:tooling)

Avasam, 2023-08-09:
As a drop-in replacement, you could try https://github.com/fmagin/networkx-stubs or downloading the stubs from https://github.com/microsoft/python-type-stubs/tree/main/stubs/networkx
There are efforts that just started to move and merge both of those towards typeshed (https://github.com/fmagin/networkx-stubs/issues/8 / https://github.com/python/typeshed/pull/10544)

daniel-zullo-frequenz, 2023-09-19:
We could fork [microsoft/python-type-stubs](https://github.com/microsoft/python-type-stubs) (suggested repo) which holds the stubs currently used in  `Pylance` and add a configuration script to create a package only for `networkx` stubs that we can reference from frequenz-SDK project configuration files. @llucax please let me know if you have any other ideas/thoughts about it

llucax, 2023-09-19:
If there are alternatives, I don't see why not using the alternatives. But in a talk with @matthias-wende-frequenz we thought maybe we can even remove the dependency in a near future, we are not really using much from networkx, so I wouldn't spend much time maintaining a fork of the stubs even if we didn't have any alternatives.

daniel-zullo-frequenz, 2023-09-19:
I've tried https://github.com/fmagin/networkx-stubs and unfortunately got 16 `mypy` errors (this is a WIP/experimental stub package)

```sh
~/projects/frequenz-sdk-python$ nox -s mypy
nox > Running session mypy
nox > Creating virtual environment (virtualenv) using python in .nox/mypy
nox > python -m pip install -e '.[dev-mypy]'
nox > mypy 
src/frequenz/sdk/microgrid/_graph.py:318: error: Missing type parameters for generic type "DiGraph"  [type-arg]
src/frequenz/sdk/microgrid/_graph.py:390: error: Incompatible types in assignment (expression has type "filter[tuple[Any, Any]]", variable has type "OutEdgeView[Any]")  [assignment]
src/frequenz/sdk/microgrid/_graph.py:474: error: Need type annotation for "new_graph"  [var-annotated]
src/frequenz/sdk/microgrid/_graph.py:764: error: Call to untyped function "is_tree" in typed context  [no-untyped-call]
src/frequenz/sdk/microgrid/_graph.py:786: error: Argument 1 to "filter" has incompatible type "Callable[[Any], bool]"; expected "Callable[[Component], TypeGuard[bool]]"  [arg-type]
src/frequenz/sdk/microgrid/_graph.py:786: error: Non-overlapping equality check (left operand type: "DiDegreeView[Any]", right operand type: "Literal[0]")  [comparison-overlap]
src/frequenz/sdk/microgrid/_graph.py:802: error: Argument 1 to "filter" has incompatible type "Callable[[Any], bool]"; expected "Callable[[Component], TypeGuard[bool]]"  [arg-type]
src/frequenz/sdk/microgrid/_graph.py:802: error: Non-overlapping equality check (left operand type: "DiDegreeView[Any]", right operand type: "Literal[0]")  [comparison-overlap]
src/frequenz/sdk/microgrid/_graph.py:812: error: Argument 1 to "filter" has incompatible type "Callable[[Any], bool]"; expected "Callable[[bool], TypeGuard[bool]]"  [arg-type]
src/frequenz/sdk/microgrid/_graph.py:812: error: "bool" has no attribute "category"  [attr-defined]
src/frequenz/sdk/microgrid/_graph.py:822: error: Non-overlapping equality check (left operand type: "DiDegreeView[Any]", right operand type: "Literal[0]")  [comparison-overlap]
src/frequenz/sdk/microgrid/_graph.py:822: error: "bool" has no attribute "component_id"  [attr-defined]
src/frequenz/sdk/microgrid/_graph.py:848: error: Unsupported operand types for < ("int" and "DiDegreeView[Any]")  [operator]
src/frequenz/sdk/microgrid/_graph.py:854: error: Non-overlapping equality check (left operand type: "DiDegreeView[Any]", right operand type: "Literal[0]")  [comparison-overlap]
src/frequenz/sdk/microgrid/_graph.py:942: error: Non-overlapping equality check (left operand type: "DiDegreeView[Any]", right operand type: "Literal[0]")  [comparison-overlap]
src/frequenz/sdk/microgrid/_graph.py:943: error: Unsupported operand types for < ("int" and "DiDegreeView[Any]")  [operator]
Warning: unused section(s) in pyproject.toml: module = ['async_solipsism', 'async_solipsism.*', 'grpc.aio.*', 'sybil', 'sybil.*']
Found 16 errors in 1 file (checked 94 source files)
nox > Command mypy  failed with exit code 1
nox > Session mypy failed.
```

while using the archived https://github.com/eggplants/networkx-stubs we got only 5 errors

```sh

$ nox -s mypy
nox > Running session mypy
nox > Creating virtual environment (virtualenv) using python in .nox/mypy
nox > python -m pip install -e '.[dev-mypy]'
nox > mypy 
src/frequenz/sdk/microgrid/_graph.py:414: error: Call to untyped function "predecessors" in typed context  [no-untyped-call]
src/frequenz/sdk/microgrid/_graph.py:440: error: Call to untyped function "successors" in typed context  [no-untyped-call]
src/frequenz/sdk/microgrid/_graph.py:758: error: Call to untyped function "number_of_nodes" in typed context  [no-untyped-call]
src/frequenz/sdk/microgrid/_graph.py:764: error: Call to untyped function "is_tree" in typed context  [no-untyped-call]
src/frequenz/sdk/microgrid/_graph.py:941: error: Call to untyped function "has_node" in typed context  [no-untyped-call]
Warning: unused section(s) in pyproject.toml: module = ['async_solipsism', 'async_solipsism.*', 'grpc.aio.*', 'sybil', 'sybil.*']
Found 5 errors in 1 file (checked 94 source files)
nox > Command mypy  failed with exit code 1
nox > Session mypy failed.

```

daniel-zullo-frequenz, 2023-09-19:
> we thought maybe we can even remove the dependency in a near future, we are not really using much from networkx, so I wouldn't spend much time maintaining a fork of the stubs even if we didn't have any alternatives.

Then I would just keep ignoring `networkx` in `mypy` configuration as the effort to have the stubs working wouldn't be worth it

daniel-zullo-frequenz, 2023-09-19:
> Alright, can you just create a PR updating the comment in the pyproject.toml file to point to this issue? Then we can close this issue as wontfix.

yes, I've just pushed the PR #685

commit c7c787a4 (2023-09-21): Update obsolete comment about networkx-stubs (#685)

The networkx-stubs package was already working as the issue mentioned in
the comment was fixed. However, the package networkx-stubs was removed a
while ago from Frequenz-SDK as it was no longer maintained and it has
been archived.
At the moment of writing, there are only outdated and incomplete
alternatives to networkx-stubs.
Besides that, the Frequenz-SDK is not heavily using networkx, so it is
not worth it to maintain our own networkx-stubs package or to contribute
to a third-party package to fix errors.

Fixes #430
PR #685 (2023-09-21): Update obsolete comment about networkx-stubs
The networkx-stubs package was already working as the issue mentioned in the comment was fixed. However, the package networkx-stubs was removed a while ago from Frequenz-SDK as it was no longer maintained and it has been archived.
At the moment of writing, there are only outdated and incomplete alternatives to networkx-stubs.
Besides that, the Frequenz-SDK is not heavily using networkx, so it is not worth it to maintain our own networkx-stubs package or to contribute to a third-party package to fix errors.

Fixes #430
closed 2023-09-21 by llucax