frequenz-floss/frequenz-sdk-python #434: Generate .dot file syntax for component graph

christianparpart, 2023-06-13:
### What's needed?

Having looked at the pure numerical output today, I think it might be an awesome addition for improved and quicker understanding the component graph by having it rendered as graph (of components).

### Proposed solution

For this we might want to look into graphviz's .dot file syntax. This could be used by the user to either feed it into the `dot` CLI tool or [directly render it via Sixel](https://pypi.org/project/PySixel/) to the terminal, if desired.

### Use cases

I am aware that this doesn't sound like a top priority feature, but visualization might help quickly understanding a graph or even be good for passing over to others for inspection (better than numbers).

### Alternatives and workarounds

number printing, as-is.

### Additional context

https://graphviz.org/

llucax, 2023-06-14:
I think it's a very good idea, but I'm not sure if this is something for the SDK, at least up until now we don't have any CLI tools in the SDK and it seems like a tool like this might be useful even for people not developing with the SDK (the tool could just use the microgrid API directly). Maybe it could be added to the `frequenz-api-microgrid` repo? @tiyash-basu-frequenz opinions?

tiyash-basu-frequenz, 2023-06-15:
I'd suggest adding it to the microgrid API now. Because:
1. we do not need to know the details of every microgrid in general. Having a a few curated microgrids representing our test cases should be enough. For such use-cases we do not need this tool.
2. if, however, people need to see the microgrid connections, we should try to do it in the UI portal. So, IMO, it's should go over to the UI :)

That said, we have an internal tool which we used in the early days during development of the microgrid API and the server. I'll share it with you.