frequenz-floss/frequenz-sdk-python #466: BatteryPool really should expose a temperature receiver

christianparpart, 2023-06-26:
### What's needed?

By migrating away from PowerDistributore and use of `BatteryData` / `InverterData`, the only thing missing is access to the `temperature` of a battery(-pool).

### Use cases

Analogous to `soc`, `power`, `power_bounds`, `capacity`, add a `temperature` receiver.

Most likely that should be called `temperature_max` to expose he max value of all batteries in a pool (at least that's wha I need in my PQ actor).

### Alternatives and workarounds

still depend on `BatteryData` and legacy APIs :)

### Additional context

This is my second attempt to migrate our PQ actor to the BatteryPool :)

thomas-nicolai-frequenz, 2023-07-26:
> BatteryPool.temperature should stream objects that contain the min/max/avg block temperatures of all batteries 
 
 We need to revise this. The battery pool should expose:
 
 1. Average temperature of the entire battery pool
 2. The highest and lowest temperature in a given pool on a per battery level (NOT block level)
 
 > min/max/avg block temperatures
 
The block concept is not a general concept and thus the smallest unit we can look at is on a per battery level.

thomas-nicolai-frequenz, 2023-07-26:
@sahas-subramanian-frequenz I need to revise the above. In this issue we should only expose the average temperature across all batteries in a given pool. That means averaging the average temperature reported by each battery. The metric for each battery is being exposed in the `MetricAggregation` message.

```Protobuf
message MetricAggregation {
  // The average value of the metric.
  float avg = 1;

  ...
}
```

llucax, 2023-07-27:
Is really average temperature useful at all? If you have 9 batteries ar 20 degrees and 1 at 70, shouldn't users know that one battery is really hot instead of thinking that all are at 25 degrees? My impression is that providing an average could make users have a false sense of security and very prone to miss important issues.

thomas-nicolai-frequenz, 2023-07-27:
> If you have 9 batteries ar 20 degrees and 1 at 70, shouldn't users know that one battery is really hot instead of thinking that all are at 25 degrees?

It depends on the use case. The max is the least helpful figure of all. If one out of ten batteries is very hot it means the available SoP is reduced by just 10%. I am not sure what one would do with the information of the max temperature to be honest. Whats important is to know the average across all batteries.

shsms, 2023-07-27:
> My impression is that providing an average could make users have a false sense of security and very prone to miss important issues.

this is also why using SoP as an indicator of temperature (or other) issues would give clearer indications to users.  SoP of the pool is not average, but the sum.  So if the SoP of one battery starts to drop,  it will be noticeable.  Further SoP drops exponentially, with linear temperature increase:  https://github.com/frequenz-io/nitrogen-core-apps/blob/v0.x.x/docs/images/bounded-exp-decay.png

thomas-nicolai-frequenz, 2023-07-27:
> this is also why using SoP as an indicator of temperature

@sahas-subramanian-frequenz agreed but it really depends on the specific use case.

PR #552 (2023-08-01): Temperature streaming from the BatteryPool
Closes #466
commit 3eae96b7 (2023-08-01): Temperature streaming from the BatteryPool (#552)

Closes #466
PR #485 (2023-08-01): Add BatteryPool.temperature and quantity Temperature
Closes #466.