frequenz-floss/frequenz-sdk-python #543: BatteryPool should stream soc and capacity independent of inverter status

shsms, 2023-07-27:


shsms, 2023-07-27:
I just remembered that this is behaving as designed, when writing the issue, so decided to write this note instead.  BatteryPool is expected to report only the usable SoC and Capacity of the pool.  And if a corresponding inverter is out of order,  then that capacity and SoC are not usable,  so the batterypool should not consider these.

I will keep this issue open until the next SDK meeting, just in case.

thomas-nicolai-frequenz, 2023-07-30:
I think this should be clearly document in case this isn't yet. 

> And if a corresponding inverter is out of order, then that capacity and SoC are not usable
That totally makes sense imho

llucax, 2023-08-02:
This wasn't mentioned in the last meeting, maybe you want to add it to the agenda @shsms ?

commit 1229824e (2023-09-05): Clarify documentation of BatteryPool soc and capacity methods (#637)

SoC and capacity from batteries whose inverters are out of order are not
usable, and therefore not considered.

Fixes #543
commit 17dd9277 (2023-09-05): Clarify documentation of BatteryPool soc and capacity methods

SoC and capacity from batteries whose inverters are out of order are not
usable, and therefore not considered.

Fixes #543
PR #637 (2023-09-05): Clarify documentation of BatteryPool soc and capacity methods
SoC and capacity from batteries whose inverters are out of order are not usable, and therefore not considered.

Fixes #543
closed 2023-09-05 by shsms