frequenz-floss/frequenz-sdk-python #546: BatteryPool should log exact reasons when components are marked as `NOT_WORKING`

shsms, 2023-07-27:


Marenz, 2023-08-24:
I am not sure what is to do for this ticket. [Every single method ](https://github.com/frequenz-floss/frequenz-sdk-python/blob/v0.x.x/src/frequenz/sdk/actor/power_distributing/_battery_status.py#L235)that indirectly causes a `NOT_WORKING` status, also outputs the reason why it does that. 
What else do we want?

tiyash-basu-frequenz, 2023-08-28:
> I am not sure what is to do for this ticket. [Every single method ](https://github.com/frequenz-floss/frequenz-sdk-python/blob/v0.x.x/src/frequenz/sdk/actor/power_distributing/_battery_status.py#L235)that indirectly causes a `NOT_WORKING` status, also outputs the reason why it does that.

True. Every method logs the reason behind failure. Including [the one for inverters](https://github.com/frequenz-floss/frequenz-sdk-python/blob/v0.x.x/src/frequenz/sdk/actor/power_distributing/_battery_status.py#L244).

llucax, 2023-09-04:
Then maybe it was a NOP. @shsms any final words before we close this?