frequenz-floss/frequenz-sdk-python #551: Add timesetamp in Power Distributing Result

david-natingga-frequenz, 2023-07-28:
### What's needed?

Add timesetamp in Power Distributing Result:
- We need to know when exactly the power was set.

### Use cases

FCR Balancing actor

### Alternatives and workarounds

Just get the time when the Result arrives, which is a very imperfect way to do this.

matthias-wende-frequenz, 2023-08-10:
> We need to know when exactly the power was set.

I don't think we can do that, since we don't get this information from the hardware. 
The best you can do is subscribe to the return channel and use the timestamp of the returned message.

Marenz, 2023-11-14:
After some talking and with the new SDK changes which uses proposals instead of power requests, a good way to get what is needed here (probably?) is to use the `battery_pool.power()`  method..

llucax, 2023-11-15:
OK, closing as invalid.

closed 2023-11-15 by llucax