frequenz-floss/frequenz-sdk-python #556: Use `start_delay` when creating the resampler `Timer`

llucax, 2023-08-01:
### What's needed?

We needed to introduce a hack to make the timer start delayed so it can be synced with the resampler alignment. We need to remove that hack eventually.

### Proposed solution

Update to a new `channels` version that supports adding a start delay to the timer.

### Additional context

Blocked on:

* https://github.com/frequenz-floss/frequenz-channels-python/pull/145

Fixes a hack introduced in:

* https://github.com/frequenz-floss/frequenz-sdk-python/pull/520

matthias-wende-frequenz, 2023-10-17:
@llucax is this done?

llucax, 2023-10-18:
Nope, it might even need a new channels release first. Super low prio though, it is just code cleanup.

matthias-wende-frequenz, 2023-10-19:
Can we move it to post 1.0 then?

llucax, 2023-10-30:
I would leave it here for now, when upgrading to channels 1.0.0 it is probably trivial to change this as part of the upgrade.

matthias-wende-frequenz, 2023-11-21:
I'm not sure if that works. The reason is that we can set a start_delay but we can't set the now when initializing the timer and thus we can't guaranty that the timer is accurately aligned.

llucax, 2023-11-22:
What do you mean by *set the now*?

llucax, 2024-07-29:
This will be resolved with #802.

closed 2024-07-29 by llucax