frequenz-floss/frequenz-sdk-python #558: Get maximum data gap size

daniel-zullo-frequenz, 2023-08-02:
### What's needed?

Implement method to return the maximum data gap in a `MovingWindow` given timestamp or index ranges representing a window size

### Proposed solution

Compute the maximum data gap size iterating through the `MovingWindow` data gaps (underlying buffer)  for the given timestamp or index ranges representing a window size

### Use cases

Actors like peak-shaving needs this functionality to avoid implementing it on their ends.

daniel-zullo-frequenz, 2023-08-02:
fyi @matthias-wende-frequenz @idlir-shkurti-frequenz 

Please feel free to amend/correct the issue description accordingly

cwasicki, 2023-08-24:
After discussion with @idlir-shkurti-frequenz we change the scope of this issue. The max gap is currently used in peak shaving but is going to be replaced with a sum of gaps soon. Therefore we plan to provide access to a list of all gaps in a defined subset of the moving window. By this current and future requirements can be achieved using something like `max(g.duration for g in gaps)` or `sum(...)`, respectively.

llucax, 2023-09-05:
If there is a PR for this, can you link it to the issue @cwasicki ?

cwasicki, 2023-09-05:
Since it's not a breaking change I think we can move it post v1.

llucax, 2023-09-06:
@daniel-zullo-frequenz said he thought there was a PR for this, so I guess this is not the case, right?

cwasicki, 2023-09-06:
I have some changes locally which I can push and link

cwasicki, 2023-09-06:
https://github.com/frequenz-floss/frequenz-sdk-python/pull/643

cwasicki, 2023-09-06:
@daniel-zullo-frequenz Could you update title and description to something like `Expose gaps in moving window` (see https://github.com/frequenz-floss/frequenz-sdk-python/issues/558#issuecomment-1692123215)

cwasicki, 2023-10-02:
Discussed with @idlir-shkurti-frequenz that this can be closed as they can use the new moving window functions 
https://github.com/frequenz-floss/frequenz-sdk-python/blob/a06db70a218cf5ae2e7174e5387e162fa52217df/src/frequenz/sdk/timeseries/_moving_window.py#L357
and 
https://github.com/frequenz-floss/frequenz-sdk-python/blob/a06db70a218cf5ae2e7174e5387e162fa52217df/src/frequenz/sdk/timeseries/_moving_window.py#L366
to determine the data quality of the window in peak shaving.

closed 2023-10-02 by cwasicki