frequenz-floss/frequenz-sdk-python #575: Revamp modules/package structure

llucax, 2023-08-10:
### What's needed?

We need to have a consistent modules and packages structure, so:

1. `import`s are clean and clear
2. Chances of ending with cyclic imports are minimized or completely removed (#84)
3. We minimize the amount of unused code that is indirectly imported

### Proposed solution

This is still to be researched and decided.

### Additional context

There are a few other issues related to this topic:

* #222 

  With the current import structure, this doesn't always make sense, for example we have a `config` module/package. Using names that are so generic are not always that clear with this scheme, as the user could also have some module called `config`.

  Related issues and discussions:

  * https://github.com/frequenz-floss/frequenz-repo-config-python/discussions/56
  * https://github.com/frequenz-floss/frequenz-sdk-python/discussions/209
  * https://github.com/frequenz-floss/frequenz-repo-config-python/issues/14 (it would be good to check what kind of enforcement this does before going for this)

* #118

  Is assuming we use google-style imports.

* Point 3 (minimize the amount of unused code that is indirectly imported) is also important. Since Python executes the code, and importing a modules can have arbitrary code execution, importing could be very expensive, and it could happen that just to import a small utility function, you end up importing the whole SDK, so it would be good to have this in consideration when thinking about a package/modules structure.

llucax, 2024-07-29:
I think the main issues with cyclic imports are:

* `frequenz.sdk.timeseries` package depends on `frequenz.sdk.actor`:
  * `BackgroundService` (fixed when moving to `core`)
  * `ChannelRegistry` (should be moved to `timeseries` or `_internal`.)
  * `ComponentMetricRequest` (by `_voltage_streamer`, this module should be moved to `frequenz.sdk.microgrid`)
  * `formula_engine` sends requests to the resampler, the resampler actor should be moved to `frequenz.sdk.timeseries.resampling` or maybe just `frequenz.sdk.resampling`.

* `frequenz.sdk.timeseries` package depends on `frequenz.sdk.microgrid`:
  * Basically anything depending on `microgrid` in `timeseries` should be moved to `microgrid`: `_voltage_streamer`, `*_pool`, `consumer`, `producer`, `grid`, `logical_meter`

* `frequenz.sdk.actor` depends on `frequenz.sdk.microgrid`
  * Actors using the microgrid (power distributing, data sourcing, power managing) should be moved to `frequenz.sdk.microgrid`

* `frequenz.sdk.microgrid` depends on `frequenz.sdk.actor`
  * Inter-actor dependencies (`Proposal`, `ReportRequest`, etc.) should probably live in their own module, that only define PODs, so they don't introduce any indirect dependency.

I think the latest point is actually the one that brings the biggest challenge in breaking cyclic imports and having a good structure.

### Proposal

In summary I propose a structure like this:

* `frequenz.sdk.config`: `ConfigManagingActor` and related (depends only on `core`/`actors` and `channels`)
* `frequenz.sdk._internal_.channels`: `ChannelRegistry`
* `frequenz.sdk.microgrid._data_sourcing`: `DataSourcingActor` and related
* `frequenz.sdk.microgrid._data_sourcing.messages`: Request and response messages (no dependencies)
* `frequenz.sdk.microgrid._power_distribution`: `PowerDistributingActor` and related (maybe even inside `power_managing`)
* `frequenz.sdk.microgrid._power_distribution.messages`: Request and response messages (no dependencies)
* `frequenz.sdk.microgrid._power_managing`: `PowerDistributingActor` and related
* `frequenz.sdk.microgrid._power_managing.messages`: Request and response messages (no dependencies)
* `frequenz.sdk.microgrid._resampling`: `ResamplingActor`, for now it depends on the microgrid, as it uses the `DataSourcingActor`, as long as we don't break that dependency I think it should live in this namespace.
* `frequenz.sdk.microgrid._resampling.messages`: Request and response messages (no dependencies)
* `frequenz.sdk.microgrid._pipeline` and other internal stuff
* `frequenz.sdk.timeseries`: `Sample` and related, `MovingWindow`, `PeriodicFeatureExtractor`, `FormulaEngine` (`RingBuffer`), `Resampler` (all of these have only dependency to the `core`/`actors`, `channels` and `quantities` repos)

Inside the `frequenz.sdk.microgrid` package, which will turn out to be pretty big, we still need to make sure to avoid cyclic dependencies, but I think that should be easy if we do the `.messages` thing, that should be the only *public* interface for actors inter-communication.

This structure will depend on making the switch to use the `core`/`actors` and `quantities` repos.

I say `core`/`actors` because at some point I though we could get away with putting `Actor` in `core`, but now I think we still need a separate repo, there is already the idea to have the `ExpiringActor` in a common repo and actually the concept of an actor comes from background service + channels, so at some point if we implement other base actors, we might want to depend on `channels` in the `actors` repo, and we can't do it if they live in `core` because `channels` will depend on `core` so we end up with a cyclic dependency.

cwasicki, 2025-02-12:
LGTM

llucax, 2025-02-13:
Soon™  😆

llucax, 2025-02-13:
I think the structure will be improved a lot naturally after we split the SDK a bit more, on top of these:

* https://github.com/frequenz-floss/frequenz-sdk-python/issues/852
* https://github.com/frequenz-floss/frequenz-sdk-python/issues/851
* https://github.com/frequenz-floss/frequenz-core-python/

I also want to split out the `config` package to a separate repo. And in the future probably the `timeseries` and `microgrid` packages too, but these won't probably happen before 1.0.0.

About google style imports, I feel it is already too late for that, we are using `from module import Symbol` in too many places by now, my feeling at the moment is cleaning this up will be an epic task that it is probably not worth it.