frequenz-floss/frequenz-sdk-python #640: The API documentation now is called "sdk" in the top navigation bar

llucax, 2023-09-05:
### What happened?

The generated website now says "sdk" in the top navigation bar.

[image]

### What did you expect instead?

It should say "API Reference", as it was up to v0.24.0

[image]

### Affected version(s)

v0.25.0

### Affected part(s)

Documentation (part:docs)

### Extra information

Nothing really changed, so this is probably a regression in a bumped dependency.

llucax, 2023-09-05:
A bisection points to e7a3c3a1d5855f3de5a4072a338d236ff6a3cba8. So we'll have to have a look at what changed in repo config from 0.5.2 to 0.6.1.

shsms, 2023-09-05:
But clicking on it takes us to this link directly: https://frequenz-floss.github.io/frequenz-sdk-python/v0.25/reference/frequenz/sdk/

and not to the top level page, like it used to.  That's very nice.  And the top level page for `API reference` appears to have gone away completely.

llucax, 2023-09-05:
No, that was a separate issue, fixed in v0.6.0. We added the file `src/conftest.py` for testing the examples, and the documentation for it was generated too (not it is excluded), so the link would take you to `conftest` because it was the most top-level module.

llucax, 2023-09-05:
The only change of dependencies I see there is in https://github.com/frequenz-floss/frequenz-repo-config-python/commit/d3cd805ac45f49d170efce5bb7b698933a16e671 (bumping from 9.1.16 to 9.1.21 of `mkdocs-material` and doing `pip install mkdocs-material==9.1.16` does solve the issue.

Strangely enough, doing a `pip install mkdocs-material==9.1.21` doesn't break it, but doing `pip install mkdocs-material==9.2.0` does, so I don't know what weird interaction was there but it seems like the update to 9.2.0+ is to blame, which happened in 889f9a43, between v0.24.0 and v0.25.0.

matthias-wende-frequenz, 2023-09-05:
removing the `section-index` plugin resolves the issue

llucax, 2023-09-05:
Reported upstream: https://github.com/squidfunk/mkdocs-material/issues/5971

llucax, 2023-09-05:
> removing the section-index plugin resolves the issue

Oh, interesting.

llucax, 2023-09-05:
Removing `mkdocs-section-index` is not appropriate either, if we do so, we get:

[image]

(Note that you need to click on the `index` menu for each package)

While with the plug-in we get no `index` on each package, which is what we want:

[image]

llucax, 2023-09-05:
`mkdocs-material` already closed the issue (as he is not looking at it saying it's an "upstream" problem), so trying with `mkdocs-section-index` and the most other likely cause of it:

* https://github.com/oprypin/mkdocs-section-index/issues/18

llucax, 2023-09-06:
For now I think the easiest workaround is to downgrade `mkdocs-material`. Done in #642.

llucax, 2023-09-06:
Marking as blocked as now we need to wait for upstream to see if we can upgrade the dependency again.

llucax, 2023-09-08:
Unblocking as it is fixed upstream, but we are still switching to the native feature in `mkdocs-material`.

PR #642 (2023-09-08): Fix mkdocs navigation bar API link
The `section-index` plugin has a bug[^1] that causes the navigation bar to display a wrong text for the **API Reference** link. There is a fix for it, but `mkdocs-material` already provides the feature natively[^2], so we can just use it instead, as it is much less likely to have issues.

[^1]: https://github.com/oprypin/mkdocs-section-index/issues/18
[^2]: https://squidfunk.github.io/mkdocs-material/setup/setting-up-navigation/#section-index-pages
Fixes #640.
commit d05f12db (2023-09-08): Fix mkdocs navigation bar API link (#642)

The `section-index` plugin has a bug[^1] that causes the navigation bar
to display a wrong text for the **API Reference** link. There is a fix
for it, but `mkdocs-material` already provides the feature natively[^2],
so we can just use it instead, as it is much less likely to have issues.

[^1]: https://github.com/oprypin/mkdocs-section-index/issues/18
[^2]:
https://squidfunk.github.io/mkdocs-material/setup/setting-up-navigation/#section-index-pages
Fixes #640.
closed 2023-09-11 by llucax