frequenz-floss/frequenz-sdk-python #646: Inconsistent gap treatment in ring buffer

cwasicki, 2023-09-06:
### What happened?

If there is no data in the ring buffer (e.g. on start up) or no valid data (e.g. only None values), there is inconsistent behavior for gaps and `__len__`, which relies on gaps.

### What did you expect instead?

Example on startup (expected 1 gap over the full window):
https://github.com/frequenz-floss/frequenz-sdk-python/blob/01b8d9d1886f298106a505bf68f6c03ced380580/tests/timeseries/test_ringbuffer.py#L211

Example for no valid data (expected length of zero):
https://github.com/frequenz-floss/frequenz-sdk-python/blob/01b8d9d1886f298106a505bf68f6c03ced380580/tests/timeseries/test_ringbuffer.py#L252

### Affected version(s)

v0.x.x

### Affected part(s)

Data pipeline (part:data-pipeline)

cwasicki, 2023-10-02:
Adding it to rc2 milestone because it would ideally be addressed for v1.0 release. But could also be addressed later since it's a bug fix which affects a very special case only.

matthias-wende-frequenz, 2023-10-17:
@cwasicki Do you plan working on this in the next 2 weeks? Otherwise I'd move it to post 1.0.

cwasicki, 2023-10-17:
@matthias-wende-frequenz Not planning to work on this, in fact I think this is better assigned someone who is more familiar with the gap algorithm.

cwasicki, 2024-08-09:
It seems this is causing problems on one of our sites where the first sample we receive after start-up is None. This results in the moving window to assume that it is completely filled but apart from the latest value contains only garbage data since the ring buffer is initialized with `np.empty`. 

Therefore we should address this for v1.0 because this makes the moving window impossible to use for our forecasting use-cases.

cwasicki, 2024-08-12:
Fixed by #1034

PR #1034 (2024-08-09): Fix gap in ring buffer when updating missing values
If there are no gaps in the ring buffer and the new value is missing and creates a gap in time, we need to start the new gap after the latest value from before.

Fixes #646.
commit aa259ecf (2024-08-09): Fix gap in ring buffer when updating missing values (#1034)

If there are no gaps in the ring buffer and the new value is missing and
creates a gap in time, we need to start the new gap after the latest
value from before.

Fixes #646.
closed 2024-08-12 by cwasicki